sparkfun/SparkFun_RTK_Everywhere_Firmware #226: Run Bluetooth Test causes exception

doug-foster, 2024-04-05:
### Subject of the issue
Run Bluetooth Test causes exception

### Your workbench
* EVK
* release_candidate branch - 2024-04-05@14-43-51 - RTK_Everywhere_rc

### Steps to reproduce
1.  serial menu
2. s - system menu
3. h - debug hardware
4. 2 - Run Bluetooth Test

### Expected behavior
Should run some kind of test.

### Actual behavior
terminal display follows ...

Rover Accuracy (m): 0.220, SIV: 29 GNSS State: 3D Fix
SparkFun RTK EVK d99.99
** Bluetooth Low-Energy broadcasting as: EVK Rover-8F06 **
Menu: Main
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
6) Configure WiFi
7) Configure Network
e) Configure Ethernet
f) Firmware Update
i) Configure Corrections Priorities
n) Configure NTP
u) Configure User Profiles
r) Configure Radios
p) Configure PointPerfect
s) Configure System
+) Enter Command line mode
x) Exit
s

System Status
2024-04-05 20:15:03.344
Mode: Rover
GNSS: Online - ZED-F9P firmware: HPG 1.32
Module ID: 3D4052C618
SIV: 29, HPA (m): 0.221, Lat: 35.60599394, Lon: -78.79442815, Altitude (m): 91.4
Display: Online
microSD: Offline
NEO-D9S L-Band: Online - 
No Keys / Corrections Received Failed / Eb/N0[dB] (>9 is good): 0.00 - NEO-D9S firmware: PMP 1.04
Bluetooth Low Energy (8F06): Online
WiFi MAC Address: E0:5A:1B:D8:8F:04
Ethernet cable: connected
Ethernet MAC Address: E0:5A:1B:D8:8F:07
Ethernet IP Address: 192.168.2.36
System Uptime: 0 00:42:16.323 (Resets: 0)
Filtered by parser: 0 NMEA / 0 RTCM / 0 UBX

Menu: System
-----  Mode Switch  -----
B) Switch to Base mode
N) Switch to NTP Server mode
R) Switch to Rover mode
W) Switch to WiFi Config mode
-----  Settings  -----
b) Set Bluetooth Mode: BLE
c) Shutdown if not charging: Disabled
d) Debug software
e) Echo User Input: On
h) Debug hardware
n) Debug network
o) Configure RTK operation
p) Configure periodic print messages
r) Reset all settings to default
z) Set time zone offset: 00:00:00
~) Setup button: Enabled
x) Exit
h

Menu: Debug Hardware
1) Print battery status messages: Enabled
2) Run Bluetooth Test
3) Print RTC resyncs: Disabled
4) Print log file messages: Disabled
5) Print log file status: Enabled
6) Run Logging Test: Disabled
7) Print SD and UART buffer sizes: Disabled
8) Print messages with bad checksums or CRCs: Disabled
9) Print GNSS Debugging: Disabled
10) Print Correction Debugging: Disabled
11) Print Tilt/IMU Debugging: Disabled
12) Print Tilt/IMU Compensation Debugging: Disabled
14) PSRAM (online): Enabled
e) Erase LittleFS
t) Test Screen
r) Force system reset
x) Exit
2
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4013833f  PS      : 0x00060130  A0      : 0x800f46bc  A1      : 0x3ffd2bc0  
A2      : 0x00000000  A3      : 0x00038400  A4      : 0x00000000  A5      : 0x000000ff  
A6      : 0x000000ff  A7      : 0x00000000  A8      : 0x00000070  A9      : 0x3ffd2bc0  
A10     : 0x00000001  A11     : 0x41524001  A12     : 0x00000000  A13     : 0x00000003  
A14     : 0x0000044c  A15     : 0x3ffd2a20  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x40093330  LEND    : 0x4009334c  LCOUNT  : 0x00000000  

Backtrace: 0x4013833c:0x3ffd2bc0 0x400f46b9:0x3ffd2c10 0x40101236:0x3ffd2ed0 0x40102cb1:0x3ffd2ef0 0x401067ca:0x3ffd2f60 0x401068e8:0x3ffd2fa0 0x40106958:0x3ffd2fc0 0x4013c4b9:0x3ffd2fe0

ELF file SHA256: b14c914d060833a9

E (6469) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
E (844) esp_core_dump_fl.Ê¯h: No core dump partition found!
E (844) esp_core_dump_flash: No core dump partition found!

LittleFS Started
PSRAM Size (bytes): 4192107
I2C Devices:
  0x42 - u-blox ZED-F9P GNSS Receiver
  0x43 - u-blox NEO-D9S Correction Data Receiver
I2C Devices:
  0x3D - SSD1306 OLED Driver
Display started
SparkFun RTK EVK d99.99
Reset reason: ESP_RST_PANIC
ZED-F9P firmware: HPG 1.32
GNSS ZED online
Profile 'Profile1' loaded

nseidle, 2024-04-05:
This is a vestige menu. Lower priority. It doesn't reset on Torch. I suspect we can remove this menu altogether.

doug-foster, 2024-04-25:
Still a problem in https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/blob/main/RTK_Everywhere_Firmware_RC-Apr_24_2024.bin

doug-foster, 2024-05-17:
Running the test with the latest RC (91ca899) on EVK 1.1 does not cause a panic, but correctly outputs:
"Bluetooth Low Energy (8ECE): Online"
It would seem that it works & serves a purpose, so it should NOT be deleted from the debug hardware menu. @nseidle  agrees, closing the issue.

closed 2024-05-17 by doug-foster