sparkfun/SparkFun_RTK_Everywhere_Firmware #227: Conf WiFi - form does not display stored/retrieved value for Ethernet HTTP Port (should be default 80)

doug-foster, 2024-04-05:
### Subject of the issue
using "Conf WiFi", the displayed web form does not display stored/retrieved value for Ethernet HTTP Port (should be default 80) - it is empty. Even if not changing "Ethernet Configuration" - say changing "System Configuration" - error checking the form displays an error since no value is in the field. Behavior is the same whether WiFi mode is "AP" or "WiFi."

### Your workbench
* EVK
* release_candidate branch - 2024-04-05@14-43-51 - RTK_Everywhere_rc

### Steps to reproduce
1.  MODE button
2. Cfg Eth
3. connect web browser to displayed address

### Expected behavior
field should be filled even if never config'd. entering value "80" removes error. Click "Save Configuration", "Exit and Reset", EVK restarts. repeat process, same result - field does not appear to be saved. 

I also saw a behavior where even though the EVK Bluetooth mode was set to "Both", the web form showed "Off". In fact, the "Both" option does not appear on the select field for the form field, only "BLE" and "SPP"

### Actual behavior
[image]
[image]

doug-foster, 2024-04-05:
Output from serial terminal session when saving web form.

Base Mode - SIV: 31
Time elapsed: 14 Accuracy (m): 0.843 SIV: 31
Time elapsed: 14 Accuracy (m): 0.843 SIV: 31
STATE_BASE_TEMP_SURVEY_STARTED --> STATE_DISPLAY_SETUP, 2024-04-05 21:26:28.067
STATE_DISPLAY_SETUP --> STATE_BASE_NOT_STARTED, 2024-04-05 21:26:41.722
Bluetooth Low-Energy broadcasting as: EVK Base-8F06
STATE_BASE_NOT_STARTED --> STATE_BASE_TEMP_SETTLE, 2024-04-05 21:26:43.942
Base Mode - SIV: 31
STATE_BASE_TEMP_SETTLE --> STATE_DISPLAY_SETUP, 2024-04-05 21:26:43.976
STATE_DISPLAY_SETUP --> STATE_WIFI_CONFIG_NOT_STARTED, 2024-04-05 21:26:51.596
Connecting WiFi... 

Network Configuration:
----------------------
         SSID: Cisco
  WiFi Status: 3
WiFi Strength: -37 dBm
          MAC: E0:5A:1B:D8:8F:04
           IP: 192.168.2.33
       Subnet: 255.255.255.0
      Gateway: 192.168.2.1
        DNS 1: 192.168.2.1
        DNS 2: 0.0.0.0
        DNS 3: 0.0.0.0

newSettings len: 11396
platformPrefix,EVK,rtkFirmwareVersion,d99.99,zedFirmwareVersion,ZED-F9P Firmware: HPG 1.32 ID: 3D4052C618,zedFirmwareVersionInt,132,deviceBTID,Device Bluetooth ID: 8F06,printDebugMessages,false,enableSD,true,enableDisplay,true,maxLogTime_minutes,1440,maxLogLength_minutes,1440,observationSeconds,60,observationPositionAccuracy,5.00,baseTypeSurveyIn,true,baseTypeFixed,false,fixedBaseCoordinateTypeECEF,true,fixedBaseCoordinateTypeGeo,false,fixedEcefX,-1280206.568,fixedEcefY,-4716804.403,fixedEcefZ,4086665.484,fixedLat,40.09029479,fixedLong,-105.18505761,fixedAltitude,1560.0890,dataPortBaud,230400,radioPortBaud,57600,zedSurveyInStartingAccuracy,1.0,measurementRateHz,4.00,debugGnss,false,enableHeapReport,false,enableTaskReports,false,dataPortChannel,0,spiFrequency,16,enableLogging,true,enableARPLogging,false,ARPLoggingInterval,10,sppRxQueueSize,2048,sppTxQueueSize,32,dynamicModel,0,lastState,1,enableResetDisplay,false,resetCount,0,enableExternalPulse,true,externalPulseTimeBetweenPulse_us,1000000,externalPulseLength_us,100000,externalPulsePolarity,1,enableExternalHardwareEventLogging,false,enableUART2UBXIn,false,ubxConstellationsGPS,true,ubxConstellationsSBAS,true,ubxConstellationsGalileo,true,ubxConstellationsBeiDou,true,ubxConstellationsGLONASS,true,profileName,Profile1,profileNumber,0,profile0Name,1: Profile1,profile1Name,2: ,profile2Name,3: ,profile3Name,4: ,profile4Name,5: ,profile5Name,6: ,profile6Name,7: ,profile7Name,8: ,serialTimeoutGNSS,1,pointPerfectDeviceProfileToken,,pointPerfectCorrectionsSource,0,autoKeyRenewal,true,pointPerfectClientID,,pointPerfectBrokerHost,,pointPerfectLBandTopic,,pointPerfectCurrentKey,,pointPerfectCurrentKeyDuration,0,pointPerfectCurrentKeyStart,0,pointPerfectNextKey,,pointPerfectNextKeyDuration,0,pointPerfectNextKeyStart,0,lastKeyAttempt,0,updateGNSSSettings,false,LBandFreq,1556290000,timeZoneHours,0,timeZoneMinutes,0,timeZoneSeconds,0,enablePrintState,false,enablePrintPosition,false,enablePrintIdleTime,false,enablePrintBatteryMessages,true,enablePrintRoverAccuracy,true,enablePrintBadMessages,false,enablePrintLogFileMessages,false,enablePrintLogFileStatus,true,enablePrintRingBufferOffsets,false,enablePrintStates,true,enablePrintDuplicateStates,false,enablePrintRtcSync,false,bluetoothRadioType,1,runLogTest,false,espnowBroadcast,true,antennaHeight,0,antennaReferencePoint,0.0,echoUserInput,true,uartReceiveBufferSize,2048,gnssHandlerBufferSize,4096,enablePrintBufferOverrun,false,enablePrintSDBuffers,false,periodicDisplay,0,periodicDisplayInterval,15000,rebootSeconds,-1,forceResetOnSDFail,false,minElev,10,coordinateInputType,0,lbandFixTimeout_seconds,180,minCNO_F9P,6,serialGNSSRxFullThreshold,50,btReadTaskPriority,1,gnssReadTaskPriority,1,handleGnssDataTaskPriority,1,btReadTaskCore,1,gnssReadTaskCore,1,handleGnssDataTaskCore,1,gnssUartInterruptsCore,1,bluetoothInterruptsCore,1,i2cInterruptsCore,1,shutdownNoChargeTimeout_s,0,disableSetupButton,false,useI2cForLbandCorrections,true,useI2cForLbandCorrectionsConfigured,false,enablePrintEthernetDiag,false,ethernetDHCP,true,ethernetIP,192.168.0.123,ethernetDNS,194.168.4.100,ethernetGateway,192.168.0.1,ethernetSubnet,255.255.255.0,httpPort,80,debugWifiState,false,wifiConfigOverAP,0,wifiNetwork0SSID,Cisco,wifiNetwork0Password,wewerewatchinghousetonight2!,wifiNetwork1SSID,,wifiNetwork1Password,,wifiNetwork2SSID,,wifiNetwork2Password,,wifiNetwork3SSID,,wifiNetwork3Password,,defaultNetworkType,2,debugNetworkLayer,false,enableNetworkFailover,true,printNetworkStatus,true,mdnsEnable,true,debugMqttClientData,false,debugMqttClientState,false,useEuropeCorrections,false,debugNtp,false,ethernetNtpPort,123,enableNTPFile,false,ntpPollExponent,6,ntpPrecision,-20,ntpRootDelay,0,ntpRootDispersion,1000,ntpReferenceId,GPS,debugNtripClientRtcm,false,debugNtripClientState,false,enableNtripClient,false,ntripClient_CasterHost,rtk2go.com,ntripClient_CasterPort,2101,ntripClient_CasterUser,test@test.com,ntripClient_CasterUserPW,,ntripClient_MountPoint,bldr_SparkFun1,ntripClient_MountPointPW,undefined,ntripClient_TransmitGGA,true,debugNtripServerRtcm,false,debugNtripServerState,false,enableNtripServer,false,ntripServer_StartAtSurveyIn,false,ntripServer_CasterHost_0,rtk2go.com,ntripServer_CasterPort_0,2101,ntripServer_CasterUser_0,test@test.com,ntripServer_CasterUserPW_0,,ntripServer_MountPoint_0,bldr_dwntwn2,ntripServer_MountPointPW_0,WR5wRo4H,ntripServer_CasterHost_1,,ntripServer_CasterPort_1,0,ntripServer_CasterUser_1,,ntripServer_CasterUserPW_1,,ntripServer_MountPoint_1,,ntripServer_MountPointPW_1,,ntripServer [... cut ...] ryDefaults,false,enableResetDisplay,false,ethernetDHCP,true,enableTcpClientEthernet,false,fileSelectAll,false,
Unknown 'ntripServer_CasterHost': 0.000
Unknown 'ntripServer_CasterPort': 2101.000
Unknown 'ntripServer_CasterUser': 0.000
Unknown 'ntripServer_CasterUserPW': 0.000
Unknown 'ntripServer_MountPoint': 0.000
Unknown 'ntripServer_MountPointPW': 0.000
Unknown 'ethernetHttpPort': 80.000
Unknown 'ethernetTcpPort': 0.000
Unknown 'hostForTCPClient': 0.000
Unknown 'enableTcpClientEthernet': 0.000

PaulZC, 2024-04-11:
Hi Doug (@doug-foster ),

Thanks for reporting... Yeah. We've got some work to do here. There is a small mismatch between what the JS web page expects / produces, and what the firmware expects. As a minimum, we need to fix:

```
Unknown 'ntripServer_CasterHost': 0.000
Unknown 'ntripServer_CasterPort': 2101.000
Unknown 'ntripServer_CasterUser': 0.000
Unknown 'ntripServer_CasterUserPW': 0.000
Unknown 'ntripServer_MountPoint': 0.000
Unknown 'ntripServer_MountPointPW': 0.000
Unknown 'ethernetHttpPort': 80.000
Unknown 'ethernetTcpPort': 0.000
Unknown 'hostForTCPClient': 0.000
Unknown 'enableTcpClientEthernet': 0.000
```

We are at the point where we need to add / update the Config WiFi HTML and JS for the Torch. It is a very different beast to the EVK / Facet. The UM980 messages are completely different to those for the ZED-F9P. If I'm not careful, I'll end up with two large blocks of HTML and JS, one for the ZED and a second for the UM980. It'll be messy to maintain. And will be further complicated when we add the mosaic X5 into the mix... It would indeed be _so_ much better if the firmware could construct the webpage - as JSON maybe - zips it - and throws it to a very simple JS plus HTML page - which unzips it and does the necessary. I'll have a good chat with @nseidle about this.

I'll say the same thing in #230 too.

Cheers!
Paul

PaulZC, 2024-04-23:
The ```httpPort``` issue is resolved in release_candidate...

doug-foster, 2024-04-25:
I confirm default httpPort now displays as 80 in https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/blob/main/RTK_Everywhere_Firmware_RC-Apr_24_2024.bin. However there are still sync issues between the web menu and the settings/serial menu - see latest comment on Issue #230.