sparkfun/SparkFun_RTK_Everywhere_Firmware #228: Remove ntripServer_StartAtSurveyIn setting

nseidle, 2024-04-05:
[image]

    8) Select survey-in radio: Bluetooth

This menu controls ntripServer_StartAtSurveyIn and looks like a vestige menu/setting. It may have been used once on the old RTK Firmware, but ntripServer_StartAtSurveyIn is also no longer used in that firmware as well. I'm going to remove.

doug-foster, 2024-04-05:
btw - #8 does not have a counterpart on the web config form.  no action needing, I was just mentioning since I'm comparing the serial menu to the web config menu.
[image]

nseidle, 2024-04-07:
A good sanity check - thank you!

commit 02ac1103 (2024-04-05): Remove ntripServer_StartAtSurveyIn

Fix #228
closed 2024-05-01 by nseidle