sparkfun/SparkFun_RTK_Everywhere_Firmware #229: serial terminal can crash EVK if in "Cfg Eth" mode

doug-foster, 2024-04-05:
### Subject of the issue
If "Cfg Eth" is chosen via the MODE button on the front panel, and the serial menu is brought up on terminal session, choosing "s" crashes the EVK. It appears other menu selections seem to work properly.

### Your workbench
- EVK
- release_candidate branch - 2024-04-05@14-43-51 - RTK_Everywhere_rc
- using POE via Ethernet to power the EVK

### Steps to reproduce
1.  MODE, select "Cfg Eth"
2. enter serial menu
3. choose s) Configure System

### Expected behavior
I would think you might want to limit access to serial menu commands while in "Cfg Eth" or "Cfg WiFi" MODE - why would you want to use both config methods at the same time? - but it should not crash the EVK. 

### Actual behavior
[image]

output from a serial terminal session:

SparkFun RTK EVK d99.99
** Bluetooth SPP broadcasting as:.**
Menu: Main
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
6) Configure WiFi
7) Configure Network
e) Configure Ethernet
f) Firmware Update
i) Configure Corrections Priorities
n) Configure NTP
u) Configure User Profiles
r) Configure Radios
p) Configure PointPerfect
s) Configure System
+) Enter Command line mode
x) Exit
s

System Status
2024-04-05 22:52:21.941
Mode: Ethernet Config
GNSS: Offline
Display: Online
microSD: Offline
NEO-D9S L-Band: Offline - 
No Keys / Corrections Received Failed / Eb/N0[dB] (>9 is good): 0.00 - NEO-D9S firmware: 
Bluetooth (8F06): GNSS Offline
WiFi MAC Address: E0:5A:1B:D8:8F:04
Ethernet cable: ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
E (978) esp_core_dump_fl.Ê¯h: No core dump partition found!
E (978) esp_core_dump_flash: No core dump partition found!

LittleFS Started
PSRAM Size (bytes): 4192107
I2C Devices:
.0x42 - u-blox ZED-F9P GNSS Receiver
.0x43 - u-blox NEO-D9S Correction Data Receiver
I2C Devices:
.0x3D - SSD1306 OLED Driver
Display started
SparkFun RTK EVK d99.99
Reset reason: ESP_RST_INT_WDT
ZED-F9P firmware: HPG 1.32
GNSS ZED online
Profile 'Profile1' loaded

PaulZC, 2024-04-11:
Hi Doug,

I'll have to dig through the code, but I'm not yet convinced this is an actual crash. Rather the firmware exiting Cfg Eth mode - which requires a restart... The ```Reset reason: ESP_RST_INT_WDT``` is a bit worrying though...

Cheers,
Paul

PaulZC, 2024-04-26:
Thanks for reporting... Resolved in PR #259

nseidle, 2024-05-01:
This is in [v1.0](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.0).

closed 2024-05-01 by nseidle