sparkfun/SparkFun_RTK_Everywhere_Firmware #231: EVK Selecting the wrong ZTP token

nseidle, 2024-04-09:
pointperfectGetToken() gets called when a provision event happens. This function looks at the available hardware and selects the ZTP token.

In the case of the EVK, a user may have select IP corrections (only) but the logic points them to the ppLbandPaidToken, which L-Band reception may or may not be available in the user's locale.

Solution: 
* Make the PointPerfect menu display service options that are compatible with the available hardware
* Make pointperfectGetToken() pick the token connected with that service level. ie if a user wants IP access only, they should be directed to the IP only ZTP token.

PaulZC, 2024-04-11:
Fixed in release_candidate. We now have LBAND_IP FREE and PAID tokens. The EVK defaults to these.

This avoids devices having to be whitelisted on both IP-only and L-Band-only plans, and potentially being charged for both plans...

PaulZC, 2024-04-11:
Removing the label. The code changes are in the release_candidate branch. But it hasn't yet been built into a RC .bin.

nseidle, 2024-04-11:
RC pushed. Tag added.

nseidle, 2024-05-01:
This is in [v1.0](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.0).

closed 2024-05-01 by nseidle