sparkfun/SparkFun_RTK_Everywhere_Firmware #232: ESP-NOW not working when WiFi is enabled

nseidle, 2024-04-09:
If ESP-NOW radio link is enabled, and WiFi is used (for NTRIP, TCP, etc) the ESP-NOW communication will fail.

This is due to a variety of things:

* By default, ESP-NOW uses WiFi channel 0 to communicate.
* When WiFi is turned on, the device will attempt to connect to the SSID the user provides. The AP for that SSID will likely be on a different channel and the WiFi.begin() moves the 2.4GHz radio to that channel. ESP-NOW is now broken.
* Additionally, with WiFi.begin(), the 2.4GHz enters a power savings mode. ESP-NOW relies on having the radio on all the time to capture incoming packets.

Possible solution 1:

We have demonstrated that ESP-NOW can transmit to a receiver (that has WiFi on) once the correct channel number is set, using the following code:

      esp_wifi_set_promiscuous(true);
      esp_err_t response = esp_wifi_set_channel(channelNumber, WIFI_SECOND_CHAN_NONE);
      if (response != ESP_OK)
      {
        char responseString[100];
        esp_err_to_name_r(response, responseString, sizeof(responseString));
        Serial.printf("setChannel failed: %s\r\n", responseString);
      }
      esp_wifi_set_promiscuous(false);

Both the transmitter and receiver can identify the WiFi channel the SSID/AP is using with the following code:

    int32_t getWiFiChannel(const char *ssid) {
      if (int32_t n = WiFi.scanNetworks()) {
        for (uint8_t i = 0; i < n; i++) {
          if (!strcmp(ssid, WiFi.SSID(i).c_str())) {
            return WiFi.channel(i);
          }
        }
      }
      return 0;
    }

Additionally, we can prevent the WiFi from putting the radio into power saving mode with

      WiFi.setSleep(false); //We must disable sleep so that ESP-NOW can readily receive packets
    
However, the transmitter needs to know the channel of any receiver. If the receiver is the one using WiFi (ie, a Rover that needs WiFi to communicate its position over TCP to a GIS app) then the transmitter needs to be able to communicate with this network (to obtain it's channel) or the receiver needs to communicate the channel number back to the transmitter (catch-22).

We could write code to make the transmitter hop channels (there's ~14) until it successfully delivers a transmission to a peer. But if there are multiple peers (aka Rovers), each with their own WiFi connections and channels, the transmitter (aka Base) will have multiple channels if may need to transmit on. This is not really tenable.

Possible Solution 2:

* Document the limitations of ESP-NOW and WiFi.
* Consider adding channel get/set options to the ESP-NOW menu.
* Add WiFi.setSleep(false) to ESP-NOW startup.

This would allow the subset of users who are affected by this issue a reasonable path forward.

nseidle, 2024-04-09:
ESPNOW_Transmit_NoWiFi and ESPNOW_Recieve_WithWiFi sketches have been added to repo that demonstrate the breaking behavior.

doug-foster, 2024-04-09:
Any ideas here? https://forum.arduino.cc/t/use-esp-now-and-wifi-simultaneously-on-esp32/1034555/6

nseidle, 2024-04-10:
@doug-foster - I don't see anything there that spreads any additional light on the problem. Perhaps I'm missing something.

nseidle, 2024-04-16:
ESP-Now is now working nicely with WiFi. [Docs updated](https://docs.sparkfun.com/SparkFun_RTK_Everywhere_Firmware/menu_radios/#setting-the-wifi-channel).

nseidle, 2024-05-01:
This is in [v1.0](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.0).

closed 2024-05-01 by nseidle