sparkfun/SparkFun_RTK_Everywhere_Firmware #246: PointPerfect - no way to configure PP using account MQQT credentials

doug-foster, 2024-04-23:
### Subject of the issue
No options to enter PointPerfect account credentials.

### Your workbench
- EVK
- https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/tree/release_candidate/Firmware/RTK_Everywhere

### Steps to reproduce
[image]

### Expected behavior
Accept MQQT credentials via both web & serial config menus
[image]

### Actual behavior
see screen snapshot of serial menu

nseidle, 2024-04-23:
[image]

It looks like your token is malformed. Are you using a release candidate binary or compiling locally? The Release Candidate binaries include this information but locally compiled binaries [are forced to use](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/main/Firmware/RTK_Everywhere/menuPP.ino#L8) the `DEVELOPMENT_TOKEN`.

If you'd like to use your own credentials, create a ZTP profile, copy/paste that token # into the 'ThingStream Device Profile Token' menu, then you can provision the device on your account. Only one other person has requested a custom key path so it's not well trodden.

doug-foster, 2024-04-24:
Compiled locally. [RTK_Everywhere_Firmware_RC-Apr_17_2024.bin](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/blob/main/RTK_Everywhere_Firmware_RC-Apr_17_2024.bin) would not run (but would when compiled locally), stuck in boot loader loop. I can retry with both [RTK_Everywhere_Firmware_RC-Apr_23_2024.bin](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/blob/main/RTK_Everywhere_Firmware_RC-Apr_23_2024.bin) and a local compile ... however ...

I think I understand now. PointPerfect ZTP (Zero Touch Provisioning & whitelisting a device) profiles are not available under a free PP developer (IP only) account.

RTK_Everywhere can only be used with a full, _paid_ PP subscription (which could yield a ZTP profile token). The lowest cost PP account level to support this would probably be "[PointPerfect 100H - 100 hour IP only @ $5.90 per month](https://portal.thingstream.io/pricing/laas/laaspointperfect)." I'll have to confirm.

nseidle, 2024-04-29:
I'm going to close this. The firmware is designed so that users do not need a ThingStream account. If they desire, they can use their own ZTP token, or they can enter the key/expirations directly.

closed 2024-04-29 by nseidle

doug-foster, 2024-04-29:
Agreed - ZTP tokens can be created for _all_ PP plans.