sparkfun/SparkFun_RTK_Everywhere_Firmware #247: Corrections priority config on the web menu: values should be 0 & positive integers, don't allow a negative value (-1, etc).

doug-foster, 2024-04-23:
### Subject of the issue
Corrections priority config on the web menu: IMO values should be 0 & positive integers, don't allow a negative value (-1, etc).

### Your workbench
- EVK
- https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/tree/release_candidate/Firmware/RTK_Everywhere

### Steps to reproduce
[image]

### Expected behavior
Limit options to 0 or positive integers

### Actual behavior
Negative integers are allowed - could confuse user as to what that means.

### Options
Perhaps a drag & drop ordering would be better?
https://www.w3schools.com/html/html5_draganddrop.asp

nseidle, 2024-04-23:
This was completely reworked to use buttons (thanks @PaulZC!). Should be fixed now.

doug-foster, 2024-04-24:
Will test on EVK with a local compile later today & comment. Thanks @PaulZC!