sparkfun/SparkFun_RTK_Everywhere_Firmware #249: RTC clock synchronisation - interrupt vs. rtcUpdate

PaulZC, 2024-04-23:
It looks like we are currently synchronising the ESP32 RTC to GNSS in two places:

```rtcUpdate``` - [here](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/ab0bb01025da6b32e25b595ce189ebfa9837cc8c/Firmware/RTK_Everywhere/RTK_Everywhere.ino#L1377) if ```(millis() - lastRTCAttempt > syncRTCInterval)```

```tpISR``` - [here](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/ab0bb01025da6b32e25b595ce189ebfa9837cc8c/Firmware/RTK_Everywhere/Begin.ino#L1394) via interrupt, on the rising edge of PPS if ```(millisNow - lastRTCSync > syncRTCInterval)```

Let's make sure ```tpISR``` wins - on platforms that support it - for the most accurate clock synchronisation...

I think we have some redundancy and probable conflict between ```lastRTCAttempt``` and ```lastRTCSync```. Let's resolve that...

PaulZC, 2024-07-04:
```rtcUpdate``` will only set the RTC once. It can't resync it once ```online.rtc``` is true.

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/d95eefe61b78b07b6af05b492d1c1fb86aa5e43a/Firmware/RTK_Everywhere/RTK_Everywhere.ino#L1412

```lastRTCAttempt``` and ```lastRTCSync``` are doing different things. ```lastRTCAttempt``` restricts how often ```rtcUpdate``` can validate the time and attempt to set the RTC. ```lastRTCSync``` controls how often ```tpISR``` can resync the RTC. Both are needed.

```syncRTCInterval``` initially limits ```rtcUpdate``` to validating the time to once per second. It later limits ```tpISR``` to resyncing the RTC once per minute.

```rtcSyncd``` indicates that the RTC has been set accurately by ```tpISR```. It controls if the system state enters ```STATE_NTPSERVER_SYNC``` and starts responding to NTP requests.

```menuSystem``` can set ```online.rtc``` back to false, but only when the timeZone is changed. That's fine. The user can't expect NTP time to be accurate after changing the time zone. I'm adding code to change the state back to ```STATE_NTPSERVER_NO_SYNC``` if ```rtcSyncd``` becomes false - i.e. because the user has changed the time zone.

PaulZC, 2024-07-07:
Resolved by #404

nseidle, 2024-07-08:
This is in release [v1.3](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.3).

closed 2024-07-08 by nseidle