sparkfun/SparkFun_RTK_Everywhere_Firmware #255: EVK ZED NMEA messages cause uninitialized um980UnicoreHandler to be called

PaulZC, 2024-04-25:
EVK v1.1 Hardware

ZED is configured to output the logging messages ( 5xNMEA + RAWX + SFRBX )

Firmware is crashing [here](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/e45ae0b3229dbc52193ec679bac25ccfde6abf78/Firmware/RTK_Everywhere/Tasks.ino#L461) because the NMEA causes the uninitialized ```um980UnicoreHandler``` to be called

I think the fix is probably as simple as adding a ```if (gnssPlatform == PLATFORM_UM980)``` in there?

nseidle, 2024-04-26:
Sorry, my fault! Yes, that'll do it. A gnssPlatform check or you could slide that section of code down under the [present.gnss_um980](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/e45ae0b3229dbc52193ec679bac25ccfde6abf78/Firmware/RTK_Everywhere/Tasks.ino#L474C9-L474C27) check.

nseidle, 2024-04-26:
Looks like gnss_um980 is removed in the RC so a platform check is a reasonable middle ground.

PaulZC, 2024-04-26:
This is resolved in the release_candidate branch. It is not yet in an RC .bin.

nseidle, 2024-05-01:
This is in [v1.0](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.0).

closed 2024-05-01 by nseidle