sparkfun/SparkFun_RTK_Everywhere_Firmware #263: NTRIP Client setting modified upon entering WiFi config

nseidle, 2024-04-29:
NTRIP Client is disabled no matter what, when entering WiFi config.

nseidle, 2024-04-29:
Fixed in [this commit](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commit/1dc5ff7aaab439482281fbf652da0063dbb94e1c).

PaulZC, 2024-04-29:
This is definitely better than it was, but has - I think - created a new issue:

When I click Save Configuration, it gets stuck on Saving...

[image]

I think because the websocket gets closed before the "receipt confirmation" is actually sent

[image]

I see "Sending receipt confirmation", but I never see it arrive in the console log. With [this line](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/4142eecb6b979a186211ff5940fee6cc4600a9c6/Firmware/RTK_Everywhere/AP-Config/src/main.js#L83) uncommented, I should see the "confirmDataReceipt,1," in the console log, but I don't...

PaulZC, 2024-04-30:
But I do see "All Saved" on my iPhone through the captive portal.... Curious!

PaulZC, 2024-04-30:
Update: Tuesday April 30th 9:40 Mountain Time
Code: release_candidate - with the #265 merged and form.h updated
Hardware: Torch

Default code settings - s,r,y - no WiFi credentials

WiFi config on Windows desktop, Chrome, _not_ captive portal

[image]

Turn on Debug WiFi Config - s,n,4

[image]

Provide sparkfun-iot credentials, enable PointPerfect - through the web config. Restart.

Torch connects to WiFi, receives PointPerfect corrections, gets RTK Float.

Return to WiFi config and save settings:

[image]

We're all good...

I'm declaring my earlier issue to be caused by connecting to the wrong Torch - which was possibly running dubious firmware....

nseidle, 2024-05-01:
This is in [v1.0](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.0).

closed 2024-05-01 by nseidle