sparkfun/SparkFun_RTK_Everywhere_Firmware #267: Correction priority not being respected

nseidle, 2024-04-30:
PointPerfect correction data is being blocked even though it is above NTRIP data on the correction priorities list.

Starting from a factory reset torch:

* Enter 1 WiFi network
* Enable NTRIP Client to RTK2Go 
* Provision the device to use PointPerfect
* Confirm the correction priority list is default (shown below, IP is above TCP)
* Enable correction debugging: s, h, 10
* Enable Debug NTRIP client --> caster GGA messages: s, n, 22

Allow NTRIP and PointPerfect data to flow until the PPL is active. 

        Pushing 512 bytes from /pp/ip/us topic to PPL for UM980
        Pushing 374 bytes from /pp/ip/us topic to PPL for UM980
        Received 573 RTCM bytes from PPL. NOT pushed to the GNSS due to priority.
        NTRIP Client received 1408 RTCM bytes, pushed to ZED
        Rover Accuracy (m): 0.008, SIV: 34 GNSS State: RTK Fix
        NTRIP Client received 1408 RTCM bytes, pushed to ZED
        Batt (96%): Voltage: 8.17V Charging: 0.94%/hr

Below is the priority list:

        A / a) Bluetooth
        B / b) IP (PointPerfect/MQTT)
        C / c) TCP (NTRIP)
        D / d) L-Band
        E / e) External Radio
        F / f) LoRa Radio
        G / g) ESP-Now

PaulZC, 2024-04-30:
The priority is being respected, but the output of the PPL is being labelled as L-Band, not IP...

The fix will be to add an extra flag which defines which source the PPL is being fed from, and then use that to classify the priority...

PaulZC, 2024-04-30:
Resolved #270

PaulZC, 2024-05-01:
Note to self:

The fix works - for Torch - because the PPL can only be fed by IP corrections.

But we will need a better solution for a future product which is both L-Band and IP capable. There we will need to decide (prioritise) which source is allowed to feed the PPL. The PPL will (almost certainly) get horribly confused if it is fed with L-Band and IP SPARTN simultaneously...

commit 82fb96d7 (2024-04-30): Resolve #267
commit ed8602c5 (2024-05-01): Merge pull request #270 from sparkfun/Resolve_#267

Resolve #267
closed 2024-05-01 by nseidle