sparkfun/SparkFun_RTK_Everywhere_Firmware #280: Torch: Need passthrough to UM980 for config and firmware update

nseidle, 2024-05-09:
There may be new features and firmware available for the UM980. We need the ability to switch all serial communication to the UM980. 

[image]

Above, the Torch muxes. For now, let's do a hard cut over of SW1/2 (and support STM32 LoRa IC as well). I don't see a good way to timeout and give control back to the ESP32, so for now, require a system reset before the ESP32 will release the muxes for normal (default) communication.

nseidle, 2024-05-09:
[image]

Ahh that won't work. SW1 is wired the other way. We'll have to receive serial from USB into ESP-UART0 and send it out ESP-UART1 with SW2 set to B0.

nseidle, 2024-05-11:
Pass through is working well. Docs for updating UM980 firmware is [here](https://docs.sparkfun.com/SparkFun_RTK_Everywhere_Firmware/firmware_update/#updating-um980-firmware).

SSMCD, 2024-05-11:
That was fast! Nice work, thank you!

nseidle, 2024-05-13:
This is in [v1.1](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.1).

closed 2024-05-13 by nseidle