sparkfun/SparkFun_RTK_Everywhere_Firmware #284: Add new features to WiFi Config

nseidle, 2024-05-14:
New features that need to be rendered on WiFi Config page:

* HAS/E6 enable/disable under GNSS
* Add Ports menu for Torch.
* Add USB GNSS output enable/disable under Ports

nseidle, 2024-05-14:
Remove Debug Zone / Reset counter

nseidle, 2024-05-14:
Update Torch default ARP and tooltip.

nseidle, 2024-05-14:
Add rebootSeconds feature to system menu

nseidle, 2024-05-14:
* Add Tilt menu.
* Enable/disable tilt
* Pole length

nseidle, 2024-05-14:
Add auto-firmware update to system menu

nseidle, 2024-05-14:
Add Measurement scale to system menu

nseidle, 2024-05-14:
Add Beeper control to system menu

PaulZC, 2024-05-16:
```Enable Tilt Compensation``` and ```Pole Length (m)``` are already in GNSS Config. Let me know if they should be somewhere else.

nseidle, 2024-05-16:
Ah! Good to know. We should break it out to its own grouping.

nseidle, 2024-07-08:
This is in release [v1.3](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.3).

closed 2024-07-08 by nseidle