sparkfun/SparkFun_RTK_Everywhere_Firmware #289: Checklist / minimum requirements for Ethernet on EVK

PaulZC, 2024-05-16:
This issue is to capture the minimum requirements for Ethernet on the EVK, and provide a "Done" & "Tested" checklist.

## Background:
At the time of writing, EVK W5500 Ethernet functionality is provided by the standard Arduino Ethernet Library 2.0.2. Configure-Via-Ethernet uses the [SparkFun_WebServer_ESP32_W5500](https://github.com/sparkfun/SparkFun_WebServer_ESP32_W5500); the hardware needs to be restarted in a dedicated Configure-Via-Ethernet mode as the WebServer does not play nicely with Arduino SPI or Interrupts. Meanwhile, Espressif have been busy and have added dedicated [Arduino support for the W5500 on SPI](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_W5500_Arduino_SPI/ETH_W5500_Arduino_SPI.ino) - but only in arduino-esp32 >= v3.0.0. We need to decide if we should upgrade the core to v3.0.0 so we can take advantage of the much-improved W5500 support (interrupt support; larger packets; seamless web serving, making the WebServer and the dedicated Configure-Via-Ethernet mode obsolete). This checklist will inform that decision.

# Checklist:

Last update: May 16th

| **Feature** | **Required** | **Done** | **Tested** |
| --- | --- | --- | --- |
| DHCP | ✔ |  |  |
| Fixed IP | ✔ |  |  |
| Configure-Via-Ethernet | ✔ |  |  |
| NTP Server | ✔ |  |  |
| PointPerfect Get Keys | ✔ |  |  |
| PointPerfect Corrections via MQTT | ✔ |  |  |
| NTRIP Client | ✔ |  |  |
| NTRIP Server | ✔ |  |  |
| TCP Client | ✔ |  |  |
| TCP Server | ✔ |  |  |
| UDP Server | ✔ |  |  |

✔❌

PaulZC, 2024-05-17:
I think I've answered my question about whether we need to upgrade to v3.0.0...

[pointperfectUpdateKeys](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/3a148439ea347771feed1429429c2c1a6beb0ded/Firmware/RTK_Everywhere/menuPP.ino#L794-L823) uses ```WiFiClientSecure``` to manage the MQTT certificates etc.. For Ethernet, if we upgrade to v3.0.0 we can continue to use ```WiFiClientSecure``` - the new arduino-esp32 Ethernet (ETH.h) inherits all of the WiFi functionality, including ClientSecure. If we don't upgrade, we will need to bolt a secure MQTT layer on top of Arduino Ethernet and we'll end up in a big old mess.

For Cellular, the LARA-R6 looks after the secure connection. We can enhance the network layer so it: uses ```WiFiClientSecure``` for both WiFi and Ethernet; adds a new layer for the LARA which calls ```setSecurityManager```, ```configSecurityProfileString```,  ```setMQTTserver```, ```setMQTTsecure```, etc..

Additionally, the KEYS states are [hardwired to WiFi](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/3a148439ea347771feed1429429c2c1a6beb0ded/Firmware/RTK_Everywhere/States.ino#L512-L541). We need to change this so they use the network layer, making it possible to update the keys over Ethernet and Cellular too...

nseidle, 2024-06-26:
Since we've migrated to v3, I thought an updated list would be good.

# Checklist:

Last update: June 26th

| **Feature** | **Required** | **Done** | **Tested** |
| --- | --- | --- | --- |
| DHCP | ✔ | ✔ | ✔ |
| Fixed IP | ✔ |  |  |
| Configure-Via-Ethernet | ✔ | ✔ | ✔ |
| NTP Server | ✔ |  |  |
| PointPerfect Get Keys | ✔ |  |  |
| PointPerfect Corrections via MQTT | ✔ |  |  |
| NTRIP Client | ✔ | ✔ | ✔ |
| NTRIP Server | ✔ | ✔ | ✔ |
| TCP Client | ✔ |  |  |
| TCP Server | ✔ |  |  |
| UDP Server | ✔ |  |  |

✔❌

PaulZC, 2024-07-05:
NTP Server - done and tested

# Checklist:

Last update: July 5th

| **Feature** | **Required** | **Done** | **Tested** |
| --- | --- | --- | --- |
| DHCP | ✔ | ✔ | ✔ |
| Fixed IP | ✔ | ✔ | ✔ |
| Configure-Via-Ethernet | ✔ | ✔ | ✔ |
| NTP Server | ✔ | ✔ | ✔ |
| PointPerfect Get Keys | ✔ | ✔ | ✔ |
| PointPerfect Corrections via MQTT | ✔ | ✔ | ✔ |
| NTRIP Client | ✔ | ✔ | ✔ |
| NTRIP Server | ✔ | ✔ | ✔ |
| TCP Client | ✔ |  |  |
| TCP Server | ✔ |  |  |
| UDP Server | ✔ |  |  |

✔❌

PaulZC, 2024-07-06:
TCP Server and Client - done and tested (with #408)

**TO DO:** retest Torch on WiFi - just to make sure we haven't broken anything!

# Checklist:

Last update: July 6th

| **Feature** | **Required** | **Done** | **Tested** |
| --- | --- | --- | --- |
| DHCP | ✔ | ✔ | ✔ |
| Fixed IP | ✔ | ✔ | ✔ |
| Configure-Via-Ethernet | ✔ | ✔ | ✔ |
| NTP Server | ✔ | ✔ | ✔ |
| PointPerfect Get Keys | ✔ | ✔ | ✔ |
| PointPerfect Corrections via MQTT | ✔ | ✔ | ✔ |
| NTRIP Client | ✔ | ✔ | ✔ |
| NTRIP Server | ✔ | ✔ | ✔ |
| TCP Client | ✔ | ✔ | ✔ |
| TCP Server | ✔ | ✔ | ✔ |
| UDP Server | ✔ | ✔ | ✔ |

✔❌

PaulZC, 2024-07-06:
# WiFi Testing:

Testing the Combined branch on EVK:

Factory reset
Provide credentials for one WiFi network
Verify the following:

Last update: July 6th

| **Feature** | **Required** | **Tested** |
| --- | --- | --- | --- |
| Configure-Via-WiFi AP | ✔ |  |
| Configure-Via-WiFi STN | ✔ |  |
| PointPerfect Get Keys | ✔ |  |
| PointPerfect Corrections via MQTT | ✔ |  |
| NTRIP Client | ✔ |  |
| NTRIP Server | ✔ |  |
| TCP Client | ✔ |  |
| TCP Server | ✔ |  |
| UDP Server | ✔ |  |

✔❌

nseidle, 2024-07-08:
This is in release [v1.3](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.3).

closed 2024-07-08 by nseidle