sparkfun/SparkFun_RTK_Everywhere_Firmware #320: Torch "...device not whitelisted..." message provides a URL to a webpage for L-Band Facet provisioning

tonycanike, 2024-05-22:
### Subject of the issue

Here's the error message with the URL that goes to a provisioning page for L-Band Facets.

```
Connecting to: https://api.thingstream.io/ztp/pointperfect/credentials
This device is not whitelisted. Please contact support@sparkfun.com or goto https://www.sparkfun.com/pointperfect to get your subscription activated. Please reference device ID: 64B7xxxxxxx

```

The URL in the error message goes to a webpage that's all about L-Band Facet provisioning.

Oh, and also **there is no form to fill out - just a big blank whitespace.**

[image]

### Your workbench

```
LittleFS Started
PSRAM Size (bytes): 2095087
I2C Devices:
  0x08 - HUSB238 Power Delivery Sink Controller
  0x0B - BQ40Z50 Battery Pack Manager / Fuel gauge
  0x5C - MP27692A Power Management / Charger
  0x60 - ATECC608A Cryptographic Coprocessor
SparkFun RTK Torch v1.1
GNSS UM980 online
UM980 firmware: 11833
Profile 'Profile1' loaded
Fuel gauge configuration complete
Batt (43%): Voltage: 7.37V Discharging: 7.60%/hr
Charger configuration complete
UM980 configuration updated
No GNSS date/time available for system RTC.
Bluetooth SPP and BLE broadcasting as: Torch Rover-0026
STATE_ROVER_NOT_STARTED --> STATE_ROVER_NO_FIX
Batt (43%): Voltage: 7.37V Charging: 7.41%/hr
No GNSS date/time available for system RTC.
```

nseidle, 2024-05-22:
We'll get that fixed. Correct link: https://www.sparkfun.com/rtk_torch_registration

What browser are you using? It's rendering for me:

[image]

tonycanike, 2024-05-22:
I was using Chrome on a W10 machine.  I've since closed and opened Chrome and the form fields appeared (sparkfun.com/pointperfect).

nseidle, 2024-05-23:
Working for me but I'll bring it up with our web folks:

[image]

nseidle, 2024-05-23:
Thanks for reporting! The landing page is now linked to platform. Should enter RC shortly.

commit 023a4bd4 (2024-05-23): Change PointPerfect landing page based on platform

Fix #320
commit 20ae5c6e (2024-05-28): Change PointPerfect landing page based on platform

Fix #320
closed 2024-05-28 by nseidle