sparkfun/SparkFun_RTK_Everywhere_Firmware #334: Tilt elevation wrong but right with tilt off.

APOS80, 2024-06-07:
### Subject of the issue
Without tilt engaged it i get the correct elevation but with tilt on i get about 24 meters above the correct hight.

### Your workbench
* What version of RTK firmware are you running? This can be found when the serial menu is opened (also in the settings.txt file, and in the serial output at power up).
1.2
* What radios are you using: Bluetooth, WiFi, and/or ESP-Now? What app are you using to connect over Bluetooth? Are you transmitting NTRIP back to the device?
Bluetooth
* Are there any additional details that may help us help you?
Polehight set to 2 meters in tilt-settings as the pole is 2 meters.

### Steps to reproduce
Tell us how to reproduce this issue. Please post any log files from serial output that may have been generated.
Set tilt-polehight to 2 meters, measure with and without tilt.

### Expected behavior
About the same elevation reading is expected.

### Actual behavior
When tilt is engaged the elevation is wrong.

APOS80, 2024-06-07:
Its with a RTK Torch

APOS80, 2024-06-08:
New test this morning!

Measured an averaged point without tilt and did stakeout of that point. It’s working millimetres without tilt, with tilt enabled I got -21,95 elevation. When turning of tilt it got back to millimetres, turning tilt on again the elevation error appeared again.

Something isn’t right.

Reading the code I can’t see Anny error, maybe it’s something wrong with the binary file.

APOS80, 2024-06-08:
Tried to downgrade to v1.1 but it didn’t go through, can’t downgrade it seems.

APOS80, 2024-06-08:
Line 349 in “tilt.ino” looks wrong but I’m not well versed in C++.

APOS80, 2024-06-09:
Looked closer on your YouTube tilt demonstration, it showed the same problem with elevation.

How did you miss this??

[image]
[image]

swells-sfefae, 2024-06-10:
While we appreciate the bug report, we don't enjoy the wheedling. "How did you miss this??" is an unproductive question that you intended to be demeaning; not a good look for you. If you'd like support, and to continue participating, please try to stay focused on collaboration rather than disparagement. We all make mistakes; let's move on from this one.

I'm working on reproducing and documenting the issue now. Please submit a PR if you can create a solution before we do! 21.95 is an interesting number to be off by; almost exactly 7π. You do have your pole height set in the Torch, under the Tilt Compensation menu, correct?

APOS80, 2024-06-10:
Sorry, I was just amazed it passed quality control that’s all.

I’ve tried a 2 meter pole with a 2 meter tilt setting and the same with 1.8 meter pole. In your YouTube video you get a 17.8 meter difference in elevation with a 1.8 meter pole.

I’ve been looking at your code a bit but I don’t really get the calculation to adjust altitude. Can you share the calculation with numbers, maybe the match is wrong.

And I’ve been wondering why I haven’t seen any vector-math because this seems to be a vector problem.

APOS80, 2024-06-10:
Sorry for that!I’ve tried using my 2 meter pole with a 2 meter setting and an 1.8 meter pole with the appropriate setting but it’s the same problem.

APOS80, 2024-06-10:
21.95/3.14=6.99 nearly 7 PI
17.21/3.14=5.48 nearly 5.5 PI

Don’t know if that helps.

swells-sfefae, 2024-06-10:
No worries, I am pushing to keep our bug-tracker from turning into a Reddit thread; it is a danger in Open Source that requires active pushback to deal with; I appreciate your understanding! 

I know tilt compensation was working correctly at some point (testing for height accuracy was our only testing metric); it seems we regressed since then without noticing.

Is the 2-meter setting you're talking about being entered into SW Maps (or some other collector software), or inside of the Torch menu settings? It's a bit confusing, because there are multiple places you can put that value.

APOS80, 2024-06-10:
By the way!

Is it possible to load older binaries, I tried downloading on the phone and uploading to the Torch but it still said “1.2” in settings.

swells-sfefae, 2024-06-10:
Got it, I was looking for clarity on where you entered the settings; sometimes people _think_ they've entered it into the device, because they don't know that the collector isn't just a control app for the Torch (on products from some other companies, the collector app **does** change settings on the device, and it is perfectly reasonable to get confused about this). 

In tilt.ino, the Orthometric height changes 1581.450 to 1580.987, and the Geoid Separation is at -21.361 for both messages. This makes sense to me; using a heavily decimated geoid model like the ZED F9P does, Geoid Separation won't change unless you move a large distance in Lat/Lon, while Orthometric will change if you move the antenna in any direction, with the largest error occurring in Z. What doesn't make sense to me is that when I run tilt comp now, I get the following:
Original GNGGA: 
$GNGGA,201946.00,4005.42738349,N,10511.11755382,W,4,22,0.7,1568.0557,M,-17.1107,M,6.0,0\*6C
Compensated GNGGA: 
$GNGGA,201946.00,4005.4274067,N,10511.1175992,W,4,22,0.7,1549.023,M,-17.1107,M,6.0,0\*57

Compensated Orthometric is suddenly -19.0327 meters of Original Orthometric, when they should be with a few centimeters of each other with error. -19.0327 is interesting to me because it is similar to, but different from, your number; this tells me it isn't a hardcoded value issue, it's a bad computation. I will continue to chase this down.

Uploading older binaries should be possible (pretty certain I downgraded from an RC version a few weeks back). I recommend using our [RTK Firmware Uploader](https://github.com/sparkfun/SparkFun_RTK_Firmware_Uploader) tool rather than the web-UI for doing anything out of the ordinary, as it executes a standard and very stable ESP32 flash operation.

APOS80, 2024-06-11:
God morning!

I’m sort of happy you also found the error and it’s not just me.

Yes it could be that the altitude it “corrected” with the wrong value.

nseidle, 2024-06-11:
Good evening! :)

As @swells-sfefae says, thanks for reporting.

I believe we are configuring the IMU with the correct pole length (see [IM19 datasheet](https://raw.githubusercontent.com/sparkfun/SparkFun_RTK_Torch/main/Datasheets/IM19_IMU_Tilt_Sensor_v1_1_10.pdf) `AT+CLUB_VECTOR` and the [CLUB_VECTOR setting](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/main/Firmware/RTK_Everywhere/Tilt.ino#L353)). Is this a geoid separation issue? In the latest version we are enabling WGS84 datum if [Galileo HAS](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/main/Firmware/RTK_Everywhere/UM980.ino#L179) is explicitly enabled but the video pre-dates the E6 feature incorporation. I'm still trying to determine what the default geoid model is for the UM980. I don't yet have a good answer why the elevation is changing. Please bear with us.

APOS80, 2024-06-11:
Don’t you mean ellipsoid?

nseidle, 2024-06-12:
Got it. The IMU was reporting ellipsoidal height. We need to remove the undulation from that number before we splice it into the NMEA sentences.

[image]

Above, alt of 1561.822m with no pole length and no tilt.

[image]

Above, tilt activated. Alt drops to 1559.883.

1561.822-1559.883 = 1.939m. The 'club length' during tilt activation is 1.916m so if we subtract that (this is basically the 1.8m pole length + the 96.45mm ARP) we see an altitude change of 23mm. This is reasonable as the pole may have indeed been tilted when I took the screen shots.

nseidle, 2024-06-12:
The RC is building now. Please give today's release candidate a try and let us know.

APOS80, 2024-06-12:
It works now but I would like to change one thing.

It would help if the app got the same reading from the Torch as if it was without tilt, so you don’t have to change the pole length in the app as you engage tilt. I think it would just be a matter of adding the “club length” to the elevation before sending it.

nseidle, 2024-06-12:
Excellent! Glad to hear it's working. I'm going to close this issue.

> It would help if the app got the same reading from the Torch as if it was without tilt, so you don’t have to change the pole length in the app as you engage tilt. I think it would just be a matter of adding the “club length” to the elevation before sending it.

I understand the request but it is unfortunately not that simple. Imagine if the pole is at a 45 degree angle: adding the pole length + ARP back to the reported elevation will return erroneous results.

For anyone else reading, SurPad has been added to the [GIS docs](https://docs.sparkfun.com/SparkFun_RTK_Everywhere_Firmware/gis_software_android/#surpad).

closed 2024-06-12 by nseidle

APOS80, 2024-06-13:
I mean adjusting the elevation to resemble a levelled pole after the tilt adjustment has been completed and you have the coordinates of the pole tip.