sparkfun/SparkFun_RTK_Everywhere_Firmware #335: (release_candidate) NetworkSecureWiFiClient destructor causes crash

PaulZC, 2024-06-12:
This line is causing a crash:

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/45aeb81a4e27fb4e7ae52d45f1a485853563c03d/Firmware/RTK_Everywhere/RTKNetworkClient.h#L302

It calls:

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/45aeb81a4e27fb4e7ae52d45f1a485853563c03d/Firmware/RTK_Everywhere/RTKNetworkClient.h#L43-L52

which looks OK, but something is going bang...

Caused by ```mqttClientStop```, which calls ```delete mqttSecureClient``` here:

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/45aeb81a4e27fb4e7ae52d45f1a485853563c03d/Firmware/RTK_Everywhere/MQTT_Client.ino#L446

Crash dump and decoder output follow:

```
Network layer stopping MQTT client
MQTT Client Connected Uptime: 0 00:00:00.000 (Reconnects: 0)
Freeing mqttClient
FreeHeap: 65152 / HeapLowestPoint: 6472 / LargestBlock: 4063220 / Used PSRAM: 87548
Freeing mqttSecureClient
Guru Meditation Error: Core  1 panic'ed (Double exception).

Core  1 register dump:
PC      : 0x4009b53a  PS      : 0x00040036  A0      : 0x800e201e  A1      : 0x3ffd05b0
A2      : 0x3f813660  A3      : 0x00000000  A4      : 0x3ffbd734  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x40080080  A9      : 0x3ffd0690
A10     : 0x00060236  A11     : 0x00040026  A12     : 0x3ffc66c0  A13     : 0x3ffc57e0
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000002
EXCVADDR: 0xffffffe0  LBEG    : 0x40095231  LEND    : 0x40095241  LCOUNT  : 0xfffffffa

Backtrace: 0x4009b537:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 |<-CONTINUES

ELF file SHA256: fd47e6c66facd55d
```

```
PC: 0x4009b53a
EXCVADDR: 0xffffffe0

Decoding stack results
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
0x400e201b: NetworkSecureWiFiClient::~NetworkSecureWiFiClient() at C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTKNetworkClient.h line 302
```

PaulZC, 2024-06-12:
Then, separately, I'm seeing:

```
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 460 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 477 bytes from /pp/Lb/eu topic to GNSS
[740463][E][ssl_client.cpp:36] _handle_error(): [data_to_read():401]: (-80) UNKNOWN ERROR CODE (0050)
[740472][E][NetworkClientSecure.cpp:308] available(): Closing connection on failed available check
[740482][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
SIV: 30, HPA (m): 0.185, Lat: 54.86654138, Lon: -1.42844321, Altitude (m): 156.0
Rover Accuracy (m): 0.185, SIV: 30 GNSS State: 3D Fix
STATE_ROVER_RTK_FIX --> STATE_ROVER_FIX, 2024-06-12 09:35:44.604
Log file size: 27093314 - Generation rate: 407.6kB/s
WiFi 'VM4924742' IP address: 192.168.0.90, RSSI: -56
Network State: *NETWORK_STATE_IN_USE
[740564][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
*MQTT_CLIENT_SERVICES_CONNECTED
[740590][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[740609][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[740627][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[740645][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[740663][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[740681][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[740699][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[740719][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[740737][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
```

followed by:

```
[770356][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[770376][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[770394][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[770413][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[770431][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[770452][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[770471][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
MQTT client data timeout. Disconnecting...
MQTT Client Connected Uptime: 0 00:00:00.000 (Reconnects: 0)
[770479][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
Freeing mqttClient
FreeHeap: 65680 / HeapLowestPoint: 6328 / LargestBlock: 4063220 / Used PSRAM: 85532
Freeing mqttSecureClient
Guru Meditation Error: Core  1 panic'ed (Double exception).

Core  1 register dump:
PC      : 0x4009b53a  PS      : 0x00040e36  A0      : 0x800e201e  A1      : 0x3ffd05b0
A2      : 0x3f813660  A3      : 0x00000018  A4      : 0x3ffc5f3c  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x40080080  A9      : 0x3ffd0690
A10     : 0x00060036  A11     : 0x00040026  A12     : 0x3ffc5f3c  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000002
EXCVADDR: 0xffffffe0  LBEG    : 0x40095231  LEND    : 0x40095241  LCOUNT  : 0xfffffffa

Backtrace: 0x4009b537:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 0x4009b537:0x3ffd0650 0x4009b537:0x3ffd0670 0x4009b537:0x3ffd0690 0x4008007d:0x3ffd05b0 0x400e201b:0x3ffd05d0 0x4009b537:0x3ffd05f0 0x4009b537:0x3ffd0610 0x4009b537:0x3ffd0630 |<-CONTINUES

ELF file SHA256: fd47e6c66facd55d
```

Again, it looks like this is caused by ```mqttClientStop``` which calls ```delete mqttSecureClient```

PaulZC, 2024-06-12:
Humm... Is it recursive somehow? Is the destructor of the inherited class somehow calling itself?

It could be caused by the renaming in this commit: https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commit/c58299259f4f6cc024777a1d4b80319c533ff81d

Maybe I one of the ```RTKNetworkClient``` should actually be ```NetworkClient```?

PaulZC, 2024-06-27:
Fixed by https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commit/9d56cd47aaa1a03877869733323b318affce1a54 but not fully resolved. It would be good to understand why the delete causes a crash. Closing... I'll open a new issue if needed.

PaulZC, 2024-06-27:
Actually, no. Let's keep this open so we don't forget to investigate if this is truly solved, and/or if the ```delete``` needs to be reinstated.

nseidle, 2024-07-08:
This is in release [v1.3](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.3).

closed 2024-07-08 by nseidle