sparkfun/SparkFun_RTK_Everywhere_Firmware #343: Pole length whit tilt

APOS80, 2024-06-14:
### Subject of the issue
When using tilt and a 0m pole length in the app(SW or Surpad) it will be a 115mm error.
The only way to be tit right it to set -0,115 pole length in the app.

It would be preferred if the NMEA sent the same elevation it would if it was a levelled pole and no tilt.

### Your workbench
* What version of RTK firmware are you running?
1.3beta
* What radios are you using: 
Bluetooth
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post any log files from serial output that may have been generated.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

APOS80, 2024-06-14:
If the code just adds the club length to the computed elevation before sending the NMEA message.

APOS80, 2024-06-16:
I tried to make a pull request but it wont do it.

nseidle, 2024-06-17:
Thank you for attempting to create a PR! That's immensely helpful. 

I'm sorry I can't help troubleshoot git. The repo is public and you should be able to submit PRs against it.

I've been thinking about this feature and am open to the idea but we need to take into account the HAE and other settings associated with base setups. Additionally, we need to think carefully about how this feature is deployed: if it is enabled by default, users who upgrade will see their altitudes drop by the default length of the pole. This could be very disruptive.

APOS80, 2024-06-17:
I’ve made a fork and pushed the change there in tilt.ino:

```
  // Remove the undulation from the IMU's altitude and adjust hight to normal
    float orthometricHeight = tiltSensor->getNaviAltitude() - undulation + settings.tiltPoleLength + present.antennaReferencePoint_mm / 1000.0;
```

It’s not that complicated it’s seems.

I would say that changing pole hight to negative 0.115 meters is disruptive and very strange.

With other rover with tilt as for example the Trimble R12i you do not need to change pole hight because you go to tilt.

APOS80, 2024-06-17:
It’s also strange that it says “orthometricHeight” as it shouldn’t be a geoid hight.

APOS80, 2024-06-20:
Hello?

nseidle, 2024-06-20:
Hi, I look forward to reviewing your PR. I reiterate the fact that this feature will need the ability to be disabled by default and enablable by a user.

APOS80, 2024-06-20:
I’m having trouble with getting the PR to be sent.

APOS80, 2024-06-21:
I do consider it a fix not a “feature” as it would make it work like the other tilt enabled rovers on the market.

nseidle, 2024-06-26:
* Menu added to allow tip altitude reporting
* Removed poleLength in place of antennaHeight_mm setting
* Added NMEA injection for three scenarios (tilt enabled+tip altitude, tilt enabled+device altitude, tilt disabled+tip altitude)

[image]

Please give the RC a try. You can follow the commits [here](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commits/AddTipAltitudeSetting/).

APOS80, 2024-06-27:
Can’t find the bin firmware

APOS80, 2024-06-27:
Update!

Tested it today.

Tilt doesn’t work at all.

nseidle, 2024-06-27:
> Tilt doesn’t work at all.

This is a bit confusing. To clarify, are you able to enter tilt compensation mode by rocking the unit back and forth on a 1.8m pole? What specifically fails?

Using [RC June 27](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/blob/main/RTK_Everywhere_Firmware_RC-Jun_27_2024.bin), everything is working well for me. I wonder if you had the pole length setting (now called Antenna Height) set to 0.0m in the firmware? When I did that, tilt mode could be entered, but any tilt translated to large changes in the position (obviously).

[image]

Above, we see the position and altitude remain nearly identical before and after tilt is activated. There's 12mm of change but my pole is not in a calibrated state.

nseidle, 2024-06-27:
I also recommend a [factory reset](https://docs.sparkfun.com/SparkFun_RTK_Everywhere_Firmware/menu_system/#factory-reset) of the device after a firmware update just to be sure the state the device is in.

APOS80, 2024-06-28:
I think the problem is that the settings doesn’t seem to be saved and when I go into settings from the phone after double press on the torch some settings are missing and tilt pole is 0 even though I’ve set it.

The rover enters tilt when rocked and moved around. 

I have not tried June 27 yet, it wasn’t there yesterday.

APOS80, 2024-06-28:
Beta Jun 27

Settings made for pole hight  and arp via serial doesn't show when going to settings via phone(WiFi).
It’s the same for firmware version and checkboxes.

nseidle, 2024-06-28:
Thanks for testing. I implement features in the serial menus first, and Web Config comes next. My apologies, I should have told you that. I will try to get the Web Config fixed shortly.

I'll consider this issue fixed. Thanks for opening a new issue for Surpad. Separate issues help us keep track more efficiently.

APOS80, 2024-06-28:
Thank you for making my work easier!

nseidle, 2024-07-08:
This is in release [v1.3](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.3).

closed 2024-07-08 by nseidle

APOS80, 2024-07-09:
Thank you, good work!