sparkfun/SparkFun_RTK_Everywhere_Firmware #353: (release_candidate) NetworkClientSecure memory allocation failures

PaulZC, 2024-06-19:
I'm seeing ```NetworkClientSecure``` memory allocation failures during auto and manual key requests on EVK with Ethernet and WiFi enabled... Something is gobbling up the heap. Note **HeapLowestPoint: 82704** before the WiFiMulti.run and NetworkClientSecure.connect, and **HeapLowestPoint: 2144** after... Maybe we're not freeing something correctly?

```
networkTypeUpdate, network->requestedNetwork: WiFi --> WiFi
Network starting WiFi
Starting WiFi
WIFI_STATE_CONNECTING
FreeHeap: 83900 / HeapLowestPoint: 82704 / LargestBlock: 4128756 / Used PSRAM: 63140
Network State: NETWORK_STATE_DELAY --> NETWORK_STATE_CONNECTING
ETH Unknown event: 9
Connecting WiFi... [ 52008][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: VM4924742
ETH Unknown event: 11
ETH Unknown event: 10
[ 62989][I][WiFiMulti.cpp:129] run(): [WIFI] scan done
[ 62996][I][WiFiMulti.cpp:134] run(): [WIFI] 14 networks found
[ 63001][D][WiFiMulti.cpp:228] run():  --->   0: [11][18:35:D1:9D:ED:09] VM4924742 (-56) (*) (visible)
[ 63011][D][WiFiMulti.cpp:233] run():         1: [6][58:CB:52:D8:5E:1E] VM6184319 (-69) (*) (visible)
[ 63020][D][WiFiMulti.cpp:233] run():         2: [1][58:CB:52:D8:49:F7] VM6184319 (-72) (*) (visible)
[ 63029][D][WiFiMulti.cpp:233] run():         3: [6][40:0D:10:48:45:51] VM6184319 (-79) (*) (visible)
[ 63038][D][WiFiMulti.cpp:233] run():         4: [11][8C:83:94:6B:6D:FB] BT-XPFH75 (-83) (*) (visible)
[ 63047][D][WiFiMulti.cpp:233] run():         5: [6][C0:05:C2:46:C3:59] VM1245360 (-84) (*) (visible)
[ 63056][D][WiFiMulti.cpp:233] run():         6: [6][04:95:E6:62:FF:D9] NOVA (-84) (*) (visible)
[ 63065][D][WiFiMulti.cpp:233] run():         7: [11][7A:83:94:6B:6D:FC] EE WiFi (-86) ( ) (visible)
[ 63074][D][WiFiMulti.cpp:233] run():         8: [1][9C:31:C3:09:B0:42] SKYCEIY6 (-89) (*) (visible)
[ 63083][D][WiFiMulti.cpp:233] run():         9: [1][D4:35:1D:17:49:49] vodafone174949 (-89) (*) (visible)
[ 63092][D][WiFiMulti.cpp:233] run():         10: [1][08:B4:B1:66:EF:FD] VM6184319 (-92) (*) (visible)
[ 63102][D][WiFiMulti.cpp:233] run():         11: [1][08:B4:B1:66:ED:5E] VM6184319 (-92) (*) (visible)
[ 63111][D][WiFiMulti.cpp:233] run():         12: [11][18:35:D1:71:EE:D1] VM2041210 (-95) (*) (visible)
[ 63120][D][WiFiMulti.cpp:233] run():         13: [1][40:0D:10:E2:46:89] VM3885188 (-96) (*) (visible)
[ 63129][I][WiFiMulti.cpp:249] run(): [WIFI] Connecting BSSID: 18:35:D1:9D:ED:09 SSID: VM4924742 Channel: 11 (-56)
[ 63140][W][STA.cpp:533] disconnect(): STA already disconnected.
ETH Unknown event: 13
ETH Unknown event: 16
[ 63429][I][WiFiMulti.cpp:272] run(): [WIFI] Connecting done.
[ 63435][D][WiFiMulti.cpp:273] run(): [WIFI] SSID: VM4924742
[ 63441][D][WiFiMulti.cpp:274] run(): [WIFI] IP: 192.168.0.90
[ 63446][D][WiFiMulti.cpp:275] run(): [WIFI] MAC: 18:35:D1:9D:ED:09
[ 63453][D][WiFiMulti.cpp:276] run(): [WIFI] Channel: 11
[ 63458][D][WiFiMulti.cpp:334] resetFails(): [WIFI] Resetting failure flags

WIFI_STATE_CONNECTED
Network connected to WiFi
Network State: NETWORK_STATE_CONNECTING --> NETWORK_STATE_IN_USE
WiFi 'VM4924742' IP address: 192.168.0.90, RSSI: -55
Rover Accuracy (m): 0.187, SIV: 31 GNSS State: 3D Fix
Log file size: 175659 - Generation rate: 9.9kB/s
*WIFI_STATE_CONNECTED
[ 63589][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
[ 63734][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 52.212.195.223
[ 64605][E][ssl_client.cpp:36] _handle_error(): [ssl_starttls_handshake():313]: (-17168) RSA - The private key operation failed : BIGNUM - Memory allocation failed
[ 64620][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -17168
Failed to connect to MQTT broker pp.services.u-blox.com
[ 64631][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
Network stopping user MQTT Client on WiFi
Pushing 528 bytes of RXM-PMP data to GNSS
STATE_ROVER_FIX --> STATE_ROVER_RTK_FIX, 2024-06-19 13:51:46.885
Network shutting down WiFi, no users
networkStop called by C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere\Network.ino 1204
Network stopping WiFi
WIFI_STATE_OFF
E (76423) wifi:NAN WiFi stop
[ 64751][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
ETH Unknown event: 14
ETH Unknown event: 12
ETH Unknown event: 8
WiFi Stopped
FreeHeap: 82264 / HeapLowestPoint: 2144 / LargestBlock: 4063220 / Used PSRAM: 65508
Network layer stopping
Network State: NETWORK_STATE_IN_USE --> NETWORK_STATE_OFF
Network layer restarting
Network failover: WiFi-->Ethernet
Network delaying 15 seconds before connection
Network State: NETWORK_STATE_OFF --> NETWORK_STATE_DELAY
Network starting user MQTT Client on Ethernet
```

```
Menu: PointPerfect Corrections
Time to first RTK Fix: 12s Restarts: 0
settings.pointPerfectKeyDistributionTopic: /pp/ubx/0236/Lb
Days until keys expire: 39
1) PointPerfect Corrections: Enabled
2) Toggle Auto Key Renewal: Enabled
3) Update Keys
4) Show device ID
c) Clear the Keys
k) Manual Key Entry
g) Geographic Region: EU
x) Exit
3
Attempting to connect to MQTT broker: pp.services.u-blox.com
[175663][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 34.249.163.35
[175715][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():298]: (-32512) SSL - Memory allocation failed
[175725][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.
[175732][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -32512
Failed to connect to MQTT Broker
Using token: ----
{
  token: ----
}
[175800][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 52.213.149.170
[175844][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():298]: (-32512) SSL - Memory allocation failed
[175854][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.
[175862][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -32512
ERROR: Failed to connect to api.thingstream.io on port 443!
Unknown provisioning error. Trying again.
Using token: ----
{
  token: ----
}
[175883][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 52.213.149.170
[175931][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():298]: (-32512) SSL - Memory allocation failed
[175941][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.
[175948][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -32512
ERROR: Failed to connect to api.thingstream.io on port 443!
Unknown provisioning error.
[175958][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
wifiStop called by C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere\menuPP.ino 1582
networkStop called by C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere\WiFi.ino 440
```

PaulZC, 2024-06-19:
I don't think it's related to WiFi. Here it is successfully connected to PP via MQTT on Ethernet, is getting corrections and is pushing them to the ZED. I've not provided WiFi credentials, it's limited to Ethernet only. If I try to update the keys manually, again it fails due to SSL memory allocation, but there is 15K of Heap available... Maybe NetworkClientSecure is just plain greedy?

```
Pushing 185 bytes from /pp/Lb/eu topic to GNSS
NOT pushing 528 bytes of RXM-PMP data to GNSS due to priority
FreeHeap: 38372 / HeapLowestPoint: 24120 / LargestBlock: 4063220 / Used PSRAM: 74188
Log file size: 141993 - Generation rate: 3.5kB/s
Rover Accuracy (m): 0.319, SIV: 30 GNSS State: 3D Fix
FreeHeap: 38372 / HeapLowestPoint: 24120 / LargestBlock: 4063220 / Used PSRAM: 74188
NOT pushing 528 bytes of RXM-PMP data to GNSS due to priority
FreeHeap: 38372 / HeapLowestPoint: 24120 / LargestBlock: 4063220 / Used PSRAM: 74188
NOT pushing 528 bytes of RXM-PMP data to GNSS due to priority
Rover Accuracy (m): 0.294, SIV: 30 GNSS State: 3D Fix
FreeHeap: 38372 / HeapLowestPoint: 24120 / LargestBlock: 4063220 / Used PSRAM: 74188
FreeHeap: 38372 / HeapLowestPoint: 24120 / LargestBlock: 4063220 / Used PSRAM: 74188
Pushing 185 bytes from /pp/Lb/eu topic to GNSS
NOT pushing 528 bytes of RXM-PMP data to GNSS due to priority

SparkFun RTK EVK d99.99
** Bluetooth SPP and BLE broadcasting as: EVK Rover-8EF2 **
Menu: Main
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
6) Configure WiFi
7) Configure TCP/UDP
e) Configure Ethernet
f) Firmware Update
i) Configure Corrections Priorities
n) Configure NTP
p) Configure PointPerfect
r) Configure Radios
s) Configure System
u) Configure User Profiles
+) Enter Command line mode
x) Exit
p

Menu: PointPerfect Corrections
Time to first RTK Fix: 12s Restarts: 0
settings.pointPerfectKeyDistributionTopic: /pp/ubx/0236/Lb
Days until keys expire: 39
1) PointPerfect Corrections: Enabled
2) Toggle Auto Key Renewal: Enabled
3) Update Keys
4) Show device ID
c) Clear the Keys
k) Manual Key Entry
g) Geographic Region: EU
x) Exit
3
Attempting to connect to MQTT broker: pp.services.u-blox.com
[ 66160][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 52.18.11.229
[ 66215][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():298]: (-32512) SSL - Memory allocation failed
[ 66225][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.
[ 66233][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -32512
Failed to connect to MQTT Broker
Using token: ----
{
  token: ----
}
[ 66276][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 52.213.149.170
[ 66318][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():298]: (-32512) SSL - Memory allocation failed
[ 66328][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.
[ 66335][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -32512
ERROR: Failed to connect to api.thingstream.io on port 443!
Unknown provisioning error. Trying again.
Using token: ----
[... 60 lines cut ...]
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 154 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 65 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 110 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 375 bytes from /pp/Lb/eu topic to GNSS
Pushing 512 bytes from /pp/Lb/eu topic to GNSS
Pushing 298 bytes from /pp/Lb/eu topic to GNSS
Pushing 185 bytes from /pp/Lb/eu topic to GNSS
Pushing 185 bytes from /pp/Lb/eu topic to GNSS
Rover Accuracy (m): 0.209, SIV: 30 GNSS State: 3D Fix
FreeHeap: 38276 / HeapLowestPoint: 15468 / LargestBlock: 4063220 / Used PSRAM: 74740
Log file size: 220270 - Generation rate: 15.7kB/s
```

PaulZC, 2024-06-19:
Possibly / probably related to #191 ?

nseidle, 2024-06-19:
It very much is. Two NetworkClientSecures are getting spun up.

nseidle, 2024-06-19:
This is really a duplicate of #191.

closed 2024-06-19 by nseidle