sparkfun/SparkFun_RTK_Everywhere_Firmware #355: (EVK) ETH init mac failed

nseidle, 2024-06-19:
Bluetooth SPP and BLE broadcasting as: EVK Rover-8EF6
      FreeHeap: 117628 / HeapLowestPoint: 116468 / LargestBlock: 4128756 / Used PSRAM: 51476
      FreeHeap: 103980 / HeapLowestPoint: 102428 / LargestBlock: 4063220 / Used PSRAM: 61480
      STATE_ROVER_NOT_STARTED --> STATE_ROVER_NO_FIX, 2024-06-19 17:24:56.734
      E (7725) gpio: gpio_install_isr_service(500): GPIO isr service already installed
      E (7826) w5500.mac: w5500_reset(305): reset timeout
      E (7826) w5500.mac: emac_w5500_init(823): reset w5500 failed
      E (7826) esp_eth: esp_eth_driver_install(228): init mac failed
      [  7833][E][ETH.cpp:659] beginSPI(): SPI Ethernet driver install failed: 263
      FreeHeap: 85752 / HeapLowestPoint: 84480 / LargestBlock: 4063220 / Used PSRAM: 61948
      NTRIP Client start

I am running ESP v3.0.1. I will try a different hardware setup shortly but I'm hesitant to believe this is a hardware issue.

nseidle, 2024-06-19:
The craziest thing... If I insert an SD card and reset, ethernet starts fine.

Note the following error is still present:

        E (8943) gpio: gpio_install_isr_service(500): GPIO isr service already installed

Ah, so if an SD card is *not* present, is SPI getting started before ethernet attempts to spin up the W5500?

        Board Info:
        ------------------------------------------
          Arduino Board     : ESP32_DEV
          Arduino Variant   : esp32
          Arduino FQBN      : esp32:esp32:esp32:FlashSize=16M,DebugLevel=debug,PSRAM=enabled
        ============ Before Setup End ============
        
        
        [  1025][D][esp32-hal-adc.c:305] __analogReadMilliVolts(): Creating cali handle for ADC_0
        [  1033][D][Begin.ino:66] identifyBoard(): Board ADC ID (mV): 3010
        [  1074][E][esp32-hal-gpio.c:166] __digitalWrite(): IO 26 is not set as GPIO.
        LittleFS Started
        [  1087][E][vfs_api.cpp:99] open(): /littlefs/configureViaEthernet.txt does not exist, no permits for creation
        [  1115][D][NVM.ino:1306] loadProfileNumber(): Using profile #0
        PSRAM Size (bytes): 4194304
        [  1417][I][esp32-hal-i2c.c:109] i2cInit(): Initializing I2C Master: sda=21 scl=22 freq=100000
        I2C Devices:
          0x42 - u-blox ZED-F9P GNSS Receiver
          0x43 - u-blox NEO-D9S Correction Data Receiver
        [  1681][I][esp32-hal-i2c.c:109] i2cInit(): Initializing I2C Master: sda=12 scl=15 freq=100000
        I2C Devices:
          0x3D - SSD1306 OLED Driver
        Display started
        SparkFun RTK EVK d99.99
        Reset reason: ESP_RST_PANIC
        ZED-F9P firmware: HPG 1.32
        GNSS ZED online
        [  2444][D][Begin.ino:514] beginSD(): SD card detected
        microSD: Online
        [  2977][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_Settings_1.txt does not exist, no permits for creation
        [  2991][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_Settings_2.txt does not exist, no permits for creation
        [  3005][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_Settings_3.txt does not exist, no permits for creation
        [  3020][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_Settings_4.txt does not exist, no permits for creation
        [  3034][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_Settings_5.txt does not exist, no permits for creation
        [  3048][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_Settings_6.txt does not exist, no permits for creation
        [  3062][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_Settings_7.txt does not exist, no permits for creation
        Profile 'Profile1' loaded
        ZED-F9x configuration update
        NEO-D9S firmware: PMP 1.04
        System time set to: Wednesday, June 19 2024 21:28:18
        [  6694][D][NVM.ino:155] recordSystemSettingsToFileSD(): Removing from SD: /SFE_EVK_Settings_0.txt
        [  6745][D][NVM.ino:174] recordSystemSettingsToFileSD(): Settings recorded to SD: /SFE_EVK_Settings_0.txt
        =========== After Setup Start ============
        INTERNAL Memory Info:
        ------------------------------------------
          Total Size        :   219276 B ( 214.1 KB)
          Free Bytes        :   180396 B ( 176.2 KB)
          Allocated Bytes   :    32544 B (  31.8 KB)
          Minimum Free Bytes:   179064 B ( 174.9 KB)
          Largest Free Block:   110580 B ( 108.0 KB)
        ------------------------------------------
        SPIRAM Memory Info:
        ------------------------------------------
          Total Size        :  4194304 B (4096.0 KB)
          Free Bytes        :  4179100 B (4081.2 KB)
          Allocated Bytes   :    12576 B (  12.3 KB)
          Minimum Free Bytes:  4175824 B (4078.0 KB)
          Largest Free Block:  4128756 B (4032.0 KB)
        ------------------------------------------
        GPIO Info:
        ------------------------------------------
          GPIO : BUS_TYPE[bus/unit][chan]
[... 43 lines cut ...]
        [  8090][D][BLEAdvertising.cpp:529] handleGAPEvent(): handleGAPEvent [event no: 1]
        [  8099][D][BLEDevice.cpp:555] getAdvertising(): get advertising
        [  8105][D][BLEAdvertising.cpp:529] handleGAPEvent(): handleGAPEvent [event no: 6]
        [  8112][I][BluetoothSerial.cpp:248] esp_spp_cb(): ESP_SPP_INIT_EVT
        [  8118][I][BluetoothSerial.cpp:254] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
        [  8126][I][BluetoothSerial.cpp:575] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
        [  8127][D][NVM.ino:174] recordSystemSettingsToFileSD(): Settings recorded to SD: /SFE_EVK_Settings_0.txt
        [  8153][I][BluetoothSerial.cpp:329] esp_spp_cb(): ESP_SPP_START_EVT
        [  8159][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
        [  8400][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
        STATE_ROVER_NOT_STARTED --> STATE_ROVER_NO_FIX, 2024-06-19 21:28:20.224
        [  8937][D][Begin.ino:1352] beginSDSizeCheckTask(): sdSizeCheck Task started
        E (8943) gpio: gpio_install_isr_service(500): GPIO isr service already installed
        Rover Accuracy (m): 0.393, SIV: 27 GNSS State: 3D Fix
        STATE_ROVER_NO_FIX --> STATE_ROVER_FIX, 2024-06-19 21:28:20.637
        Rover Accuracy (m): 0.363, SIV: 27 GNSS State: 3D Fix
        Rover Accuracy (m): 0.341, SIV: 27 GNSS State: 3D Fix
        SD card size: 29.1 GB / Free space: 29.1 GB
        [ 13793][D][Begin.ino:1364] deleteSDSizeCheckTask(): sdSizeCheck Task deleted
        [ 13805][D][menuMessages.ino:318] beginLogging(): Using last log file.
        Log file name: /SFE_EVK_240619_212540.ubx
        Rover Accuracy (m): 0.322, SIV: 27 GNSS State: 3D Fix
        Rover Accuracy (m): 0.310, SIV: 27 GNSS State: 3D Fix
        Rover Accuracy (m): 0.300, SIV: 27 GNSS State: 3D Fix
        Rover Accuracy (m): 0.292, SIV: 27 GNSS State: 3D Fix

PaulZC, 2024-06-20:
I think this has to be a ```SPI.begin``` issue...

```SPI.begin``` is only called in ```resetSPI``` and ```sdCardPresentSoftwareTest```.

```sdCardPresentSoftwareTest``` is only called by ```sdCardPresent```, but won't be called if pin_microSD_CardDetect says no card is present.

```resetSPI``` is called by ```sdCardPresent```, but won't be called if pin_microSD_CardDetect says no card is present.

```resetSPI``` is called by ```beginSD```, but ```sdCardPresent``` is called first.

Long story short, ```SPI.begin``` doesn't get called if no SD card is present.

I think we need a ```beginSPI```.

PaulZC, 2024-06-20:
We have no control over ```E (6715) gpio: gpio_install_isr_service(500): GPIO isr service already installed``` - that's being called by the core

PaulZC, 2024-06-20:
Fixed in #375

commit 32bb703d (2024-06-20): Fix #355
commit 494cc249 (2024-06-20): Merge pull request #375 from sparkfun/Fix_#355

Fix #355
closed 2024-07-08 by nseidle