sparkfun/SparkFun_RTK_Everywhere_Firmware #356: (EVK) Device reporting large rover inaccuracy after factory reset

nseidle, 2024-06-19:
E (7021) w5500.mac: emac_w5500_init(823): reset w5500 failed
      E (7021) esp_eth: esp_eth_driver_install(228): init mac failed
      [  7027][E][ETH.cpp:659] beginSPI(): SPI Ethernet driver install failed: 263
      Rover Accuracy (m): 429496.719, SIV: 0 GNSS State: No Fix
      System time set to: Wednesday, June 19 2024 17:50:40
      Rover Accuracy (m): 429496.719, SIV: 0 GNSS State: No Fix
      Rover Accuracy (m): 429496.719, SIV: 0 GNSS State: No Fix
      Rover Accuracy (m): 429496.719, SIV: 0 GNSS State: No Fix

To replicate, factory reset a device and the rover accuracy will be huge.

This is likely due to GNSS reset. Should we just not display rover accuracy until we get a fix?

PaulZC, 2024-06-20:
We have no control over the 429496.719. That's just the default hAcc while the module gets a fix. Suspiciously close to 0xFFFFFFFF * 0.1mm!

We can easily cap this at ">30m" [like we do on the display](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/fcaf7cab99e4c40ff118e5b8d8c9e332135b58f5/Firmware/RTK_Everywhere/Display.ino#L1306).

PaulZC, 2024-06-20:
Fixed in #376

commit a4d69ef0 (2024-06-20): Fix #356
commit 0f947737 (2024-06-26): Merge pull request #376 from sparkfun/Fix_#356

Fix #356
closed 2024-07-08 by nseidle