sparkfun/SparkFun_RTK_Everywhere_Firmware #362: MQTT broker fails to connect post provisioning

nseidle, 2024-06-19:
To replicate:

* Factory reset (s, r, y)
* Turn off Ethernet/WiFi failover (currently TCP menu 7, f)
* Set default network to WiFi (currently TCP menu 7, d)
* Enter WiFi credentials (6)
* Enable PointPerfect (p, 1)
* Turn on heap reporting (s, d, 1)
* Exit menus to save
* Force system reset (s, d, r)

Device will connect to WiFi and successfully provision (yay!).

Once device has fully booted, it will start WiFi again but WiFi fails to connect because of RAM limitations.

Device should connect to WiFi.

    [ 42363][W][STA.cpp:533] disconnect(): STA already disconnected.

The above warning is a bit disconcerting.

        ------------------------------------------
        GPIO Info:
        ------------------------------------------
          GPIO : BUS_TYPE[bus/unit][chan]
          --------------------------------------
             0 : GPIO
             1 : UART_TX[0]
             2 : GPIO
             3 : UART_RX[0]
             4 : GPIO
             5 : GPIO
            12 : I2C_MASTER_SDA[1]
            15 : I2C_MASTER_SCL[1]
            21 : I2C_MASTER_SDA[0]
            22 : I2C_MASTER_SCL[0]
            25 : UART_RX[2]
            26 : GPIO
            27 : GPIO
            32 : GPIO
            33 : UART_TX[2]
            34 : GPIO
            35 : ADC_ONESHOT[0][7]
            36 : GPIO
            39 : GPIO
        ============ After Setup End =============
        [  6953][D][States.ino:527] stateUpdate(): PointPerfect Keys starting WiFi
        FreeHeap: 180632 / HeapLowestPoint: 178152 / LargestBlock: 4128756 / Used PSRAM: 14520
        FreeHeap: 180632 / HeapLowestPoint: 178152 / LargestBlock: 4128756 / Used PSRAM: 14520
        STATE_KEYS_STARTED --> STATE_KEYS_PROVISION_WIFI_STARTED, 2024-06-19 20:00:07.155
        Connecting WiFi... [  7091][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: sparkfun-iot
        [  9890][I][WiFiMulti.cpp:129] run(): [WIFI] scan done
        [  9895][I][WiFiMulti.cpp:134] run(): [WIFI] 38 networks found
        [  9901][D][WiFiMulti.cpp:233] run():         0: [11][74:AC:B9:E7:7E:4C] sparkfun-internal (-58) (*) (visible)
        [  9911][D][WiFiMulti.cpp:228] run():  --->   1: [11][7E:AC:B9:E7:7E:4C] sparkfun-iot (-58) (*) (visible)
        [  9920][D][WiFiMulti.cpp:233] run():         2: [11][82:AC:B9:E7:7E:4C] sparkfun-funions (-59) (*) (visible)
        [  9930][D][WiFiMulti.cpp:233] run():         3: [11][7A:AC:B9:E7:7E:4C] sparkfun-guest (-61) ( ) (visible)
        [  9940][D][WiFiMulti.cpp:228] run():  --->   4: [6][EA:63:DA:25:70:7C] sparkfun-iot (-69) (*) (visible)
        [  9949][D][WiFiMulti.cpp:233] run():         5: [6][EE:63:DA:25:70:7C] sparkfun-funions (-69) (*) (visible)
        [  9959][D][WiFiMulti.cpp:233] run():         6: [6][E0:63:DA:25:70:7C] sparkfun-internal (-69) (*) (visible)
        [  9968][D][WiFiMulti.cpp:233] run():         7: [6][E6:63:DA:25:70:7C] sparkfun-guest (-69) ( ) (visible)
        [  9978][D][WiFiMulti.cpp:233] run():         8: [11][7E:A7:B0:ED:EB:4A] DIRECT-0uLexmark MC3326adw-4aeb (-75) (*) (visible)
        [  9989][D][WiFiMulti.cpp:233] run():         9: [6][FA:8F:CA:69:68:D2] Den TV (-76) ( ) (visible)
        [  9998][D][WiFiMulti.cpp:233] run():         10: [6][E0:63:DA:28:0A:84] sparkfun-internal (-79) (*) (visible)
        [ 10008][D][WiFiMulti.cpp:228] run():  --->   11: [6][EA:63:DA:28:0A:84] sparkfun-iot (-79) (*) (visible)
        [ 10017][D][WiFiMulti.cpp:233] run():         12: [6][E6:63:DA:28:0A:84] sparkfun-guest (-80) ( ) (visible)
        [ 10027][D][WiFiMulti.cpp:233] run():         13: [6][EE:63:DA:28:0A:84] sparkfun-funions (-80) (*) (visible)
        [ 10037][D][WiFiMulti.cpp:233] run():         14: [1][60:D0:2C:93:D3:48] NI_IOT (-83) (*) (visible)
        [ 10046][D][WiFiMulti.cpp:228] run():  --->   15: [1][B6:8B:A9:DA:31:69] sparkfun-iot (-83) (*) (visible)
        [ 10055][D][WiFiMulti.cpp:233] run():         16: [1][BA:8B:A9:DA:31:69] sparkfun-funions (-83) (*) (visible)
[... 160 lines cut ...]
        [ 42353][I][WiFiMulti.cpp:249] run(): [WIFI] Connecting BSSID: 7E:AC:B9:E7:7E:4C SSID: sparkfun-iot Channel: 11 (-57)
        [ 42363][W][STA.cpp:533] disconnect(): STA already disconnected.
        [ 42572][I][WiFiMulti.cpp:272] run(): [WIFI] Connecting done.
        [ 42578][D][WiFiMulti.cpp:273] run(): [WIFI] SSID: sparkfun-iot
        [ 42584][D][WiFiMulti.cpp:274] run(): [WIFI] IP: 10.10.10.121
        [ 42589][D][WiFiMulti.cpp:275] run(): [WIFI] MAC: 7E:AC:B9:E7:7E:4C
        [ 42595][D][WiFiMulti.cpp:276] run(): [WIFI] Channel: 11
        [ 42601][D][WiFiMulti.cpp:334] resetFails(): [WIFI] Resetting failure flags
        
        WiFi 'sparkfun-iot' IP address: 10.10.10.121, RSSI: -58
        Rover Accuracy (m): 0.305, SIV: 32 GNSS State: 3D Fix
        FreeHeap: 60260 / HeapLowestPoint: 58400 / LargestBlock: 4063220 / Used PSRAM: 87608
        [ 42769][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 34.250.253.204
        [ 44937][E][ssl_client.cpp:36] _handle_error(): [ssl_starttls_handshake():313]: (-17168) RSA - The private key operation failed : BIGNUM - Memory allocation failed
        [ 44951][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -17168
        Failed to connect to MQTT broker pp.services.u-blox.com
        [ 44963][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
        FreeHeap: 59872 / HeapLowestPoint: 1388 / LargestBlock: 4063220 / Used PSRAM: 94416
        FreeHeap: 59872 / HeapLowestPoint: 1388 / LargestBlock: 4063220 / Used PSRAM: 94416
        FreeHeap: 59872 / HeapLowestPoint: 1388 / LargestBlock: 4063220 / Used PSRAM: 90384
        Rover Accuracy (m): 0.305, SIV: 32 GNSS State: 3D Fix
        E (61891) wifi:NAN WiFi stop
        [ 45032][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
        FreeHeap: 81904 / HeapLowestPoint: 1388 / LargestBlock: 4063220 / Used PSRAM: 73148
        Network delaying 7 seconds befor

PaulZC, 2024-06-20:
See: https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/issues/367#issuecomment-2181139566

Possible solution on its way...!

nseidle, 2024-06-26:
Connecting WiFi... [ 26283][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: sparkfun-iot
      [ 37261][I][WiFiMulti.cpp:129] run(): [WIFI] scan done
      [ 37266][I][WiFiMulti.cpp:134] run(): [WIFI] 32 networks found
      [ 37272][D][WiFiMulti.cpp:233] run():         0: [11][74:AC:B9:E7:7E:4C] sparkfun-internal (-50) (*) (visible)
      [ 37282][D][WiFiMulti.cpp:233] run():         1: [11][82:AC:B9:E7:7E:4C] sparkfun-funions (-50) (*) (visible)
      [ 37292][D][WiFiMulti.cpp:233] run():         2: [11][7A:AC:B9:E7:7E:4C] sparkfun-guest (-52) ( ) (visible)
      [ 37301][D][WiFiMulti.cpp:228] run():  --->   3: [11][7E:AC:B9:E7:7E:4C] sparkfun-iot (-52) (*) (visible)
      [ 37311][D][WiFiMulti.cpp:233] run():         4: [6][E0:63:DA:25:70:7C] sparkfun-internal (-73) (*) (visible)
      [ 37320][D][WiFiMulti.cpp:233] run():         5: [6][E0:63:DA:28:0A:84] sparkfun-internal (-74) (*) (visible)
      [ 37330][D][WiFiMulti.cpp:228] run():  --->   6: [6][EA:63:DA:25:70:7C] sparkfun-iot (-74) (*) (visible)
      [ 37340][D][WiFiMulti.cpp:233] run():         7: [6][E6:63:DA:25:70:7C] sparkfun-guest (-74) ( ) (visible)
      [ 37349][D][WiFiMulti.cpp:233] run():         8: [6][EE:63:DA:25:70:7C] sparkfun-funions (-74) (*) (visible)
      [ 37359][D][WiFiMulti.cpp:233] run():         9: [6][E6:63:DA:28:0A:84] sparkfun-guest (-75) ( ) (visible)
      [ 37368][D][WiFiMulti.cpp:228] run():  --->   10: [6][EA:63:DA:28:0A:84] sparkfun-iot (-75) (*) (visible)
      [ 37378][D][WiFiMulti.cpp:233] run():         11: [6][EE:63:DA:28:0A:84] sparkfun-funions (-75) (*) (visible)
      [ 37388][D][WiFiMulti.cpp:233] run():         12: [11][7E:A7:B0:ED:EB:4A] DIRECT-0uLexmark MC3326adw-4aeb (-76) (*) (visible)
      [ 37399][D][WiFiMulti.cpp:233] run():         13: [11][60:D0:2C:53:DE:98] NI-Guest (-84) (*) (visible)
      [ 37408][D][WiFiMulti.cpp:233] run():         14: [1][B2:8B:A9:DA:31:69] sparkfun-guest (-86) ( ) (visible)
      [ 37418][D][WiFiMulti.cpp:233] run():         15: [11][60:D0:2C:93:DB:38] NI_IOT (-86) (*) (visible)
      [ 37427][D][WiFiMulti.cpp:233] run():         16: [11][60:D0:2C:93:DE:98] NI_IOT (-86) (*) (visible)
      [ 37436][D][WiFiMulti.cpp:233] run():         17: [1][BA:8B:A9:DA:31:69] sparkfun-funions (-87) (*) (visible)
      [ 37446][D][WiFiMulti.cpp:233] run():         18: [1][60:D0:2C:13:D3:48] NI-Wireless (-87) (*) (visible)
      [ 37455][D][WiFiMulti.cpp:228] run():  --->   19: [1][B6:8B:A9:DA:31:69] sparkfun-iot (-87) (*) (visible)
      [ 37465][D][WiFiMulti.cpp:233] run():         20: [6][FA:8F:CA:69:68:D2] Den TV (-87) ( ) (visible)
      [ 37474][D][WiFiMulti.cpp:233] run():         21: [11][60:D0:2C:13:DB:38] NI-Wireless (-87) (*) (visible)
      [ 37483][D][WiFiMulti.cpp:233] run():         22: [11][60:D0:2C:53:DB:38] NI-Guest (-87) (*) (visible)
      [ 37492][D][WiFiMulti.cpp:233] run():         23: [1][60:D0:2C:93:D6:88] NI_IOT (-88) (*) (visible)
      [ 37501][D][WiFiMulti.cpp:233] run():         24: [11][60:D0:2C:93:D7:48] NI_IOT (-88) (*) (visible)
      [ 37510][D][WiFiMulti.cpp:233] run():         25: [1][60:D0:2C:13:D6:88] NI-Wireless (-89) (*) (visible)
      [ 37520][D][WiFiMulti.cpp:233] run():         26: [1][60:D0:2C:53:D6:88] NI-Guest (-89) (*) (visible)
      [ 37529][D][WiFiMulti.cpp:228] run():  --->   27: [6][EA:63:DA:28:14:B4] sparkfun-iot (-90) (*) (visible)
      [ 37538][D][WiFiMulti.cpp:233] run():         28: [11][60:D0:2C:13:D7:48] NI-Wireless (-90) (*) (visible)
      [ 37548][D][WiFiMulti.cpp:233] run():         29: [6][E0:63:DA:28:14:B4] sparkfun-internal (-91) (*) (visible)
      [ 37558][D][WiFiMulti.cpp:233] run():         30: [6][EE:63:DA:28:14:B4] sparkfun-funions (-91) (*) (visible)
      [ 37567][D][WiFiMulti.cpp:233] run():         31: [11][60:D0:2C:53:D7:48] NI-Guest (-92) (*) (visible)
      [ 37577][I][WiFiMulti.cpp:249] run(): [WIFI] Connecting BSSID: 7E:AC:B9:E7:7E:4C SSID: sparkfun-iot Channel: 11 (-52)
      [ 37587][W][STA.cpp:533] disconnect(): STA already disconnected.
      [ 37725][I][WiFiMulti.cpp:272] run(): [WIFI] Connecting done.
      [ 37731][D][WiFiMulti.cpp:273] run(): [WIFI] SSID: sparkfun-iot
      [ 37737][D][WiFiMulti.cpp:274] run(): [WIFI] IP: 10.10.10.125
      [ 37742][D][WiFiMulti.cpp:275] run(): [WIFI] MAC: 7E:AC:B9:E7:7E:4C
      [ 37749][D][WiFiMulti.cpp:276] run(): [WIFI] Channel: 11
      [ 37754][D][WiFiMulti.cpp:334] resetFails(): [WIFI] Resetting failure flags
      
      WiFi 'sparkfun-iot' IP address: 10.10.10.125, RSSI: -51
      Rover Accuracy (m): 0.298, SIV: 32 GNSS State: 3D Fix
      FreeHeap: 58780 / HeapLowestPoint: 57060 / LargestBlock: 4063220 / Used PSRAM: 87212
      Log file size: 44326 - Generation rate: 7.8kB/s
      [ 37978][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 34.247.204.55
      [ 39438][D][ssl_client.cpp:322] ssl_starttls_handshake(): Protocol is TLSv1.2 Ciphersuite is TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
      [ 39450][D][ssl_client.cpp:324] ssl_starttls_handshake(): Record expansion is 29
      FreeHeap: 58496 / HeapLowestPoint: 56200 / LargestBlock: 3997684 / Used PSRAM: 139204
      Rover Accuracy (m): 0.298, SIV: 32 GNSS State: 3D Fix
      FreeHeap: 58464 / HeapLowestPoint: 56200 / LargestBlock: 3997684 / Used PSRAM: 139204
      FreeHeap: 58464 / HeapLowestPoint: 56200 / LargestBlock: 3997684 / Used PSRAM: 139204

Working well! Thanks!

commit bdb88202 (2024-06-26): Merge pull request #390 from sparkfun/Fix_#362

Fix #362
closed 2024-07-08 by nseidle