sparkfun/SparkFun_RTK_Everywhere_Firmware #363: Getting keys over Ethernet or WiFi

nseidle, 2024-06-19:
The device should use ethernet, if available, to provision/update keys during POR.

To replicate:

* Factory reset
* Disable network failover
* Set default network to ethernet
* Enable PointPerfect
* Exit menus to save settings
* Reset the device

Device will fail to provision because in `States.ino` the line:

    else if (wifiNetworkCount() == 0 && strlen(settings.pointPerfectCurrentKey) == 0)

is WiFi centric. 

Instead, we should check for ethernet connectivity as well and enter `STATE_KEYS_PROVISION_WIFI_STARTED` (please rename states as needed) or add states to start ethernet.

PaulZC, 2024-06-27:
```mqttClientStart``` is only called by ```mqttClientUpdate``` when ```mqttClientState``` is ```MQTT_CLIENT_OFF```.

```mqttClientStart``` calls ```mqttClientStop(false)```. ```mqttClientStop``` should be advancing the state to ```mqttClientSetState(MQTT_CLIENT_ON)```. Is it?

Turning on ```debugMqttClientState``` and ```debugNetworkLayer```.

```
[ 13983][D][NVM.ino:174] recordSystemSettingsToFileSD(): Settings recorded to SD: /SFE_EVK_Settings_0.txt
[ 14004][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
[ 14233][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
STATE_ROVER_NOT_STARTED --> STATE_ROVER_NO_FIX, 2024-06-27 13:17:48.233
[ 14768][D][Begin.ino:1386] deleteSDSizeCheckTask(): sdSizeCheck Task deleted
Log file name: /SFE_EVK_240627_131748.ubx
Ethernet connected
FreeHeap: 84256 / HeapLowestPoint: 79092 / LargestBlock: 4128756 / Used PSRAM: 63240
MQTT Client start
MQTT Client Off Last Uptime: 0 00:00:00.000 (Reconnects: 0)
FreeHeap: 84256 / HeapLowestPoint: 79092 / LargestBlock: 4128756 / Used PSRAM: 63240
MQTT_CLIENT_OFF --> MQTT_CLIENT_ON
Rover Accuracy (m): 0.153, SIV: 31 GNSS State: 3D Fix
STATE_ROVER_NO_FIX --> STATE_ROVER_FIX, 2024-06-27 13:17:49.397
Log file size: 4125 - Generation rate: 0.8kB/s
*WIFI_STATE_OFF
Ethernet started with DHCP
Network starting user MQTT Client on Active
Network request to start Active
Network layer starting Active
Network State: NETWORK_STATE_OFF --> NETWORK_STATE_DELAY
MQTT_CLIENT_ON --> MQTT_CLIENT_NETWORK_STARTED
networkTypeUpdate, network->type: Active --> Ethernet
networkTypeUpdate, network->requestedNetwork: Active --> Ethernet
Network starting Ethernet
Network State: NETWORK_STATE_DELAY --> NETWORK_STATE_CONNECTING
Network connected to Ethernet
Network State: NETWORK_STATE_CONNECTING --> NETWORK_STATE_IN_USE
Ethernet IP address: 192.168.0.91
MQTT_CLIENT_NETWORK_STARTED --> MQTT_CLIENT_CONNECTING_2_SERVER
[ 16009][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_certificate_0.txt does not exist, no permits for creation
Failed to read from LittleFS: /SFE_EVK_certificate_0.txt
MQTT Client Connecting Last Uptime: 0 00:00:00.000 (Reconnects: 0)
Freeing mqttSecureClient
FreeHeap: 84224 / HeapLowestPoint: 79092 / LargestBlock: 4063220 / Used PSRAM: 69584
FreeHeap: 84224 / HeapLowestPoint: 79092 / LargestBlock: 4063220 / Used PSRAM: 65552
Network stopping user MQTT Client on Ethernet
MQTT_CLIENT_CONNECTING_2_SERVER --> MQTT_CLIENT_OFF
FreeHeap: 84224 / HeapLowestPoint: 79092 / LargestBlock: 4063220 / Used PSRAM: 65552
MQTT Client start
MQTT Client Off Last Uptime: 0 00:00:00.000 (Reconnects: 0)
FreeHeap: 84224 / HeapLowestPoint: 79092 / LargestBlock: 4063220 / Used PSRAM: 65552
MQTT_CLIENT_OFF --> MQTT_CLIENT_ON
Network starting user MQTT Client on Ethernet
MQTT_CLIENT_ON --> MQTT_CLIENT_NETWORK_STARTED
MQTT_CLIENT_NETWORK_STARTED --> MQTT_CLIENT_CONNECTING_2_SERVER
[ 16116][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_certificate_0.txt does not exist, no permits for creation
Failed to read from LittleFS: /SFE_EVK_certificate_0.txt
MQTT Client Connecting Last Uptime: 0 00:00:00.000 (Reconnects: 0)
Freeing mqttSecureClient
FreeHeap: 84192 / HeapLowestPoint: 79092 / LargestBlock: 4063220 / Used PSRAM: 71888
FreeHeap: 84192 / HeapLowestPoint: 79092 / LargestBlock: 4063220 / Used PSRAM: 67856
Network stopping user MQTT Client on Ethernet
MQTT_CLIENT_CONNECTING_2_SERVER --> MQTT_CLIENT_OFF
FreeHeap: 84192 / HeapLowestPoint: 79092 / LargestBlock: 4063220 / Used PSRAM: 67856
MQTT Client start
```

PaulZC, 2024-06-27:
OK. This is critical:

```
Failed to read from LittleFS: /SFE_EVK_certificate_0.txt
MQTT Client Connecting Last Uptime: 0 00:00:00.000 (Reconnects: 0)
Freeing mqttSecureClient
```

SFE_EVK_certificate_0.txt doesn't exist because it is provided during ZTP. And we haven't yet performed ZTP.

If I open the PointPerfect menu and then Provision Device, all goes well:

```
MQTT Client start
MQTT Client Off Last Uptime: 0 00:00:00.000 (Reconnects: 0)
FreeHeap: 83940 / HeapLowestPoint: 79064 / LargestBlock: 4063220 / Used PSRAM: 83976
MQTT_CLIENT_OFF --> MQTT_CLIENT_ON

SparkFun RTK EVK d99.99
** Bluetooth SPP and BLE broadcasting as: EVK Rover-8EF2 **
Menu: Main
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
6) Configure WiFi
7) Configure TCP/UDP
e) Configure Ethernet
f) Firmware Update
i) Configure Corrections Priorities
n) Configure NTP
p) Configure PointPerfect
r) Configure Radios
s) Configure System
t) Configure Instrument Setup
u) Configure User Profiles
+) Enter Command line mode
x) Exit
p

Menu: PointPerfect Corrections
Days until keys expire: No keys
1) PointPerfect Corrections: Enabled
2) Toggle Auto Key Renewal: Enabled
3) Provision Device
4) Show device ID
c) Clear the Keys
k) Manual Key Entry
g) Geographic Region: EU
x) Exit
3
[ 23711][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_certificate_0.txt does not exist, no permits for creation
[ 23722][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
[ 24033][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 52.209.93.91
[ 24040][V][ssl_client.cpp:68] start_ssl_client(): Free internal heap before TLS 83696
[ 24048][V][ssl_client.cpp:75] start_ssl_client(): Starting socket (domain 2)
[ 24130][V][ssl_client.cpp:166] start_ssl_client(): Seeding the random number generator
[ 24139][V][ssl_client.cpp:174] start_ssl_client(): Setting up the SSL/TLS structure...
[ 24148][V][ssl_client.cpp:194] start_ssl_client(): Loading CA cert
[ 24157][V][ssl_client.cpp:288] start_ssl_client(): Setting hostname for TLS session...
[ 24167][V][ssl_client.cpp:301] start_ssl_client(): Free internal heap after mbedtls_ssl_setup 83416
[ 24176][V][ssl_client.cpp:311] ssl_starttls_handshake(): Performing the SSL/TLS handshake...
[ 24612][V][ssl_client.cpp:332] ssl_starttls_handshake(): Verifying peer X.509 certificate...
[ 24621][V][ssl_client.cpp:340] ssl_starttls_handshake(): Certificate verified.
[ 24628][V][ssl_client.cpp:356] ssl_starttls_handshake(): Free internal heap after TLS 83216
client connected to api.thingstream.io on port 443
[ 24637][V][HTTPClient.cpp:230] beginInternal(): url: https://api.thingstream.io/ztp/pointperfect/credentials
[ 24651][D][HTTPClient.cpp:275] beginInternal(): switching host from '' to 'api.thingstream.io'. disconnecting first
[ 24662][D][HTTPClient.cpp:366] disconnect(): tcp stop
[... 151 lines cut ...]
Rover Accuracy (m): 0.155, SIV: 31 GNSS State: 3D Fix
Rover Accuracy (m): 0.155, SIV: 31 GNSS State: 3D Fix
Rover Accuracy (m): 0.155, SIV: 31 GNSS State: 3D Fix
Log file size: 2190257 - Generation rate: 3.5kB/s
Rover Accuracy (m): 0.155, SIV: 31 GNSS State: 3D Fix
Rover Accuracy (m): 0.155, SIV: 31 GNSS State: 3D Fix
Log file size: 2207795 - Generation rate: 3.5kB/s
Rover Accuracy (m): 0.155, SIV: 31 GNSS State: 3D Fix
Rover Accuracy (m): 0.155, SIV: 31 GNSS State: 3D Fix
Rover Accuracy (m): 0.153, SIV: 31 GNSS State: 3D Fix
*WIFI_STATE_OFF
Log file size: 2225325 - Generation rate: 3.5kB/s
Rover Accuracy (m): 0.671, SIV: 31 GNSS State: RTK Float
STATE_ROVER_FIX --> STATE_ROVER_RTK_FLOAT, 2024-06-27 13:29:33.515
Rover Accuracy (m): 0.224, SIV: 31 GNSS State: RTK Float
Log file size: 2243228 - Generation rate: 3.6kB/s
Rover Accuracy (m): 0.157, SIV: 31 GNSS State: RTK Float
Rover Accuracy (m): 0.120, SIV: 31 GNSS State: RTK Float
Rover Accuracy (m): 0.099, SIV: 31 GNSS State: RTK Float
Log file size: 2260479 - Generation rate: 3.5kB/s
Rover Accuracy (m): 0.087, SIV: 31 GNSS State: RTK Float
Rover Accuracy (m): 0.080, SIV: 31 GNSS State: RTK Float
```

I think this is just a cart-before-horse issue. If SFE_EVK_certificate_0.txt does not exist, we need to attempt ZTP to get it.

PaulZC, 2024-06-27:
Has the keys states WiFi dependency been fixed? No. That's still there.

```menuPointPerfect``` option 3 (Provision Device / Update Keys) has had a facelift. It now works for non-WiFi networks. I.e. Ethernet. ```pointperfectProvisionDevice``` is called if the certificate file does not exist or the current key or key distribution topic are null length or the certificates are invalid. That's excellent. We need a similar strategy for the states.

```pointperfectUpdateKeys``` is different. It is only called when the certificates are present and valid. It gets the keys by subscribing to the MQTT key distribution topic (also provided during ZTP). It falls back to ```pointperfectProvisionDevice``` if it can't connect to the broker or the topic.

```pointperfectProvisionDevice``` calls ```pointperfectTryZtpToken```. ```pointperfectTryZtpToken``` does an HTTP POST to retrieve the certificates, key distribution topic, regional correction topics, and keys.

So, we have two things to do. The states also need to check if valid certificates etc. are present. And we need to fix the WiFi dependency.

PaulZC, 2024-06-28:
```pointperfectUpdateKeys``` uses a separate instance of ```NetworkClientSecure``` (```secureClient```) to connect to the PointPerfect broker via MQTT. It uses a separate instance of ```MqttClient``` (```menuppMqttClient```) on top of that ```secureClient```. It isn't 'integrated' with the main MQTT functionality - subscribing to the correction topic. It isn't a network user. It does a blind ```connect``` and hopes that there is a network there to provide connectivity. If no network is connected, it will fail. It has no way to request and wait on a network before attempting to connect.

PaulZC, 2024-06-28:
```pointperfectProvisionDevice``` calls ```pointperfectTryZtpToken```. ```pointperfectTryZtpToken``` also uses a separate instance of ```NetworkClientSecure``` (```client```). It uses a separate instance of ```HTTPClient``` (```httpClient```) on top of that ```client```. It isn't a network user. It does a blind ```begin``` and hopes there is a network to provide connectivity. If no network is connected, it will fail. It has no way to request and wait on a network before attempting to connect.

I think we need to update both ```pointperfectUpdateKeys``` and ```pointperfectProvisionDevice``` so they use the network layer in a more integrated way...

PaulZC, 2024-06-28:
```MQTT_Client``` already subscribes to the ```pointPerfectKeyDistributionTopic```. The keys arrive in UBX format and get pushed straight to the ZED. There's duplication / redundancy here we can use.

PaulZC, 2024-06-28:
I'm in the process of adding a ```HTTP_Client``` to handle ZTP. Work in progress:

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commits/Fix_%23363/

nseidle, 2024-06-28:
I believe the rolling `MQTT Client start` was fixed by [Lee's PR](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/pull/396) merged a few minutes ago.

PaulZC, 2024-06-29:
Agreed! I'll revert the new CHECK_CERTS state. And add the new timeout code to HTTP_Client.

PaulZC, 2024-07-02:
OK. Much progress...

We have a shiny new ZTP state machine: ```updateProvisioning```. The state changes are tracked through ```provisioningSetState```; it has full Periodic Display and Dead Man Walking support. The new ```HTTP_Client``` is controlled by ```httpClientModeNeeded```. The ZTP success / failure is reported back through ```ztpResponse```.

It needs a bit more polish, but I'm seeing many good things:

From a factory reset (ensuring any previous certs and private keys are erased):

BLE starts up as normal, then MQTT gets its foot in the door:

```
[ 11340][I][BluetoothSerial.cpp:248] esp_spp_cb(): ESP_SPP_INIT_EVT
[ 11346][I][BluetoothSerial.cpp:254] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
[ 11354][I][BluetoothSerial.cpp:575] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[ 11366][I][BluetoothSerial.cpp:329] esp_spp_cb(): ESP_SPP_START_EVT
[ 11368][D][NVM.ino:174] recordSystemSettingsToFileSD(): Settings recorded to SD: /SFE_EVK_Settings_0.txt
[ 11404][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
[ 11640][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
FreeHeap: 103988 / HeapLowestPoint: 102412 / LargestBlock: 4128756 / Used PSRAM: 62020
STATE_ROVER_NOT_STARTED --> STATE_ROVER_NO_FIX, 2024-07-02 12:28:12.522
[ 12188][D][Begin.ino:1363] beginSDSizeCheckTask(): sdSizeCheck Task started
[ 12195][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type ETHERNET_SPI (48) successfully set to 0x401184a8
[ 12207][V][esp32-hal-periman.c:174] perimanSetPinBusExtraType(): Successfully set extra_type ETH_CS[0] for pin 27
E (12056) gpio: gpio_install_isr_service(500): GPIO isr service already installed
[ 12582][V][ETH.cpp:104] _onEthEvent(): eth0 Started
[ 12587][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 1 - ETH_START
ETH Started
[ 12595][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 39 successfully set to type ETHERNET_SPI (48) with bus 0x3ffc7408
[ 12607][V][esp32-hal-periman.c:174] perimanSetPinBusExtraType(): Successfully set extra_type ETH_IRQ for pin 39
FreeHeap: 81328 / HeapLowestPoint: 80168 / LargestBlock: 4128756 / Used PSRAM: 63072
MQTT Client start
mqttClientStop(false) called by C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere\MQTT_Client.ino 496
MQTT Client Off Last Uptime: 0 00:00:00.000 (Reconnects: 0)
FreeHeap: 81328 / HeapLowestPoint: 80168 / LargestBlock: 4128756 / Used PSRAM: 63072
MQTT_CLIENT_OFF --> MQTT_CLIENT_ON
Rover Accuracy (m): 0.188, SIV: 32 GNSS State: 3D Fix
Provisioning: PROVISIONING_OFF --> PROVISIONING_WAIT_RTC
FreeHeap: 81328 / HeapLowestPoint: 80168 / LargestBlock: 4128756 / Used PSRAM: 63072
STATE_ROVER_NO_FIX --> STATE_ROVER_FIX, 2024-07-02 12:28:13.035
Network starting user MQTT Client on Active
Network request to start Active
Network layer starting Active
Network State: NETWORK_STATE_OFF --> NETWORK_STATE_DELAY
MQTT_CLIENT_ON --> MQTT_CLIENT_NETWORK_STARTED
```

MQTT will fail because we don't yet have a cert and private key.

Here we see ZTP provisioning start. The PROVISIONING_CHECK_REMAINING state checks the certs, sees they don't exist, and so proceeds to PROVISIONING_STARTED which starts the HTTP_Client. Meanwhile, the MQTT_Client discovers it also lacks certs and private key, and so shuts down (```MQTT_CLIENT_CONNECTING_2_SERVER no certificate available```):

```
Provisioning: PROVISIONING_WAIT_RTC --> PROVISIONING_NOT_STARTED
networkTypeUpdate, network->type: Active --> Ethernet
networkTypeUpdate, network->requestedNetwork: Active --> Ethernet
Network starting Ethernet
Network State: NETWORK_STATE_DELAY --> NETWORK_STATE_CONNECTING
Provisioning: PROVISIONING_NOT_STARTED --> PROVISIONING_CHECK_REMAINING
[ 12823][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_certificate_0.txt does not exist, no permits for creation
File /SFE_EVK_certificate_0.txt does not exist on LittleFS
ERROR - Invalid certificate format!
Certificate is corrupt.
[ 12848][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_privateKey_0.txt does not exist, no permits for creation
File /SFE_EVK_privateKey_0.txt does not exist on LittleFS
ERROR - Invalid private key format!
PrivateKey is corrupt.
Stored certificates are NOT valid
Invalid certificates or keys. Starting provisioning
Provisioning: PROVISIONING_CHECK_REMAINING --> PROVISIONING_CHECK_NETWORK
Provisioning: PROVISIONING_CHECK_NETWORK --> PROVISIONING_STARTING
[ 14694][V][ETH.cpp:95] _onEthEvent(): eth0 Connected
[... 251 lines cut ...]
Pushing 528 bytes of RXM-PMP data to GNSS
L-Band Eb/N0[dB] (>9 is good): 8.88
Log file size: 78430 - Generation rate: 3.5kB/s
Rover Accuracy (m): 0.176, SIV: 32 GNSS State: 3D Fix
FreeHeap: 81884 / HeapLowestPoint: 79960 / LargestBlock: 3997684 / Used PSRAM: 128668
NOT pushing 192 bytes from /pp/Lb/eu topic to GNSS due to priority
Pushing 528 bytes of RXM-PMP data to GNSS
FreeHeap: 81884 / HeapLowestPoint: 79960 / LargestBlock: 3997684 / Used PSRAM: 128668
Rover Accuracy (m): 0.176, SIV: 32 GNSS State: 3D Fix
FreeHeap: 81884 / HeapLowestPoint: 79960 / LargestBlock: 3997684 / Used PSRAM: 128668
Pushing 528 bytes of RXM-PMP data to GNSS
L-Band Eb/N0[dB] (>9 is good): 9.12
FreeHeap: 81884 / HeapLowestPoint: 79960 / LargestBlock: 3997684 / Used PSRAM: 128668
Rover Accuracy (m): 0.176, SIV: 32 GNSS State: 3D Fix
FreeHeap: 81884 / HeapLowestPoint: 79960 / LargestBlock: 3997684 / Used PSRAM: 128668
Pushing 528 bytes of RXM-PMP data to GNSS
Log file size: 95934 - Generation rate: 3.5kB/s
FreeHeap: 81884 / HeapLowestPoint: 79960 / LargestBlock: 3997684 / Used PSRAM: 128668
NOT pushing 279 bytes from /pp/Lb/eu topic to GNSS due to priority
NOT pushing 1674 bytes from /pp/Lb/eu topic to GNSS due to priority
NOT pushing 1332 bytes from /pp/Lb/eu topic to GNSS due to priority
Pushing 528 bytes of RXM-PMP data to GNSS
```

Happy days!

PaulZC, 2024-07-02:
Fixed in #401

commit d95eefe6 (2024-07-03): Merge pull request #401 from sparkfun/Fix_#363

Fix #363
closed 2024-07-08 by nseidle