sparkfun/SparkFun_RTK_Everywhere_Firmware #365: No logging icon when SD present

nseidle, 2024-06-19:
To replicate:

* Insert SD card
* Factory reset device
* Wait for log file creation

I see the log file created but the icon on the display is the circular 'I'm alive' icon. I expect to see the standard set of Page, P, and C icons correlating to default (Page), NMEA+RAWX+SFRBX (P), and Custom (C) message logging.

      No GNSS date/time available for system RTC.
      Rover Accuracy (m): 429496.719, SIV: 0 GNSS State: No Fix
      System time set to: Wednesday, June 19 2024 21:19:32
      [ 14935][D][NVM.ino:155] recordSystemSettingsToFileSD(): Removing from SD: /SFE_EVK_Settings_0.txt
      [ 14986][D][NVM.ino:174] recordSystemSettingsToFileSD(): Settings recorded to SD: /SFE_EVK_Settings_0.txt
      Log file name: /SFE_EVK_240619_211932.ubx
      Rover Accuracy (m): 429496.719, SIV: 0 GNSS State: No Fix
      Rover Accuracy (m): 429496.719, SIV: 0 GNSS State: No Fix
      Rover Accuracy (m): 429496.719, SIV: 0 GNSS State: No Fix
      STATE_ROVER_NO_FIX --> STATE_ROVER_FIX, 2024-06-19 21:19:38.179
      Rover Accuracy (m): 8.374, SIV: 5 GNSS State: 3D Fix

PaulZC, 2024-06-20:
I can't replicate this...

The only way this can happen is if ```online.logging``` is false or if ```logIncreasing``` is false.

Were you possibly in NTP mode? In that mode, you will only see the "N" logging icon if NTP requests are arriving and ```ntpLogIncreasing``` is true. Otherwise you get the circle of life.

PaulZC, 2024-06-20:
It can take up to 5s for the icon to update... Maybe you didn't wait long enough?

If you can get it to do it again, please turn on "5) Print Log File Status" from the Debug Hardware menu and check for "No increase in file size".

nseidle, 2024-06-20:
Interesting, that was not my experience. Many minutes passed without a icon change. I'll test again next week.

nseidle, 2024-06-26:
This was a hardware issue. I was using v1.0 PCB when I should have been on v1.1. Works great on correct hardware.

closed 2024-06-26 by nseidle