sparkfun/SparkFun_RTK_Everywhere_Firmware #366: IO 26 being controlled at startup but not set as output

nseidle, 2024-06-19:
To replicate:

* Factory reset
* Look at startup output

        [  1025][D][esp32-hal-adc.c:305] __analogReadMilliVolts(): Creating cali handle for ADC_0
        [  1033][D][Begin.ino:66] identifyBoard(): Board ADC ID (mV): 3011
        [  1074][E][esp32-hal-gpio.c:166] __digitalWrite(): IO 26 is not set as GPIO.
        LittleFS Started
        [  1082][E][vfs_api.cpp:99] open(): /littlefs/configureViaEthernet.txt does not exist, no permits for creation
        [  1094][E][vfs_api.cpp:99] open(): /littlefs/profileNumber.txt does not exist, no permits for creation
        [  1103][D][NVM.ino:1288] loadProfileNumber(): profileNumber.txt not found
        [  1115][D][NVM.ino:1306] loadProfileNumber(): Using profile #0
        [  1122][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_Settings_0.txt does not exist, no permits for creation
        PSRAM Size (bytes): 4194304

See `__digitalWrite(): IO 26 is not set as GPIO.` Not a big deal but something we should look into.

PaulZC, 2024-06-20:
Fixed in #375

nseidle, 2024-07-08:
This is in release [v1.3](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.3).

closed 2024-07-08 by nseidle