sparkfun/SparkFun_RTK_Everywhere_Firmware #370: (EVK) Firmware update fails over WiFi

nseidle, 2024-06-19:
Likely related to #369 .

To replicate:

* Confirm ethernet is not plugged in
* Factory reset
* Turn off Ethernet/WiFi failover (currently TCP menu 7, f)
* Set default network to WiFi (currently TCP menu 7, d)
* Enter WiFi credentials (6)
* Exit menus to save settings
* Update firmware (f, c)

Device fails to even start WiFi. It looks like it's trying to use ethernet.

      Current firmware version: d99.99
      Checking to see if an update is available from https://raw.githubusercontent.com/sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/main/RTK-Everywhere-Firmware.json
      [105807][D][HTTPClient.cpp:241] beginInternal(): unexpected protocol: https, expected http
      [105822][D][HTTPClient.cpp:281] beginInternal(): protocol: https, host: raw.githubusercontent.com port: 443 url: /sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/main/RTK-Everywhere-Firmware.json
      [105840][D][HTTPClient.cpp:552] sendRequest(): request type: 'GET' redirCount: 0
      
      [105847][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'raw.githubusercontent.com' with error '-54'
      [105857][E][ssl_client.cpp:116] start_ssl_client(): connect on fd 48, errno: 118, "Host is unreachable"
      [105867][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.
      [105874][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
      [105882][D][HTTPClient.cpp:1083] connect(): failed connect to raw.githubusercontent.com:443
      [105890][W][HTTPClient.cpp:1399] returnError(): error(-1): connection refused
      [105897][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
      [105906][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
      [105914][D][HTTPClient.cpp:379] disconnect(): tcp is closed
      
      Firmware server not available
      Failed to get version number from server.

PaulZC, 2024-07-07:
I'm seeing the same thing when using the Web config on WiFi - in AP mode:

```
settingName: wifiNetwork_0SSID value: VM4924742
settingName: wifiNetwork_0Password value: s8qtDpfnqgqh
settingName: wifiNetwork_1SSID value:
settingName: wifiNetwork_1Password value:
settingName: wifiNetwork_2SSID value:
settingName: wifiNetwork_2Password value:
settingName: wifiNetwork_3SSID value:
settingName: wifiNetwork_3Password value:
settingName: enableRCFirmware value: true
settingName: checkNewFirmware value: 1
Checking for new OTA Pull firmware
sendStringToWebsocket: checkingNewFirmware,1,
Current firmware version: d99.99-Jul  7 2024
Checking to see if an update is available from https://raw.githubusercontent.com/sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/main/RTK-Everywhere-RC-Firmware.json
[628423][V][HTTPClient.cpp:230] beginInternal(): url: https://raw.githubusercontent.com/sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/main/RTK-Everywhere-RC-Firmware.json
[628448][D][HTTPClient.cpp:241] beginInternal(): unexpected protocol: https, expected http
[628456][V][HTTPClient.cpp:230] beginInternal(): url: https://raw.githubusercontent.com/sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/main/RTK-Everywhere-RC-Firmware.json
[628471][D][HTTPClient.cpp:281] beginInternal(): protocol: https, host: raw.githubusercontent.com port: 443 url: /sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/main/RTK-Everywhere-RC-Firmware.json
[628489][D][HTTPClient.cpp:552] sendRequest(): request type: 'GET' redirCount: 0

[628497][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
[628503][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'raw.githubusercontent.com' with error '-54'
[628513][V][ssl_client.cpp:68] start_ssl_client(): Free internal heap before TLS 73412
[628520][V][ssl_client.cpp:75] start_ssl_client(): Starting socket (domain 2)
[628528][E][ssl_client.cpp:116] start_ssl_client(): connect on fd 53, errno: 118, "Host is unreachable"
[628537][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.
[628544][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
[628552][V][ssl_client.cpp:362] stop_ssl_socket(): Cleaning SSL connection.
[628559][D][HTTPClient.cpp:1083] connect(): failed connect to raw.githubusercontent.com:443
[628568][W][HTTPClient.cpp:1399] returnError(): error(-1): connection refused
[628575][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[628583][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[628591][D][HTTPClient.cpp:379] disconnect(): tcp is closed

[628597][V][ssl_client.cpp:362] stop_ssl_socket(): Cleaning SSL connection.
[628603][V][ssl_client.cpp:362] stop_ssl_socket(): Cleaning SSL connection.
Firmware server not available
Sending error to AP config page
sendStringToWebsocket: newFirmwareVersion,ERROR,
Sending receipt confirmation of settings
sendStringToWebsocket: confirmDataReceipt,1,
[628626][D][NVM.ino:155] recordSystemSettingsToFileSD(): Removing from SD: /SFE_EVK_Settings_0.txt
[628732][D][NVM.ino:174] recordSystemSettingsToFileSD(): Settings recorded to SD: /SFE_EVK_Settings_0.txt
[628759][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
[628982][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
sendStringToWebsocket: geodeticLat,54.86654214,geodeticLon,-1.42846185,geodeticAlt,153.582,ecefX,3677698.704,ecefY,-91709.042,ecefZ,5192973.427,batteryIconFileName,src/BatteryBlank.png,batteryPercent, ,
Provisioning: *PROVISIONING_NOT_STARTED
Provisioning: *PROVISIONING_NOT_STARTED
```

Checking STN mode now...

PaulZC, 2024-07-07:
OK. Updating from web config when in STN (WiFi) mode _does_ work. Because WiFi is already up and running when connecting to the SparkFun repo.

The solution will be to stall the HTTPClient until WiFi has had time to start. This almost certainly needs to be integrated with the new HTTP_Client network user?

PaulZC, 2024-07-07:
There's a bug in ```networkGetActiveType```. ```network``` is never initialized, making type corrupt. Fix coming...

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/d95eefe61b78b07b6af05b492d1c1fb86aa5e43a/Firmware/RTK_Everywhere/Network.ino#L485-L488

PaulZC, 2024-07-07:
Resolved by #411

PaulZC, 2024-07-07:
This issue isn't _fully_ resolved by #411 

With the web config in AP mode:
The firmware version check succeeds, but the firmware update fails with ```OTA Update: Firmware server not available```.

(This requires simultaneous AP and STN. But we know WiFi is capable of that. Investigating...)

PaulZC, 2024-07-07:
I found a nasty flaw in ```wifiConnect```. It's been there for quite a while. Embarrassing really... Fixed in #412

PaulZC, 2024-07-07:
This is resolved...

commit 09e81352 (2024-07-07): Fix error in networkGetActiveType - resolves #370
commit 3fd8fa5c (2024-07-07): Fix error in networkGetActiveType - resolves #370
commit 38281344 (2024-07-08): Merge pull request #411 from sparkfun/Fix_#370

Fix error in networkGetActiveType - resolves #370
closed 2024-07-08 by nseidle