sparkfun/SparkFun_RTK_Everywhere_Firmware #371: (EVK) NTP reports UDP error

nseidle, 2024-06-19:
To replicate:

* Confirm ethernet is connected
* Factory reset
* Enter into NTP mode from system menu

A non-stop error is reported. The display looks pretty good. 

      [ 23006][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23013][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23020][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23027][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23034][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23041][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23048][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23055][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23062][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23069][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23076][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9
      [ 23083][E][NetworkUdp.cpp:309] parsePacket(): could not receive data: 9

PaulZC, 2024-07-05:
Yeah... NTP was very broken. Resolved in #404

nseidle, 2024-07-08:
This is in release [v1.3](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.3).

PR #404 (2024-07-08): NTP updates
Resolves #371

NTP mode is now working nicely on EVK Ethernet. Tested with NTP on Raspberry Pi.

I struggled with ```ntpServer->parsePacket();``` and ```ntpServer->available()``` for a while. It wasn't capturing the Raspberry Pi NTP requests. The solution was to call ```ntpServer->clear();``` after each check. This ensures the internal ```rx_buffer``` is deleted correctly. Without this ```parsePacket``` returns 0 and won't receive the next packet.
closed 2024-07-08 by nseidle