sparkfun/SparkFun_RTK_Everywhere_Firmware #395: Tilt with Surpad(app) doesn’t work well

APOS80, 2024-06-28:
### Subject of the issue
Describe your issue here. Additionally, screenshots are easy to paste into github.

Everything works well with SW Maps but not using tilt with Surpad.

Without tilt it works well but when enabling tilt Surpad thinks it’s thousands of kilometres away, but the elevation is correct.

It makes me think that something changes with the NMEA message when enabling tilt.

### Your workbench
* What version of RTK firmware are you running? This can be found when the serial menu is opened (also in the settings.txt file, and in the serial output at power up).
Torch
June 27 binary
* What radios are you using: Bluetooth, WiFi, and/or ESP-Now? What app are you using to connect over Bluetooth? Are you transmitting NTRIP back to the device?
Bluetooth with Ntrip via phone
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post any log files from serial output that may have been generated.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

APOS80, 2024-06-29:
Without tilt
[image]
With tilt
[image]

Hope that helps

nseidle, 2024-06-30:
Thanks for reporting! I've got a demo version of Surpad working. 

0.11m to 6555559m is a bit of a problem!

> It makes me think that something changes with the NMEA message when enabling tilt.

Agreed. I suspect Surpad is parsing the NMEA in a different way. We are outputting (we believe) fully formed NMEA but there's always small changes. And different parsers can withstand varying levels of difference. AFAIK the lat/lon/alt NMEA sentences we output during tilt compensation is the same length and indistinguishable from non-tilt NMEA but obviously something is amiss.

Unfortunately I am out on vacation for a bit. I hope this doesn't block you from work for too long. I know you are capable of compiling the source so please poke at the problem if you have the desire.

APOS80, 2024-06-30:
I’m also going away for a while.

Sadly I haven’t solved the compiling part at this moment.

APOS80, 2024-07-09:
Back from the tripp?

APOS80, 2024-07-11:
I wonder if it’s as simple as a LF instead of a CR character.

nseidle, 2024-07-21:
I've tried to replicate this a few times and can't yet. What version of Surpad are you using? What view are you in? 

I'm using Surpad v20240612, with Everywhere firmware v1.3, using the Point Survey mode. 

[image]

[image]

Data in and out of tilt compensation mode seem to work just fine. 

I know there is a newer version of Surpad but my trial version is expired so I haven't been able to test it yet. 

Also, in the release candidate we've removed an [extra null terminator character](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commit/3265f59f0477c46f1fd2de2cd8c48e7a70058e0e) from the output. But I can't say it fixed anything until I can replicate the problem.

APOS80, 2024-07-22:
I can try the release candidate and see if it works.

nseidle, 2024-07-22:
I was testing with v1.3 and I did not detect a problem.

APOS80, 2024-07-22:
I was in the measuring view

APOS80, 2024-07-23:
Surpad v4.2.220626.232644

nseidle, 2024-07-23:
I'm on v4.2.240104.164515. You might try upgrading. I'll see if I can find an older version of Surpad but after a quick search, it doesn't look possible.

APOS80, 2024-07-24:
I’m unable to upgrade my version as of now.

But are you sure the NMEA messages is correct after going into tilt?

APOS80, 2024-09-01:
Problem persists in v1.4

nseidle, 2024-09-16:
Everything is working for us. What version of SurPad are you using?

> But are you sure the NMEA messages is correct after going into tilt?

Yes.

nseidle, 2025-05-21:
Closing old issue due to lack of activity.

closed 2025-05-21 by nseidle