sparkfun/SparkFun_RTK_Everywhere_Firmware #409: (EVK) PointPerfect key update (HTTP_Client) fails on WiFi

PaulZC, 2024-07-06:
EVK running the combined branch release_candidate

Disconnect Ethernet
Reset to factory defaults - to ensure Ethernet / WiFi Failover is enabled
Provide credentials for one WiFi network
Set the PointPerfect geographical region
Enable PointPerfect corrections
Exit the menus

The firmware gives up on Ethernet and failover to WiFi
WiFi connects, bu then immediately disconnects without key retrieval via the HTTP_Client 
I think the ```Network shutting down WiFi, no users``` is probably critical? Is the HTTP_Client User being registered correctly?

```
Network: Ethernet connection timed out
networkStop called by C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere\Network.ino 1183
Network layer stopping
Network State: NETWORK_STATE_CONNECTING --> NETWORK_STATE_OFF
Network layer restarting
Network failover: Ethernet-->WiFi
Network delaying 15 seconds before connection
Network State: NETWORK_STATE_OFF --> NETWORK_STATE_DELAY
Log file size: 4458545 - Generation rate: 3.6kB/s
Rover Accuracy (m): 0.137, SIV: 32 GNSS State: 3D Fix
Rover Accuracy (m): 0.138, SIV: 32 GNSS State: 3D Fix
Rover Accuracy (m): 0.138, SIV: 32 GNSS State: 3D Fix
*WIFI_STATE_OFF
Log file size: 4476696 - Generation rate: 3.6kB/s
Rover Accuracy (m): 0.138, SIV: 32 GNSS State: 3D Fix
Rover Accuracy (m): 0.139, SIV: 32 GNSS State: 3D Fix
Log file size: 4494753 - Generation rate: 3.6kB/s
Rover Accuracy (m): 0.138, SIV: 32 GNSS State: 3D Fix
Rover Accuracy (m): 0.138, SIV: 32 GNSS State: 3D Fix
networkTypeUpdate, network->requestedNetwork: WiFi --> WiFi
Network starting WiFi
Starting WiFi
WIFI_STATE_CONNECTING
FreeHeap: 83176 / HeapLowestPoint: 58192 / LargestBlock: 4063220 / Used PSRAM: 74836
Network State: NETWORK_STATE_DELAY --> NETWORK_STATE_CONNECTING
[1302235][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 9 - WIFI_READY
ETH Unknown event: 9
Connecting WiFi... [1302246][I][WiFiMulti.cpp:86] addAP(): [WIFI][APlistAdd] add SSID: VM4924742
[1302247][V][STA.cpp:184] _onStaEvent(): STA Started
[1302260][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 11 - STA_START
[1302267][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
ETH Unknown event: 11
[1313243][V][WiFiGeneric.cpp:77] _arduino_event_cb(): SCAN Done: ID: 133, Status: 0, Results: 15
[1313252][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 10 - SCAN_DONE
ETH Unknown event: 10
[1313260][I][WiFiMulti.cpp:129] run(): [WIFI] scan done
[1313267][I][WiFiMulti.cpp:134] run(): [WIFI] 15 networks found
[1313273][V][WiFiMulti.cpp:175] run(): known ssid: VM4924742, has failed: no
[1313280][V][WiFiMulti.cpp:208] run(): rssi_scan: -66, bestNetworkDb: -2147483648
[1313287][V][WiFiMulti.cpp:211] run(): best network is now: VM4924742
[1313294][D][WiFiMulti.cpp:228] run():  --->   0: [1][18:35:D1:9D:ED:09] VM4924742 (-66) (*) (visible)
[1313303][D][WiFiMulti.cpp:233] run():         1: [6][58:CB:52:D8:5E:1E] VM6184319 (-67) (*) (visible)
[1313312][D][WiFiMulti.cpp:233] run():         2: [6][40:0D:10:48:45:51] VM6184319 (-70) (*) (visible)
[1313321][D][WiFiMulti.cpp:233] run():         3: [1][58:CB:52:D8:49:F7] VM6184319 (-80) (*) (visible)
[1313331][D][WiFiMulti.cpp:233] run():         4: [11][7A:83:94:6B:6D:FC] EE WiFi (-81) ( ) (visible)
[1313340][D][WiFiMulti.cpp:233] run():         5: [6][C0:05:C2:46:C3:59] VM1245360 (-83) (*) (visible)
[1313349][D][WiFiMulti.cpp:233] run():         6: [11][8C:83:94:6B:6D:FB] BT-XPFH75 (-83) (*) (visible)
[1313358][D][WiFiMulti.cpp:233] run():         7: [6][04:95:E6:62:FF:D9] NOVA (-85) (*) (visible)
[1313367][D][WiFiMulti.cpp:233] run():         8: [11][D4:35:1D:17:49:49] vodafone174949 (-89) (*) (visible)
[1313377][D][WiFiMulti.cpp:233] run():         9: [1][08:B4:B1:66:EF:FD] VM6184319 (-91) (*) (visible)
[1313386][D][WiFiMulti.cpp:233] run():         10: [11][98:AA:FC:36:F8:2B] Wi-Fi Socket (-92) ( ) (visible)
[1313395][D][WiFiMulti.cpp:233] run():         11: [6][18:35:D1:71:EE:D1] VM2041210 (-93) (*) (visible)
[1313405][D][WiFiMulti.cpp:233] run():         12: [11][9C:31:C3:09:B0:42] SKYCEIY6 (-94) (*) (visible)
[1313414][D][WiFiMulti.cpp:233] run():         13: [11][6C:A0:B4:84:68:2A] SKYCEIY6 (-95) (*) (visible)
[1313423][D][WiFiMulti.cpp:233] run():         14: [1][08:B4:B1:66:ED:5E] VM6184319 (-97) (*) (visible)
[1313432][V][WiFiMulti.cpp:239] run(): foundCount = 1, failCount = 0
[1313439][I][WiFiMulti.cpp:249] run(): [WIFI] Connecting BSSID: 18:35:D1:9D:ED:09 SSID: VM4924742 Channel: 1 (-66)
[1313449][W][STA.cpp:533] disconnect(): STA already disconnected.
[1313525][V][STA.cpp:204] _onStaEvent(): STA Connected: SSID: VM4924742, BSSID: 18:35:d1:9d:ed:09, Channel: 1, Auth: WPA2_PSK
[1313537][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 13 - STA_CONNECTED
[1313545][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 13 - STA_CONNECTED
ETH Unknown event: 13
[1313751][V][NetworkInterface.cpp:66] _onIpEvent(): sta Got New IP: 192.168.0.90 MASK: 255.255.255.0 GW: 192.168.0.1
[1313761][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 16 - STA_GOT_IP
[1313769][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 16 - STA_GOT_IP
[1313776][V][STA.cpp:169] _onStaArduinoEvent(): STA IP: 192.168.0.90, MASK: 255.255.255.0, GW: 192.168.0.1
ETH Unknown event: 16
[1313786][I][WiFiMulti.cpp:272] run(): [WIFI] Connecting done.
[1313793][D][WiFiMulti.cpp:273] run(): [WIFI] SSID: VM4924742
[1313799][D][WiFiMulti.cpp:274] run(): [WIFI] IP: 192.168.0.90
[1313805][D][WiFiMulti.cpp:275] run(): [WIFI] MAC: 18:35:D1:9D:ED:09
[1313812][D][WiFiMulti.cpp:276] run(): [WIFI] Channel: 1
[1313818][D][WiFiMulti.cpp:334] resetFails(): [WIFI] Resetting failure flags

WIFI_STATE_CONNECTED
Network connected to WiFi
Network State: NETWORK_STATE_CONNECTING --> NETWORK_STATE_IN_USE
WiFi 'VM4924742' IP address: 192.168.0.90, RSSI: -65
Rover Accuracy (m): 0.138, SIV: 32 GNSS State: 3D Fix
Log file size: 4546381 - Generation rate: 10.3kB/s
*WIFI_STATE_CONNECTED
Network shutting down WiFi, no users
networkStop called by C:\Users\pc235\Documents\GitHub\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere\Network.ino 1236
Network stopping WiFi
WIFI_STATE_OFF
E (1325945) wifi:NAN WiFi stop
[1314340][V][STA.cpp:214] _onStaEvent(): STA Disconnected: SSID: VM4924742, BSSID: 18:35:d1:9d:ed:09, Reason: 8
[1314356][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 14 - STA_DISCONNECTED
[1314357][V][STA.cpp:188] _onStaEvent(): STA Stopped
[1314369][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 14 - STA_DISCONNECTED
[1314377][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
ETH Unknown event: 14
[1314384][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 12 - STA_STOP
[1314393][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 12 - STA_STOP
ETH Unknown event: 12
[1314405][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 8 - WIFI_OFF
ETH Unknown event: 8
WiFi Stopped
FreeHeap: 83144 / HeapLowestPoint: 58192 / LargestBlock: 4063220 / Used PSRAM: 74836
Network layer stopping
Network State: NETWORK_STATE_IN_USE --> NETWORK_STATE_OFF
Rover Accuracy (m): 0.139, SIV: 32 GNSS State: 3D Fix
Rover Accuracy (m): 0.139, SIV: 32 GNSS State: 3D Fix
Log file size: 4564294 - Generation rate: 3.6kB/s
```

PaulZC, 2024-07-06:
Ah. OK. I think I understand what's going wrong here.

The HTTP_Client (Provisioning) has already timed out and is in state ```Provisioning: *PROVISIONING_WAIT_ATTEMPT```

This happened because Ethernet wasn't available and it took a long time for the failover to WiFi to happen.

What to do? Extend the Provisioning timeout? Or have the failover reset the Provisioning state?

Trying the timeout first...

PaulZC, 2024-07-06:
Extending the Provisioning timeout from 20s (!) to 2mins allowed the provisioning to take place successfully over WiFi (after Ethernet failed first).

I'm now expecting to see PP correction data being delivered over WiFi, but that's not happening. I think we possibly have a similar issue on MQTT due to ```MQTT client data timeout```.

PaulZC, 2024-07-06:
Yeah. ```mqttClientLastDataReceived``` wasn't being initialized correctly - when the MQTT connection is made - and was going into immediate timeout.

I'll add these two fixes to the relevant branches.

This is resolved...

nseidle, 2024-07-08:
This is in release [v1.3](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.3).

closed 2024-07-08 by nseidle