sparkfun/SparkFun_RTK_Everywhere_Firmware #418: Display correction source

nseidle, 2024-07-08:
I'm currently using an EVK. The device is in RTK Fix and has 14mm HPA, but it's not immediately clear to me how. It doesn't appear to be L-Band or Bluetooth. It must be IP via PointPerfect, but 14mm is stellar. Ah, no, it was an NTRIP client I didn't realize was on.

Should we periodically print the source being used? 

Should we display an icon showing the source? This could get messy on the screen, but we have plenty of room.

PaulZC, 2024-07-09:
I like the idea of displaying an icon. And we have room for it on the bottom row - between the full IP address and the logging icon. Let's identify the most recently pushed corrections source:

* CORR_RADIO_EXT - use DynamicModel_1_Portable ?
  * [image]
* CORR_ESPNOW - use ESPNOW_Symbol_3 ?
  * [image]
* CORR_RADIO_LORA - new "[RKO Pictures](https://en.wikipedia.org/wiki/RKO_Pictures#/media/File:RKO_Radio_Pictures_transmitter_ident.jpg)" antenna mast icon?
  * [image]
* CORR_BLUETOOTH - use BT_Symbol ?
  * [image]
* CORR_TCP - this is NTRIP, so maybe the SNIP logo?
  * [image]
* CORR_LBAND - use SIV_Antenna ?
  * [image]
* CORR_IP - this is PointPerfect so maybe a simplified 
  * [image]

nseidle, 2024-07-09:
WFM! How to signify the word 'source'? Maybe just put these icon types next to the current down arrow? I believe the down arrow only blinks with Bluetooth and WiFi traffic. If we can expand this to the other sources, I think that would be great.

nseidle, 2024-08-27:
Lee has laid the groundwork [here](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/pull/452/files). Priorities menu is now showing who/what is active. `correctionGetSourceName()` is helpful.

[image]

PaulZC, 2024-10-29:
The changes for EVK are in #492

PaulZC, 2024-10-29:
Full set:

    CORR_RADIO_EXT
    CORR_ESPNOW
    CORR_RADIO_LORA
    CORR_BLUETOOTH

[image]

    CORR_USB
    CORR_TCP
    CORR_LBAND
    CORR_IP

[image]

nseidle, 2024-10-29:
Beauty!

PaulZC, 2024-11-07:
On 64x48, I went with blinking the source over the Horizontal accuracy. It only happens when you are in RTK Float / Fix. The normal dual-crosshair is shown for 6 out of 8 of the 0.5 second intervals. The correction source is shown for the other two. For Float, the source blinks to match the blinking crosshair. Let me know if this hits the spot...

[image]
[image]

nseidle, 2024-11-07:
This looks great to me!

nseidle, 2024-12-18:
This was rolled out in production release v2.0.

closed 2024-12-18 by nseidle