sparkfun/SparkFun_RTK_Everywhere_Firmware #419: EVK retrevies keys twice

nseidle, 2024-07-08:
Using an EVK on d1.3:

* Connect ethernet
* No SD card
* Factory reset device
* Wait for 3D Fix (~10 seconds post startup)
* Enable PointPerfect
* Exit menus and watch

What I see:
* Getting Keys is displayed. Ethernet fails to get keys every time, so 7 second wait.
* Getting Keys is displayed. Keys are obtained. 48 days until expiry shown.
* L-Band Config is displayed.
* Getting Keys is displayed. Keys are obtained a second time. 48 days until expiry shown.
* L-Band Config is displayed again.

Below is the terminal output. I can accept the first Ethernet failure as it seems this is the network layer turning on (not entirely sure). But the duplicate key obtain and L-Band config could be improved.

    Menu: PointPerfect Corrections
    Days until keys expire: No keys
    1) PointPerfect Corrections: Enabled
    2) Toggle Auto Key Renewal: Enabled
    3) Request Key Update: Requested
    4) Use localized distribution: Disabled
    a) Use AssistNow: Disabled
    c) Clear the Keys
    i) Show device ID
    k) Manual Key Entry
    g) Geographic Region: US
    x) Exit
    x
    
    SparkFun RTK EVK d1.3
    ** Bluetooth SPP and BLE broadcasting as: EVK Rover-8EFA **
    Menu: Main
    1) Configure GNSS Receiver
    2) Configure GNSS Messages
    3) Configure Base
    4) Configure Ports
    5) Configure Logging
    6) Configure WiFi
    7) Configure TCP/UDP
    e) Configure Ethernet
    f) Firmware Update
    i) Configure Corrections Priorities
    n) Configure NTP
    p) Configure PointPerfect
    r) Configure Radios
    s) Configure System
    t) Configure Instrument Setup
    u) Configure User Profiles
    +) Enter Command line mode
    x) Exit
    x
    [ 46909][D][States.ino:729] requestChangeState(): Requested System State: 0
    [ 47063][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
    [ 47271][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
    Rover Accuracy (m): 0.503, SIV: 23 GNSS State: 3D Fix
    STATE_ROVER_FIX --> STATE_ROVER_NOT_STARTED, 2024-07-08 19:05:18.177
    [ 47661][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
    [ 47868][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
    STATE_ROVER_NOT_STARTED --> STATE_ROVER_NO_FIX, 2024-07-08 19:05:19.243
    [ 48414][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_certificate_0.txt does not exist, no permits for creation
    [ 48427][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_privateKey_0.txt does not exist, no permits for creation
    STATE_ROVER_NO_FIX --> STATE_ROVER_FIX, 2024-07-08 19:05:19.322
    Ethernet IP address: 10.7.0.31
    [ 48483][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_certificate_0.txt does not exist, no permits for creation
    [ 48511][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
    [ 48717][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
    Network delaying 7 seconds before connection
    HTTP Client start
    Rover Accuracy (m): 0.468, SIV: 23 GNSS State: 3D Fix
    Rover Accuracy (m): 0.404, SIV: 23 GNSS State: 3D Fix
    Rover Accuracy (m): 0.384, SIV: 23 GNSS State: 3D Fix
    Rover Accuracy (m): 0.362, SIV: 23 GNSS State: 3D Fix
    Ethernet IP address: 10.7.0.31
    [ 58307][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
    [ 58349][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 52.19.81.15
    [ 59041][D][HTTPClient.cpp:275] beginInternal(): switching host from '' to 'api.thingstream.io'. disconnecting first
    [ 59052][D][HTTPClient.cpp:366] disconnect(): tcp stop
    [ 59061][D][HTTPClient.cpp:281] beginInternal(): protocol: https, host: api.thingstream.io port: 443 url: /ztp/pointperfect/credentials
    Rover Accuracy (m): 0.350, SIV: 23 GNSS State: 3D Fix
    [ 59128][D][HTTPClient.cpp:552] sendRequest(): request type: 'POST' redirCount: 0
    
    [ 59136][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
    [ 59144][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 52.19.81.15
    [ 59844][D][HTTPClient.cpp:1090] connect():  connected to api.thingstream.io:443
    [ 60156][D][HTTPClient.cpp:1235] handleHeaderResponse(): code: 200
    [ 60162][D][HTTPClient.cpp:1238] handleHeaderResponse(): size: 5667
    [ 60168][D][HTTPClient.cpp:596] sendRequest(): sendRequest code=200
    
    [ 60195][D][HTTPClient.cpp:364] disconnect(): tcp keep open for reuse
    [ 60212][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_certificate_0.txt does not exist, no permits for creation
    [ 60283][D][NVM.ino:1477] recordFile(): File recorded to LittleFS: /SFE_EVK_certificate_0.txt
    [ 60295][E][vfs_api.cpp:99] open(): /littlefs/SFE_EVK_privateKey_0.txt does not exist, no permits for creation
    [ 60471][D][NVM.ino:1477] recordFile(): File recorded to LittleFS: /SFE_EVK_privateKey_0.txt
    [ 60517][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
    [ 60711][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
    HTTP Client stopping
    Days until PointPerfect keys expire: 48
    Rover Accuracy (m): 0.321, SIV: 23 GNSS State: 3D Fix
    [ 65027][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
    [ 65233][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
    Ethernet IP address: 10.7.0.31
    [ 65278][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
    [ 65484][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
    [ 67640][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 34.241.240.139
    [ 68812][D][ssl_client.cpp:322] ssl_starttls_handshake(): Protocol is TLSv1.2 Ciphersuite is TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
    [ 68824][D][ssl_client.cpp:324] ssl_starttls_handshake(): Record expansion is 29
    HTTP Client start
    Rover Accuracy (m): 0.315, SIV: 24 GNSS State: 3D Fix
    [ 69319][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
    [ 69522][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
    [ 69740][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 52.19.81.15
    [ 70456][D][HTTPClient.cpp:275] beginInternal(): switching host from '' to 'api.thingstream.io'. disconnecting first
    [ 70467][D][HTTPClient.cpp:366] disconnect(): tcp stop
    [ 70476][D][HTTPClient.cpp:281] beginInternal(): protocol: https, host: api.thingstream.io port: 443 url: /ztp/pointperfect/credentials
    [ 70546][D][HTTPClient.cpp:552] sendRequest(): request type: 'POST' redirCount: 0
    
    [ 70553][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
    [ 70562][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 52.19.81.15
    [ 71271][D][HTTPClient.cpp:1090] connect():  connected to api.thingstream.io:443
    [ 71570][D][HTTPClient.cpp:1235] handleHeaderResponse(): code: 200
    [ 71576][D][HTTPClient.cpp:1238] handleHeaderResponse(): size: 5667
    [ 71582][D][HTTPClient.cpp:596] sendRequest(): sendRequest code=200
    
    [ 71610][D][HTTPClient.cpp:364] disconnect(): tcp keep open for reuse
    [ 71660][D][NVM.ino:1465] recordFile(): Removing LittleFS: /SFE_EVK_certificate_0.txt
    [ 71720][D][NVM.ino:1477] recordFile(): File recorded to LittleFS: /SFE_EVK_certificate_0.txt
    [ 71775][D][NVM.ino:1465] recordFile(): Removing LittleFS: /SFE_EVK_privateKey_0.txt
    [ 71844][D][NVM.ino:1477] recordFile(): File recorded to LittleFS: /SFE_EVK_privateKey_0.txt
    Rover Accuracy (m): 0.292, SIV: 26 GNSS State: 3D Fix
    [ 71980][D][NVM.ino:204] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
    [ 72156][D][NVM.ino:216] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_EVK_Settings_0.txt
    HTTP Client stopping
    Days until PointPerfect keys expire: 48
    Rover Accuracy (m): 0.273, SIV: 26 GNSS State: 3D Fix
    Rover Accuracy (m): 0.267, SIV: 26 GNSS State: 3D Fix

PaulZC, 2024-07-09:
Humm... I think this was being caused by the RTC coming online 'late', and by some poor logic in ```updateProvisioning```.

Resolved in #422

closed 2024-08-29 by nseidle

nseidle, 2024-08-29:
Fix in place in [v1.4](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.4).

PR #422 (2024-07-09): Fix #419
This PR resolves #419 by improving ```updateProvisioning```:
* Restart from PROVISIONING_CHECK_REMAINING, not PROVISIONING_STARTING
* Avoid using ```lastKeyAttempt``` in PROVISIONING_WAIT_ATTEMPT as the RTC may come online _during_ this state
* Only update and save ```lastKeyAttempt``` when the RTC is online
* Add _ms and _millis units clarifiers
commit d554418f (2024-07-09): Merge pull request #422 from sparkfun/Fix_#419

Fix #419
commit 765fac4c (2024-07-09): Merge pull request #424 from sparkfun/Fix_#419

Fix #419
commit 1e1a3012 (2024-08-29): Merge pull request #423 from sparkfun/revert-422-Fix_#419

Revert "Fix #419"