sparkfun/SparkFun_RTK_Everywhere_Firmware #446: Not able to receive RTCM over BLE

nseidle, 2024-07-17:
Because of [this patch](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commit/94b188ef4af0a5f3c72362d95afaf368545d7bb7), and [related issue](https://github.com/avinabmalla/ESP32_BleSerial/issues/15), BLE RX (data from the phone to the RTK Device) is acting very oddly.

Almost no bytes (but some!) are getting through:

[image]

Using

    String value = pCharacteristic->getValue();

in BleSerial.cpp seems to be working for me. 

[image]

To test, I am using GNSS Master on Android because it seems to be the only NTRIP over BLE compatible with Avinab's BLESerial library (SW Maps on Android supports Generic BLE instrument, but that instrument has the NTRIP client disabled).

PaulZC, 2024-07-18:
Referencing your commit https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commit/59421084b3add61d1b730860f60203306a009d1b

PaulZC, 2024-07-18:
Please see https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commit/59421084b3add61d1b730860f60203306a009d1b#commitcomment-144371193

nseidle, 2024-08-29:
Fix in place in [v1.4](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v1.4).

closed 2024-08-29 by nseidle