sparkfun/SparkFun_RTK_Everywhere_Firmware #449: Add a unique device ID to Messages list

becorey, 2024-08-02:
### Subject of the issue
With an RTK EVK sending messages using TCP Client, there is currently no unique device ID available to identify the device. 
With a unique ID, it would enable the possibility of deploying a fleet of RTK EVK devices which all report data back to a central server, and keeping track of which device is associated with which incoming data.
The ID could be the UBX-SEC-UNIQID, the ESP32 MAC address, or some combination of those.
Reference to forum post [Unique device ID in RTK EVK messages?](https://community.sparkfun.com/t/unique-device-id-in-rtk-evk-messages/59680)

### Your workbench
* What version of RTK firmware are you running? 
* * Beta d1.4
* What radios are you using: Bluetooth, WiFi, and/or ESP-Now? What app are you using to connect over Bluetooth? Are you transmitting NTRIP back to the device? 
* * Wifi, using TCP Client. Yes NTRIP to the device from PointPerfect.
* Are there any additional details that may help us help you?
* * The RTK EVK is installed with a vehicle and powered by the vehicle's 12V battery. A 4G modem/router provides Wifi to the EVK. The EVK sends messages using TCP Client to a server which logs the messages to files. Data is collected for several hours each day to track the movements of the vehicle.

### Steps to reproduce
Tell us how to reproduce this issue. Please post any log files from serial output that may have been generated.
Put RTK EVK into Wifi Config mode, visit the config screen with a browser. Open the list of Messages available to send. There are many available but as far as I can tell none of them include a unique ID to identify the device that is sending messages.

### Expected behavior
Tell us what should happen
It could implement a similar approach as [SparkFun_RTK_Firmware #476](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/476#issuecomment-1567346348), adding a $GNTXT message which sends the GNSS unique chip ID and the last four digits of the ESP32 MAC address.
It could be sent periodically like any other message and likely would be set to a low frequency such as once every 20 fixes.

Thank you!

nseidle, 2024-08-09:
This is a great feature to add but will require a considerable amount of work to implement. We'll need to add a custom message generator which has no precedence. Please give us some time to consider how it might be added. In the mean time, if you have any ability to contribute to the codebase, we welcome PRs.

PaulZC, 2024-10-30:
I've been thinking about this:

Injecting GNTXT messages periodically into the message stream to the TCP server is ~difficult. I think we would need to change the way messages are parsed, and allow injection of GNTXT when the parser finds the end of a valid message. Maybe we could modify the parser so that it searches for GNGGA, and 'appends' the GNTXT to that? Messy, but, in my mind, it could work.

What would be _much_ easier, would be to send the GNTXT message once when the TCP client connects to the server. I.e. it is the first thing sent to the server, before the stream of other messages begins. We would need to add a ```client->write``` [here](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/4ce85f32f42975648255732faa12e4b2366a811d/Firmware/RTK_Everywhere/TcpClient.ino#L308).

Corey (@becorey ): would the easy option meet your needs?

PaulZC, 2024-10-30:
I.e. take [this code](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/40607b9cc08a84f523a227f6ab671c49953dfc3b/Firmware/RTK_Surveyor/menuMessages.ino#L597-L628) and change it to ```client->write```'s

And while we're in there, we should add the [brand and platform prefix](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/4ce85f32f42975648255732faa12e4b2366a811d/Firmware/RTK_Everywhere/menuMain.ino#L136-L139) to the [firmware version](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/40607b9cc08a84f523a227f6ab671c49953dfc3b/Firmware/RTK_Surveyor/menuMessages.ino#L606). The version date on its own isn't that useful...