sparkfun/SparkFun_RTK_Everywhere_Firmware #462: Multiple disconnects over Bluetooth with SWmaps

APOS80, 2024-09-01:
### Subject of the issue
Disconnects all the time in v1.4
SWMaps

### Your workbench
* What version of RTK firmware are you running? This can be found when the serial menu is opened 
Version 1.4
* What radios are you using: Bluetooth, WiFi, and/or ESP-Now? What app are you using to connect over Bluetooth? Are you transmitting NTRIP back to the device?
Bluetooth with ntrip
Android
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post any log files from serial output that may have been generated.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

APOS80, 2024-09-01:
Same with IPhone it seems

Viatrak, 2024-11-21:
RTK Torch,  iPad Pro M4 11 Inch, BLE, Firmware v1.4, 2Hz. 

Lots of disconnects.

I have since disabled most of the NMEA messages. 

I noticed that if I turn off the NTRIP client (stop sending corrections to the Torch) it no longer disconnects from bluetooth.

nseidle, 2024-12-19:
BLE connectivity has changed significantly in the recent RTK Everywhere v2.0 release. We believe this has been fixed.

Other changes pertinent to the BLE connectivity:

* ESP32 Arduino core update to 3.0.7
* BLESerial library update to 2.0.1
* SW Maps update to v3.0

While I was intermittently able to reproduce this issue using RTK Everywhere v1.4, it was not often seen. With the updates in place, I haven't seen a hiccup in a few days worth of testing. Closing for now. Please try v2.0 and let us know if the issue persists.

closed 2024-12-19 by nseidle