sparkfun/SparkFun_RTK_Everywhere_Firmware #480: WiFi Config - Base Configuration - NTRIP Server (1) - Save Configuration fails...

PaulZC, 2024-09-25:
Reported [here](https://portal.u-blox.com/s/question/0D5Oj00000Qu4oiKAB/i-would-like-to-transmit-rtcm-from-sparkfun-rtk-evk-via-the-ntrip-server) and [here](https://community.sparkfun.com/t/how-to-setup-rtcm-via-ntrip-tcp-for-precise-point-positioning-ppp/60341).

We have a gremlin in main.js when using ```checkElementString``` on the NTRIP Server (1) configuration:

[image]

Investigating...

PaulZC, 2024-09-25:
This issue has already been resolved in the release_candidate branch...

I've created a separate "hotfix" branch which includes: the fix for this issue; and increases the number of corrections sources to 8.

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commits/Hotfix_%23480/

PaulZC, 2024-09-25:
@nseidle : this is done. Please release an RC from the Hotfix branch when time permits.

PaulZC, 2024-09-25:
I suggest we **don't** merge the Hotfix branch into main. It is safe to do so, the changes should be identical to the changes already in release_candidate, but they'll probably appear as a conflict...

nseidle, 2024-09-25:
Nice fix! RC has been posted [Sep-25](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware_Binaries/blob/main/RTK_Everywhere_Firmware_RC-Sep_25_2024.bin).

nseidle, 2024-12-18:
This was rolled out in production release v2.0.

closed 2024-12-18 by nseidle