sparkfun/SparkFun_RTK_Everywhere_Firmware #505: Postcard: Base mode not showing SIV

nseidle, 2024-12-18:
When in Base mode, 0 SIV is reported. RTCM output looks like it is reduced to 25 bytes (not nearly enough).

PaulZC, 2024-12-19:
@nseidle : Could this be the same issue I'm seeing on Facet mosaic? If the BLE task is preventing the GNSS UART task from being serviced, could that explain loss of incoming SIV and RTCM data? Please try the same test - set Bluetooth Mode to Classic or Off - and see if that improves delivery of SIV and RTCM.

nseidle, 2024-12-19:
Different I believe. [The solution](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commit/1a0baed6ec73b093dae12ec4fa498a75b24a389d) is related to specifically how the LG290P reports SIV. Initially we were reporting SIV used in a fix, and in Base mode on the LG290P, there is no fix, so there's no SIV used in a fix.

PaulZC, 2024-12-19:
Thanks. Agreed. I'm struggling to find the root cause of the BLE-stopping-the-GNSS-UART-task on Facet mosaic. I don't see the same issue on Postcard. I do see the same ```[D][BluetoothSerial.cpp:792] waitForConnect(): connection closed!``` messages whenever BluetoothSerial is enabled (Dual or Classic). But I _think_ those are a red herring. The GNSS data keeps on arriving. I'll keep digging...

commit 1a0baed6 (2024-12-18): Use getSatellitesInViewCount()

Fix #505. In Base mode, getSatellitesInViewCount is correctly calculated where as getSatellitesUsedCount is always zero.
closed 2025-02-20 by nseidle