sparkfun/SparkFun_RTK_Everywhere_Firmware #513: No accuracy information available with SW Maps or GNSS Master (no NMEA GST)

mmoduu, 2025-01-13:
When connecting to RTK Postcard via Bluetooth with SW Maps or GNSS Master, there isn't accuracy information displayed, and GNSS Master even says "No NMEA GST". This does not change if I enable NTRIP client in the app (NTRIP client in the RTK Postcard does not work, I'll create a separate issue if I remember).

However, if I open the serial terminal, you can see status strings that specify the accuracy, so the data is there somewhere.

Weirdly, Quectel also confirm that they do not output that data currently (https://forums.quectel.com/t/the-lg290p-doesnt-seem-to-support-nmea-gst-message/39242), but is it possible that you could inject that data to the output?

nseidle, 2025-01-13:
Indeed, I've confirmed the GST message is not supported on the LG290P and requested it in a future firmware release. Quectel says maybe in a future firmware update but I haven't seen *any* firmware updates yet so who knows.

> is it possible that you could inject that data to the output?

Anything is possible but there's not so many hours available. We can generate NMEA easily enough but we don't yet have a scheduler to add messages to the outbound circular buffer (what you see in your GIS app). This feature would be related to the [unique ID messages request](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/issues/449). Same plumbing, different message to report.

nseidle, 2025-01-13:
The timing couldn't be better. I was just notified by Quectel that they released new firmware that supports GST. 

[Quectel_LG290P_Firmware_Release_V0104S.zip](https://github.com/user-attachments/files/18401943/Quectel_LG290P_Firmware_Release_V0104S.zip)

We'll get started implementing this.

nseidle, 2025-01-14:
GST is working on the Postcard in the release candidate. We have a bunch of docs to update but you should be able to update your LG290P firmware on Postcard with this:

Start the update procedure from QGNSS: 

[image]

Within 20 seconds, you have to reset the LG290P. Do this from **s**ystem->**h**ardware->**13**) LG290P reset for firmware update:

[image]

Firmware updates took me about 75 seconds.

Release candidate is scanning for old (< v4) and new (>=v4) firmware types and settings things up accordingly.

Note: GST is currently reporting single decimals so the accuracy during normal 3D Fix will look fine, but in RTK Fix mode, some GIS softwares (ie, SW Maps) will report an unknown lat/lon positional error because of this:

> $GNGST,190701.500,0.7,0.0,0.0,85.7,0.0,0.0,0.0*4F

Error in [GST](https://receiverhelp.trimble.com/alloy-gnss/en-us/NMEA-0183messages_GST.html) is reported as '0.0' when really it's 0.013 (13mm). Their binary protocol confirms this but the GST message is cutting off those all important decimals. I've pinged Quectel to see if there's a remedy.

mmoduu, 2025-01-14:
Thanks for the hard work on this!

I need to find an Windows PC first to try that firmware upgrade, I'll report my progress in about 20 hours...

nseidle, 2025-01-15:
Np! Glad it's coming together. Also, our [RTK loader](https://github.com/sparkfun/SparkFun_RTK_Firmware_Uploader) is available on Win/iOS/*inux.

mmoduu, 2025-01-19:
Hi!

Sorry for the long delay. I was able to flash the firmware (both ESP32 and LG290P) and see the results. Let's hope Quectel can fix that data problem soon.

ryanmurray, 2025-01-25:
Glad I wasn't the only one. I'm gonna do the firmware upgrade. The hdop and vdop I'm seeing with the L1/l2/L5 antenna far exceed what I saw with the facet making the postcard a winning combo if we can get hrms and vrms 

I use surpad but wasn't getting any RMS info so stumbled here

[image]

ryanmurray, 2025-01-25:
I did complete the v4 upgrade to the lgp290 and will see what survpad says tomorrow.

nseidle, 2025-02-05:
As mentioned, 

Note: GST is currently reporting single decimals so the accuracy during normal 3D Fix will look fine, but in RTK Fix mode, some GIS softwares (ie, SW Maps) will report an unknown lat/lon positional error because of this:

$GNGST,190701.500,0.7,0.0,0.0,85.7,0.0,0.0,0.0*4F

Error in [GST](https://receiverhelp.trimble.com/alloy-gnss/en-us/NMEA-0183messages_GST.html) is reported as '0.0' when really it's 0.013 (13mm). Their binary protocol confirms this but the GST message is cutting off those all important decimals. I've pinged Quectel to see if there's a remedy.

ryanmurray, 2025-02-05:
Yes I was able to get 0.000 on surpad so that's solved. If quectel can fix it this chip is phenomenal for surveying. I mean it still is by RMS would be nice to see.

ryanmurray, 2025-02-10:
Anyway to force minimum to equal 0.01 instead of rounding to 0.00? Field genius refuses to store points with no gst value

nseidle, 2025-02-10:
We could disable the GST sentence the module produces and produce our own GST sentence. Not something I really want to do since this is Quectel's fault. But if it's breaking compatibility with GIS software, I'll consider it.

ryanmurray, 2025-02-10:
Yeah basically on most apps it prevents you from storing a position
On surpad I could get around it but field genius won't let you override the hrms or vrms with 0

I've told quectel we really need more decimals but who knows how long it will take. 

The postcard would be a beautiful device if you could possibly fix this one little issue :D

nseidle, 2025-02-10:
> if you could possibly fix this one little issue

Oh, I've heard that one before! We'll do our best :)

mmoduu, 2025-02-10:
Should we just bombard Quectel forums? Nothing else we can do I guess?

ryanmurray, 2025-02-10:
> Should we just bombard Quectel forums? Nothing else we can do I guess?

I've been doing that and I think sparkfun has too.

nseidle, 2025-02-11:
Quectel told me end of Q2 for the next firmware update that fixes the malformed GST sentence so I'm not willing to wait.

I've got a work around in place. It's not completely accurate, but we've extended the GST sentence to include 3 decimals.

[image]

Above, GST with a RTK Fix that is waning. Note that the vertical error (that's the 0.066m that follows the two 0.033m fields) is basically made up. 0.033 is the LG290P's reported horizontal error for both the lat and lon error, and the 0.066 is double the horizontal error for the height error. This is just a rough hack until Quectel properly outputs the GST sentence. 

[image]

Above, SW Maps with a good RTK Fix.

While `horzAccuracy * 2` is not correct, it should be good enough to allow GIS software (looking at you Field Genius) to be able to take points.

I haven't tested this on anything but SW Maps so please test today's (Feb 10) RC and let me know.

nseidle, 2025-02-11:
Glad to hear it's working! I pushed another small change to prevent this GST patch from being applied unless RTK Fix/Float is happening. I'd rather trust the original GST if the HPA is >0.1m. It's only when we're in the mm range that we need the extra decimals.

nseidle, 2025-02-14:
Quectel delivers again! We were given v04 beta firmware that fixes the GST decimals to 3 digits. Please upgrade your LG290P to v04 Beta found [here](https://github.com/sparkfun/SparkFun_RTK_Postcard/tree/main/Firmware). I've reverted the changes in the RC of Feb 14 to remove our modification of the GST sentence and now it passes through unchanged. Please update the ESP32 (Postcard) [firmware](https://github.com/sparkfun/sparkFun_RTK_Everywhere_Firmware_binaries) to today or later RC.

Below, the GST sentence shows 1.119, 1.047, 2.018 correctly in three decimals.

[image]

smyrgl, 2025-02-15:
Loaded both the new firmware from Quectel and the RTK Everywhere (RC Feb 14th) and I can confirm that the issue seems to be resolved.  Nice job! 

[image]

mmoduu, 2025-02-17:
Thanks for all the reports! Currently I do not have access to Windows computer that could run QGNSS for updating the module firmware :)

Off-topic from the actual issue but.... @nseidle do you think it would be possible to have the GNSS module firmware upgrade process done with other tools?

nseidle, 2025-02-18:
Yes, Quectel does a nice job of [publishing how their bootloader works](https://docs.sparkfun.com/SparkFun_LG290P_Quadband_GNSS_RTK_Breakout/assets/component_documentation/Quectel_LG290P(03)_Firmware_Upgrade_Guide_V1.0.pdf) so, theoretically, any tool could be created. For now, we're going to leverage Quectel's QGNSS app, but if someone wanted to write a python script, they could.

We've talked internally about building these tools into the RTK Everywhere firmware itself, but:
* We're limited in flash space
* We don't know all the engines (UM980 bootloader is a mystery so we have to leverage their tool for now)
* GNSS receiver firmware upgrades are a lot less frequent than our updates so we tend to focus on our update tools

nseidle, 2025-02-18:
Closing for now.

closed 2025-02-18 by nseidle