sparkfun/SparkFun_RTK_Everywhere_Firmware #515: On Postcard, Ntrip Client does not send GGA messages to Caster

anathan, 2025-01-19:
With the LG290P on the postcard, GGA messages are not being sent via the embedded NTRIP client on the ESP32. Verified the device is connecting to the server successfully and is authenticating properly and that the LG290P has a valid position solution.

anathan, 2025-01-19:
example output with some debugging on:
``NTRIP Client connected to polaris.pointonenav.com:2101 via 192.168.0.161:52445 at 7:07:49
NTRIP Client: NTRIP_CLIENT_WAIT_RESPONSE --> NTRIP_CLIENT_CONNECTED
Rover Accuracy (m): 1.737, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 2.045, SIV: 11 GNSS State: 3D Fix
NTRIP Client state: NTRIP_CLIENT_CONNECTED
Correction Source: None
Rover Accuracy (m): 2.927, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 2.470, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 2.778, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 2.210, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 1.323, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 1.809, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 2.688, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 1.585, SIV: 11 GNSS State: 3D Fix
NTRIP Client state: NTRIP_CLIENT_CONNECTED
Correction Source: None
Rover Accuracy (m): 1.526, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 2.395, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 2.296, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 1.660, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 1.553, SIV: 11 GNSS State: 3D Fix
NTRIP Client timeout receiving data at 7:08:19
FreeHeap: 43112 / HeapLowestPoint: 41332 / LargestBlock: 1998836 / Used PSRAM: 89452
NTRIP Client: NTRIP_CLIENT_CONNECTED --> NTRIP_CLIENT_ON
NTRIP Client Connecting - polaris.pointonenav.com/polaris:2101 Last Uptime: 49 17:02:47.295 (Reconnects: 12)
NTRIP Client trying again in 40 minutes.
NTRIP Client: NTRIP_CLIENT_ON --> NTRIP_CLIENT_NETWORK_STARTED
FreeHeap: 43316 / HeapLowestPoint: 41332 / LargestBlock: 1998836 / Used PSRAM: 89524
NTRIP Client: NTRIP_CLIENT_NETWORK_STARTED --> NTRIP_CLIENT_NETWORK_CONNECTED
Rover Accuracy (m): 1.798, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 1.300, SIV: 11 GNSS State: 3D Fix
NTRIP Client state: NTRIP_CLIENT_NETWORK_CONNECTED
Correction Source: None
Rover Accuracy (m): 2.632, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 3.004, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 1.775, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 2.380, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 1.522, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 2.124, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 2.199, SIV: 11 GNSS State: 3D Fix
Rover Accuracy (m): 1.383, SIV: 11 GNSS State: 3D Fix
NTRIP Client state: NTRIP_CLIENT_NETWORK_CONNECTED
Correction Source: None
```

nseidle, 2025-01-20:
Indeed. Thanks for reporting! Working on it.

nseidle, 2025-01-20:
The GGA push to caster is now fixed in the RC. I can now successfully use Point One Nav for corrections:

[image]

Sorry for that!

commit 10d5a924 (2025-01-20): Push GGA on non ZED platforms

Fix #515
closed 2025-02-20 by nseidle