sparkfun/SparkFun_RTK_Everywhere_Firmware #535: Use RTK Everywhere with an ESP32-PICO-DevKitM-2

GeomPozzoli, 2025-02-04:
Hello everyone,
I wanted to ask if it is possible to use the firmware RTK Everywhere with an ESP32-PICO-DevKitM-2 and a generic UM980 breakout board

I tried to compile the source code by commenting out the non-essential parts in the file RTK_Everywhere.ino (other board, imu, etc.) and using the custom partition file RTKEverywhere_8MB.csv
Once the firmware has been loaded, when I start the ESP32 without connecting the GNSS card via the serial connection I get the following message:

```
=======================
Board ADC ID (mV): 3098
=======================
Identified variant: Unknown
========================
Device MAC: D4D4DA8D5776
========================
HALTED: Product variant unknown. Unable to proceed. Please contact SparkFun with the "Device MAC" and the "Board ADC ID (mV)" reported above.
```

Looking inside the code in the Begin.ino file, I found that based on the mV it defines which board is and then the pins are defined, 
but it is not exactly clear to me if it is possible to insert a new generic board and what are the criteria to calculate the tolerance. 

Is it possible to have some advice on this?   

I probably got into something that is beyond my reach.... and I should continue to simply use the UART BT bridge

Have a good day
Stefano

nseidle, 2025-02-05:
Welcome! And congratulations for compiling from source.

If I were you, I would force the board identification. For example, at the end of identifyBoard(), I would 

    productVariant = RTK_CUSTOM; // You'll have to add this enum

Then, within beginBoard() add an RTK_CUSTOM platform and define how your hardware is connected and what hardware is available. For example:

    else if (productVariant == RTK_CUSTOM)
    {
        // Specify the GNSS radio
    #ifdef COMPILE_UM980
            gnss = (GNSS *)new GNSS_UM980();
    #else  // COMPILE_UM980
            gnss = (GNSS *)new GNSS_None();
            systemPrintln("<<<<<<<<<< !!!!!!!!!! UM980 NOT COMPILED !!!!!!!!!! >>>>>>>>>>");
    #endif // COMPILE_UM980
    
            present.brand = BRAND_SPARKFUN;
            present.psram_2mb = true;
            present.gnss_um980 = true;
            present.gnss_to_uart = true;
            present.galileoHasCapable = true;
            present.multipathMitigation = true; // UM980 has MPM, other platforms do not
            present.minCno = true;
            present.minElevation = true;
            present.dynamicModel = true;
    
            pin_I2C0_SDA = 15;
            pin_I2C0_SCL = 4;
    
            pin_GnssUart_RX = 26;
            pin_GnssUart_TX = 27;
    
            pin_bluetoothStatusLED = 32;
            pinMode(pin_bluetoothStatusLED, OUTPUT);
            bluetoothLedOn();
    
            settings.dataPortBaud = 115200; // Override settings. Use UM980 at 115200bps.
        }

Above, you will need to set your pin numbers according to how you wire the UM980 and I2C bus on your board.

I personally would use any general LED you have available on the ESP32-PICO-DevKitM2 board to indicate Bluetooth connectivity. 

Good luck and let us know how it goes!

closed 2025-02-05 by nseidle

rcdemuth, 2025-03-23:
@GeomPozzoli did you get this working? Would you mind sharing your code if so?