sparkfun/SparkFun_RTK_Everywhere_Firmware #604: Lora serial settings

ryanmurray, 2025-03-24:
Trying to find a way to get Lora to work with postcard since it's quite noisy. Maybe I need to disable much of the nmea on the base. But I also noted for another project using rp2040

Our system uses the Raspberry Pi RP2040 processor on the main board to interact with the GPS receiver and the radio. The software is written for the Arduino IDE and is available, at our Serial Modem project on GitHub. For the base station the RP2040 receives the RTCM formatted correction data from the GNSS receiver and then sends it out in packets over the LoRa radio. LoRa packets are restricted to 256 bytes by the protocol, so the RP2040 breaks up the messages into that sized packets. On the rover, the RP2040 receives the packets from the LoRa radio, parses them into messages (including data spread across packets), and then sends them to the RXD2 port on the rover’s GNSS receiver. The GNSS receiver uses these RTCM messages to produce an an RTK solution with centimeter level accuracy.

Is there a way to break up packets? 

I haven't found a definitive guide to what rtcm opus needs but my goal is to set up a base logging while providing corrections over Loraserial and have my rover do work and correct my base later. Espnow works fantastic so I believe I'm overload lora.

PaulZC, 2025-03-25:
Hi Ryan (@ryanmurray ),

Please post a link to your Serial Modem project. Thanks!

Best wishes,
Paul

ryanmurray, 2025-03-25:
https://tinkerbugrobotics.com/how-to-send-rtk-corrections/

Paul I am using sparkfuns Lora serial radios. I believe this may be an issue of the postcard being very noisy. Reading the manual a little more last night is there guidance (I tried low bandwidth defaults) on maybe what minimal nmea messages are needed (maybe none?).

Also am I understanding I could turn the frequency of the rtcm messages down? 

Finally my idea was to adjust the Lora airspeed. 

Basically I'm just trying to get Lora to work with the postcard (I've also tried with the facet as a base) since espnow works great I'm assuming that the link is just way too noisy. So maybe some guidance on messages for rinex conversion, rtcm messages needed to the rover for RTK, minimal nmea messages. I was going to try to play around a little more but where others gave up on the Loraserial I believe it probably can work.

ryanmurray, 2025-04-01:
Think I found a better solution in rover mode disable uart3 transmit. Once I disconnected the jst wire the Lora worked fine from base to rover. I used 19200 airspeed I am going to retest decreasing that.

PaulZC, 2025-04-01:
Thanks for the update Ryan (@ryanmurray ),

We haven't yet had time to look into this. Please keep the updates coming - and thank you for spending time on this.

Best,
Paul

ryanmurray, 2025-04-01:
Exactly Nathan and since rtk postcards just have the one port for uart 3 that's what I've been connecting loras to, as soon as I disconnected the transmit wire on the rover postcard I got corrections. It was a moment of glory. On the base I disabled nmea and only broadcast 1005, and the msm7 messages for glo, gal, and gps. I obtained an instant 8mm by 11mm fix. Basically the rover back feeding over serial (I'm in multipoint mode btw) overwhelms the link. I was using the jst cables. as you stated the facet I own does not behave this way. And would make sense why 2 torches can do it fine also. 

The only issue I could see ( I use corrections on my phone when using state network) is that you need gga on that but my application does that versus the rover itself. But maybe if someone was using onboard ntrip that would break?

cturvey, 2025-04-01:
I had suggested removing the Radio RX pin connectivity on the Rover side to basically prove/demonstrate in a hard fashion whether the traffic there was an issue or not.
The LoRa radio basically being a half-duplex channel, and synchronizing to the burst output of the GNSS aligned to top-of-second, and potentially self-jamming or treading on itself.
So basically the ability of Rover(s) to be silent on radio side comms. I've generally avoided disabling radio side UART RX, as this just makes (re)configuration and enabling debugging extra difficult. But data TX over radio could be gated or suppressed there too.
Now for NTRIP, in a one-to-one connection, you typically need GLL or GGA to establish a location to find closest mount point, allow for VRS, act as a walled garden, or establish continued life/operation of the connection

Jeffpostcn, 2025-05-02:
> I believe what you are asking is for the ability to disable serial transmission on the RADIO port, during rover, correct?
> 
> * Postcard - Turning off messages from UARTX on UARTY was a feature that was not supported in the stock firmware of the LG290P. We'll need to build in version checking of the receiver firmware but it should be doable.
> * EVK - The ZED is connected to the external TX2/RX2. We can disable NMEA/RTCM/UBX on the outgoing UART2.
> * Torch - No RADIO port but do we suppress serial from Rover flowing to the LoRa radio? I'm pretty confident but I need to confirm.
>...

Ok this Comment has me thinking I do not really understand the "Configure Ports" Serial Menu for the RTK Postcard:

[image]

My thinking was that option 1 was to set the Baud Rate for the internal ESP-NOW radio and option 2 set the baud rate for the "4-pin locking JST-GH connector" which I *now* see  listed on the Schematic as the RADIO Port 🤯.  Please Confirm? also what then is the "Data Port" ?  (My guess: not functional item for the RTK Postcard?)

nseidle, 2025-05-02:
RADIO port is the JST-GH connector. DATA is USB (C connector) port B on the CH342.

PaulZC, 2025-07-10:
The ports menu now looks like this - on mosaic-X5. Postcard has the same option.

[image]

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/release_candidate/docs/img/Terminal/SparkFun%20RTK%20Everywhere%20-%20Ports%20Menu.png

nseidle, 2025-07-16:
This feature has been added to v2.2 and is documented [here](https://docs.sparkfun.com/SparkFun_RTK_Everywhere_Firmware/menu_ports/#limit-radio-port-output-to-rtcm).

PR #668 (2025-07-10): Add print profile option to menuUserProfiles
Resolves #609 

Also resolves #604
closed 2025-07-16 by nseidle