sparkfun/SparkFun_RTK_Everywhere_Firmware #635: Error in the accuracy display on the portability protector screen. (postcard)

amlago, 2025-05-07:
Error in the accuracy display on the portability shield screen.

In rover mode, the accuracy displayed on the screen is roughly the same as that reported by the control program, such as SWmaps or GNSS Master.
However, if ntrip corrections are provided, the accuracy reported on the display is several meters. I obtained up to 14 meters, and the accuracy in the app was within a few decimeters, for example, 0.45 or 0.35.
When the position fix is ​​obtained, the accuracy is corrected on the display and equals that shown in the app.
Something is wrong.
Regards

nseidle, 2025-05-21:
Hi amlago - what version of the firmware do you have loaded onto the Postcard?

nseidle, 2025-05-21:
> However, if ntrip corrections are provided, the accuracy reported on the display is several meters. I obtained up to 14 meters, and the accuracy in the app was within a few decimeters, for example, [...]

We are displaying the value that the LG290P is reporting as its horizontal accuracy. I know that SW Maps is calculating its own horizontal accuracy from the NMEA sentences so you will see some discrepancy. Who is correct? SW Maps or the LG290P? For now, I will trust the LG290P.

Below, firmware version RTK_Everywhere_Firmware_v2_1 shows a reasonable 18mm during RTK Fix 

[image]

Below, SW Maps shows a much better accuracy of 6mm:

[image]

During RTK Float (the phase where you report 14.0m horizontal accuracy) it could be true that the LG290P reports a larger horizontal accuracy, while SW Maps reports a smaller one. We don't have control over this. All HPAs should be taken with a grain of salt, especially during RTK Float.

Since we can't change the LG290P behavior, I'm going to close this issue. Hopefully future LG290P firmware releases will show some change in behavior (I did my testing with v5, the latest, unfortunately).

closed 2025-05-21 by nseidle

amlago, 2025-05-21:
Hi Nathan. It's actually unexpected behavior. Facet behaves differently, and its reported accuracy begins to improve, as expected, after making corrections via ntrip.
What I see on the portability protector screen is the opposite.
I start with a 1-meter error and make corrections. Instead of improving, it worsens, reaching values ​​of 14 meters on one occasion, and then plummets when I reach fix mode, where the values ​​on the screen and the app you're using coincide.
I use several apps like Landstar 8, Survey Master, SW Maps, etc.
I'm not convinced that this huge number is correct, but...

amlago, 2025-05-21:
If the correction is very fast you won't notice it, but if you are connected to an area with poor internet access and normal to low satellite reception, then the values ​​in question will appear.