sparkfun/SparkFun_RTK_Everywhere_Firmware #651: [Release Candidate] The TCP Server/Client and UDP Server do not work

nseidle, 2025-05-21:
The TCP Server will start a local WiFi AP rather than connect to designated WiFi networks.

wjc10240, 2025-07-09:
The use of the "Configure device via WiFi Access Point or connect to WiFi" setting to control the WiFi behavior of operational modes (Rover, Base) as well as the WiFi behavior of Configure mode is undesirable.

For some uses, e.g. Rover mode with QField running on iOS, the operational Rover mode would need this setting to be set to "WiFi". But if there were a problem in the field and the receiver was unable to connect to the configured WiFi net, and no computer and USB cable were available, the user would be unable to connect to the receiver in Configure mode. Yes, if another of the receiver's profiles had this setting set to "AP", then the user could switch to that other profile and rebuild the entire, desired configuration — but that could be quite inconvenient and error-prone

The WiFi behavior in Configure mode and the WiFi behavior in operational modes are logically distinct, and they should have separate settings.

If implementing a separate WiFi behavior setting for the operational modes is not feasible for the next RC or full release, then there should at least be a change of the name of the existing setting to something like "WiFi Mode - AP or connect to existing net". The current name is misleading, in that it implies that the setting affects only Configure mode.

Should I open a new issue for this?

nseidle, 2025-07-09:
I agree. We'll add a setting to the TCP/UDP configuration shortly.

In the mean time - configuration should fall back to AP if WiFi station does not connect. I'll have to test and solve that issue so that users don't get locked out (there's always [Bluetooth](https://docs.sparkfun.com/SparkFun_RTK_Everywhere_Firmware/configure_with_bluetooth/)).

nseidle, 2025-07-10:
What are the four apps you've tried? Current tested iOS apps are [here](https://docs.sparkfun.com/SparkFun_RTK_Everywhere_Firmware/gis_software_ios/).

nseidle, 2025-07-11:
@wjc10240 - The TCP setting feature can be monitored on [this issue](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/issues/685). What you are also reporting is a BLE issue. I'll open a new issue for that.

I believe this original issue has been fixed.

closed 2025-07-16 by nseidle