sparkfun/SparkFun_RTK_Everywhere_Firmware #675: During SoftAP Web config entering IP address fails to load page

nseidle, 2025-07-08:
To replicate:

* Factory reset
* Put device into Web Config mode
* Connect to "RTK Config ####" from phone
* Open browser on phone
* Enter 192.168.4.1 <- page fails to load
* Enter 'rtk.local' <- page will load

nseidle, 2025-07-10:
Different android phone: IP / rtk.local works
Back to my primary phone: rtk.local works. IP works in incognito mode. I suspect this issue is specific to something weird on my phone. I think it's got an http*s*://192.168.4.1 that it is insisting on using.

I've got captive portal working separately. I'll get that pushed to a branch and we'll test more on iPhone.

PaulZC, 2025-07-10:
I don't know if the Socket issue is RAM-related?
I just tried increasing the two stack size settings to 16K, [closer to where they were previously](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blame/ab8937474f50d931ed9159b04ebaab5b5e8ed8e4/Firmware/RTK_Everywhere/WebServer.ino#L67-L68), and - after a few unscientific tests - the Socket seems better behaved.
No obvious change on the 192.168.4.1 vs rtk.local issue though...

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blame/ab8937474f50d931ed9159b04ebaab5b5e8ed8e4/Firmware/RTK_Everywhere/WebServer.ino#L67-L68

LeeLeahy2, 2025-07-10:
On my Android phone, both 192.168.4.1 and rtk.local seem to connect fine.  However for both addresses, the initial load shows the green buttons down to "System Configuration" are displayed followed by the buttons "Save Configuration" and "Exit and Reset".  Reloading the page adds the "Ethernet Configuration", "NTP Configuration" and "File Manager" buttons.

nseidle, 2025-07-10:
MDNS is not enabled during AP. I've got that fix into the captive portal branch. That fixes my iPhone.

    static const int webServerTaskStackSize = 1024 * 4;
    static const int webSocketStackSize = 1024 * 20; // Needs to be large enough to hold the file manager file list

Working well with ~50 files on the SD card. Moving to testing with EVK and a full stack to stress RAM.