sparkfun/SparkFun_RTK_Everywhere_Firmware #681: Crash in u-blox library, may be related to I2C stack

LeeLeahy2, 2025-07-10:
Bluetooth SPP and BLE broadcasting as: EVK Rover-8F16                                                                                                                                                              
Network: Calling networkUserRemove(WEB_CONFIG) from /home/lee/SparkFun/rc/Everywhere/Firmware/RTK_Everywhere/WebServer.ino at line 1159                                                                            
Network: Calling networkInterfaceInternetConnectionLost(Ethernet)                                                                                                                                                  
Ethernet previously lost internet access                                                                                                                                                                           
Rover Accuracy (m): > 30.000, SIV: 0 GNSS State: No Fix                                                                                                                                                            
Provisioning: PROVISIONING_OFF --> PROVISIONING_CHECK_REMAINING                                                                                                                                                    
requestKeyUpdate is true.                                                                                                                                                                                          
Starting provisioning                                                                                                                                                                                              
Network: Calling networkConsumerAdd(POINTPERFECT_KEY_UPDATE, NETWORK_ANY) from /home/lee/SparkFun/rc/Everywhere/Firmware/RTK_Everywhere/menuPP.ino at line 1521                                                    
rtkMode: 0x08 (RTK_MODE_ROVER)                                                                                                                                                                                     
Network: Adding consumer POINTPERFECT_KEY_UPDATE                                                                                                                                                                   
Network: Calling networkStart(Cellular) from /home/lee/SparkFun/rc/Everywhere/Firmware/RTK_Everywhere/Network.ino at line 469                                                                                      
Starting network: Cellular                                                                                                                                                                                         
Network: Calling networkSequenceStart(Cellular) from /home/lee/SparkFun/rc/Everywhere/Firmware/RTK_Everywhere/Network.ino at line 2205                                                                             
Cellular sequencer idle                                                                                                                                                                                            
--------------- Cellular Start Sequence Starting ---------------                                                                                                                                                   
Cellular: Stopped --> Starting                                                                                                                                                                                     
rtkMode: 0x08 (RTK_MODE_ROVER)                                                                                                                                                                                     
Provisioning: PROVISIONING_CHECK_REMAINING --> PROVISIONING_WAIT_FOR_NETWORK                                                                                                                                       
Cellular: Verify the LARA priority                                                                                                                                                                                 
--------------- WiFi Station Starting ---------------                                                                                                                                                              
WIFI_STATION_STATE_OFF --> WIFI_STATION_STATE_STARTING                                                                                                                                                             
wifiStationOn called in /home/lee/SparkFun/rc/Everywhere/Firmware/RTK_Everywhere/WiFi.ino at line 1025                                                                                                             
WiFi: Starting (Station)                                                                                                                                                                                           
SD card size: 119.1 GB / Free space: 118.7 GB                                                                                                                                                                      
event: 9 (WIFI_READY)                                                                                                                                                                                              
event: 11 (STA_START)                                                                                                                                                                                              
WiFi Event: Station start: MAC: e0:5a:1b:d8:8f:14                                                                                                                                                                  
WiFI starting scan for remote APs                                                                                                                                                                                  
event: 10 (SCAN_DONE)                                                                                                                                                                                              
WiFi scan complete, found 41 remote APs                                                                                                                                                                            
 dBm   Chan   Authorization     SSID                                                                                                                                                                               
----   ----   ---------------   --------------------------------                                                                                                                                                   
 -25      9   WPA2_PSK   Work                                                                                                                                                                                      
WiFi: Found remote AP: Welo 79                                                                                                                                                                                     
 -35      8   WPA2_PSK   Welo 79                                                                                                                                                                                   
 -48      6   WPA2_PSK   setup3A130                                                                                                                                                                                
 -61     11   WPA2_PSK   setup37ED0                                                                                                                                                                                
 -63      4   WPA2_PSK   Welo 80                                                                                                                                                                                   
 -70     11   WPA2_PSK   setup01C5.ybd                                                                                                                                                                             
[... cut ...]
                                                                                                                                                                                                                   
      PC        Stack Ptr   Symbol
----------   ----------   --------------------
0x4009b27a   0x3ffd3c80   vPortTaskWrapper
0x4015cf74   0x3ffd3c60   loopTaskP
0x4011864c   0x3ffd3c40   loop
0x400ec395   0x3ffd3c20   updateLBandCorrections
0x4013379e   0x3ffd3c00   DevUBLOXGNSS10checkUbloxEh
0x4013375b   0x3ffd3bd0   DevUBLOXGNSS18checkUbloxInternalEP9ubxPacketh
0x40133540   0x3ffd3b90   DevUBLOXGNSS13checkUbloxI2CEP9ubxPacketh
0x4030151f   0x3ffd3b70   DevUBLOXGNSS9availableE

nseidle, 2025-07-10:
What are the steps to reproduce this?

LeeLeahy2, 2025-07-12:
> Ah. Hang on... Each GNSS instance (ZED and NEO) currently uses their own
> independent iAmLocked semaphore. As a minimum, we need them to share a
> global gnssIsLocked semaphore.
>
> But this is only an issue if any of the tasks running in the background
> attempt to access the GNSS "asynchronously". The way we have the loop
> structured will prevent most badness. gnss->update() and displayUpdate()
> are called consecutively and separately.
>
> The crash occurs during updateLBandCorrections (i2cLBand.checkUblox())
> which is called by the loop - well after gnss->update(). It seems
> unlikely that it is the ZED and NEO fighting over the bus...
>
> Looking at the ticker background tasks, are any of them attempting to
> access the GNSS or even just the I2C bus? tickerGnssLedUpdate calls
> gnss->isRTKFix(). On the ZED, isRTKFix() doesn't generate any bus
> traffic. It only returns the saved _carrierSolution. It's not that. And
> the EVK doesn't have a gnssStatusLED anyway. It does have pin_baseStatusLED
> though.
>
> Oh. This might be it...? If the PointPerfect Library task is running,
> updatePplTask is called periodically. In there, if the PPL is generating
> corrections, zed->updateCorrectionsSource(1); can be called. And that
> _does_generate bus traffic (_zed->setVal8). It also does gnss->pushRawData
> from inside the task. Yeah. That's potentially bad. If that happens while
> [... 27 lines cut ...]
> connector has its own auxiliary I2C bus.
>
> Facet Flex: if the GNSS supports I2C, it will be shared with the OLED,
> charger (MP2762AGV), USB Hub (U5), the battery, the authentication IC, I/O
> expander (TCA9534).

Hi Paul,

The I2C driver in the core (NOT the TwoWire library) does use a semaphore
to access the I2C bus.

nseidle, 2025-07-16:
0x40214773: DevUBLOXGNSS::available() at C:\Users\nathan.seidle\Dropbox\Apps\Arduino\user\libraries\SparkFun_u-blox_GNSS_v3\src\u-blox_GNSS.cpp line 771
      0x401256e0: DevUBLOXGNSS::checkUbloxI2C(ubxPacket*, unsigned char, unsigned char) at C:\Users\nathan.seidle\Dropbox\Apps\Arduino\user\libraries\SparkFun_u-blox_GNSS_v3\src\u-blox_GNSS.cpp line 1165
      0x401258fb: DevUBLOXGNSS::checkUbloxInternal(ubxPacket*, unsigned char, unsigned char) at C:\Users\nathan.seidle\Dropbox\Apps\Arduino\user\libraries\SparkFun_u-blox_GNSS_v3\src\u-blox_GNSS.cpp line 1140
      0x4012593e: DevUBLOXGNSS::checkUblox(unsigned char, unsigned char) at C:\Users\nathan.seidle\Dropbox\Apps\Arduino\user\libraries\SparkFun_u-blox_GNSS_v3\src\u-blox_GNSS.cpp line 1129
      0x400e6c08: updateLBandCorrections() at C:\Users\nathan.seidle\Dropbox\Projects\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/menuPP.ino line 746
      0x40109d4b: loop() at C:\Users\nathan.seidle\Dropbox\Projects\SparkFun_RTK_Everywhere_Firmware\Firmware\RTK_Everywhere/RTK_Everywhere.ino line 1403
      0x4013fe1c: loopTask(void*) at C:\Users\nathan.seidle\Dropbox\Apps\Arduino\data\packages\esp32\hardware\esp32\3.0.7\cores\esp32\main.cpp line 74
      0x40093142: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 162

To replicate:

* RC branch
* EVK, factory reset
* Give WiFi credentials
* Enable MQTT IP deprecated service

Unit will get keys, apply them, then reset:

      Skipping ZED Rover configuration
      [  6943][D][NVM.ino:199] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
      STATE_ROVER_NOT_STARTED --> STATE_ROVER_NO_FIX, 2025-07-16 15:00:50.391
      Rover Accuracy (m): 0.214, SIV: 32 GNSS State: 3D Fix
      STATE_ROVER_NO_FIX --> STATE_ROVER_FIX, 2025-07-16 15:00:50.440
      Days until PointPerfect keys expire: 34
      PointPerfect keys applied
      [ 11929][D][NVM.ino:199] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_EVK_Settings_0.txt
      Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
      
i2cLBand.checkUblox() is getting called uninitialized. Bug is likely due to the fact that the EVK is the only device on the PSRAM product line that has the NEO-D9S L-Band receiver.

nseidle, 2025-07-16:
This is fixed.

closed 2025-07-16 by nseidle