sparkfun/SparkFun_RTK_Everywhere_Firmware #699: PointPerfect not connecting to Wifi After 2.2 upgrade on Postcard With Portability Shield

joef12345, 2025-07-18:
I had valid keys before the upgrade, and I have PointPerfect enabled; however, the Postcard does not attempt to connect to wifi while in rover mode to pull correction data. I even tried to enable TCP Server, but the device does not attempt to pull correction data. I am also now missing the option to update the keys.

joef12345, 2025-07-19:
Thank you, I am new to the RTK and PointPerfect and did not know about the change. 

Here is my ID is: 64B70885E45A05

And here is my current setting for PP.

[image]

Thanks!

nseidle, 2025-07-21:
Got it. I've added 64B70885E45A05 to the allowlist for Flex NTRIP/RTCM. Please make sure your unit has WiFi credentials, then reboot it. It should now get NTRIP credentials and start downloading corrections.

closed 2025-11-14 by nseidle