sparkfun/SparkFun_RTK_Everywhere_Firmware #735: Postcard RTK: Choose msm4 or msm7 message format

amlago, 2025-09-27:
Postcard RTK: Choose msm4 or msm7 message format

I need to choose the msm4 or msm7 message format.
The Postcard settings only allow you to choose between on and off.
I would like to be able to choose between msm4 and msm7, like in the Facet settings.
Regards
Angel

amlago, 2025-10-02:
Hi, any answers on msm7 in Postcard RTK?
I need it urgently because I need rinex with full GLONASS data, and msm4 is missing phase data.
Regards

PaulZC, 2025-10-03:
Hi Angel (@amlago ),

The LG290P MSM selection has changed frequently over the past few weeks as we add better support for PPP logging.

For MSM7, it looks like Release 2.2 may currently be your best option?

To make it 'switchable', we will need to add a new setting with full menu, command and configuration support, and implement it on all other supported GNSS platforms. It is certainly possible, and we will add it, but it is not a 'simple' change.

I hope this helps,
Paul

**In the main branch (release v2.2):**

`enableRTCMBase` :

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/68884d8f5fb5d608c3a454e69378a5b1a2d6d675/Firmware/RTK_Everywhere/GNSS_LG290P.ino#L801

`enableRTCMRover` :

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/68884d8f5fb5d608c3a454e69378a5b1a2d6d675/Firmware/RTK_Everywhere/GNSS_LG290P.ino#L941

**In today's release_candidate (unreleased):**

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/68884d8f5fb5d608c3a454e69378a5b1a2d6d675/Firmware/RTK_Everywhere/GNSS_LG290P.ino#L801

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/68884d8f5fb5d608c3a454e69378a5b1a2d6d675/Firmware/RTK_Everywhere/GNSS_LG290P.ino#L941

**In release_candidate on September 30th:**

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/5e8192f2617237f7dbaa2173f403898d2ba8e257/Firmware/RTK_Everywhere/GNSS_LG290P.ino#L853

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/5e8192f2617237f7dbaa2173f403898d2ba8e257/Firmware/RTK_Everywhere/GNSS_LG290P.ino#L1013

**In release_candidate on September 22nd:**

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/b3b04923087bd88ce8702274d3c6ce06a9494bed/Firmware/RTK_Everywhere/GNSS_LG290P.ino#L853

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/b3b04923087bd88ce8702274d3c6ce06a9494bed/Firmware/RTK_Everywhere/GNSS_LG290P.ino#L1013

**In release_candidate on September 19th:**

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/ec5b17f701aca06b5e3650b2d789933fa55cc101/Firmware/RTK_Everywhere/GNSS_LG290P.ino#L853

https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/blob/ec5b17f701aca06b5e3650b2d789933fa55cc101/Firmware/RTK_Everywhere/GNSS_LG290P.ino#L1013

amlago, 2025-10-03:
Hi Paul.
Thanks for your concern.
I know each change is a lot of work.
Ideally, it should be switchable between msm4 and msm7.

Regards
Angel

cturvey, 2025-10-03:
I didn't do a comprehensive search of the usage, initially threw it at the whole of GitHub to find the string/command, but this looked like the one file in the SparkFun domain.

I suspect having it adjustable is helpful, the MSM4 are definitely more compact and a better choice in lower bandwidth radio links where size and speed/latency.

MSM4 vs MSM7 there's also a precision aspect, the doppler can be inferred, and is often used in a code vs carrier sense to identify cycle slippage. When I was checking RINEX files I'd try and map the code and carrier advancement into Hz. The expectation being about +/- 3 KHz as I recall, but this pulls one way or another depending if the receiver's local clock is fast or slow. On uBlox they measure the clock rather than try an discipline it explicitly like say LEICA, and expand the doppler search space to perhaps +/- 5 KHz to find the skewed signal.

Ideally for RINEX that we use raw measurements more directly in a native form like uBlox's UBX-RXM-RAWX, where the post-processing solves for receiver time (offset, bias, drift, etc) it ends up with more information to work with and doesn't have to un-cook things. Not sure if Quectel has a raw format or if that just generates way more data than desired.

PaulZC, 2025-10-24:
Hi Angel,

Please tell us which version of RTK Everywhere Firmware you are using. Is it the latest Release Candidate?

Please tell us which version of LG290P firmware you have installed.

Please tell us if you have Galileo HAS enabled.

The MSM Selection option should be available in: the Configure GNSS Messages menu; and in the Base \ RTCM Messages menu.

[image]

[image]

If I enter WiFi web config mode, change that setting, save and exit, I see the change:

[image]

If I then put the Postcard into Base mode, give it WiFi credentials, enable NTRIP, I see MSM7 on RTK2go:

[image]

All seems well?

Best wishes,
Paul

amlago, 2025-10-24:
Hi Paul
In short, after a couple of reboots it worked.
The strange thing is that yesterday I switched to msm7 with Wi-Fi config and it was still in msm4 despite the msm7 check mark.
I also found it in Bluetooth config, my apologies. It must be due to tiredness from the workday.
I'm using the latest v2.3 from 10/15 and v06 test for the LG290p.
It works fine.
I don't know if it needs an extra reboot to get it right.
Thanks, Paul.

amlago, 2025-10-24:
Thank you very much, Paul.
It was a pleasure working with you.
Best regards
Angel

TCIII, 2025-10-28:
@PaulZC 
Thanks for the response much appreciated.
I was basing my perception of requiring MSM7 for the LG290P to reach a RTK Fix solution on this [post](https://community.sparkfun.com/t/cant-get-better-than-rtk-float-with-the-sparkfun-lg290p-gnss-rtk-breakout-module/67085) I made on the Sparkfun GPS/GNSS Forum.
Based on your response, I have tried to go back to RTK Everywhere firmware version "RTK_Everywhere_Firmware_v2_2.bin" and LG290P firmware version "LG290P03AANR01A05S.pkg" on the RTK Postcard that was working with the ZED-F9P RTK GNSS module on the Rover, but now I cannot set the Telemetry Radio baud rate to 57600 as the RTK Everywhere UI keeps saying "the value is out of range".
Suggestions as to how I can get the RTK Postcard Radio baud rate set to 57600 so I can continue to troubleshoot the issue of the LG290P RTK GNSS module not reaching a RTK Fix solution on the Rover?

Regards,
TCIII

TCIII, 2025-10-28:
@PaulZC 
Thanks for the quick response.
On the other hand, I was able to use QGNSS app and the Console tool to change the UART3 baud rate to 57600 using the following commands:
$PQTMGNSSSTOP*09    Stop the GNSS Engine
$PQTMCFGUART,W,3,57600*34    Change UART3 to 57600
$PQTMSAVEPAR*5A    Save
$PQTMSRR*4B   Reboot

The RTK Everywhere UI also shows that the Radio baud rate changed from the default 115200 to 57600.

So it looks like there is an error in the LG290P Protocol Specification where it precludes the 57600 baud rate?

I performed the same baud rate change with the Rover Sparkfun LG290P RTK GNSS module and I now have the RTK Postcard Base Station communicating with the Sparkfun LG290P RTK GNSS module Rover over the telemetry radio link at 57600 baud.

Now, about the Sparkfun LG290P RTK GNSS Breakout module on the Rover not obtaining a RTK Fix solution. My Sparkfun ZED-F9P RTK GNSS module on the same Rover obtains a RTK Fix solution when communicating with the RTK Postcard while the Sparkfun LG290P RTK GNSS module will only obtain a RTK Float solution when placed in the same location outdoors and communicating with the RTK Postcard. My Rover has only an L1/L2 active antenna which works with the Sparkfun ZED-F9P RTK GNSS module. Since the Sparkfun LG290P RTK GNSS module is L1/L2/L5 capable, could that be an issue?

Can you suggest how I might trouble shoot this issue with the Sparkfun LG290P RTK GNSS module not obtaining a RTK Fix solution when used under the same outdoor conditions where the Sparkfun ZED-F9P RTK GNSS module obtains a RTK Fix solution?

Should I open an issue ticket on the Sparkfun RTK Postcard/Firmware Binaries webpage?

Regards,
TCIII

PaulZC, 2025-10-28:
Hi @TCIII ,

I don't understand your statement:

> The RTK Everywhere UI also shows that the Radio baud rate changed from the default 115200 to 57600.

Can you please explain further? And or attach a screenshot?

Any changes you make to the Postcard LG290P with QGNSS are likely to be overwritten by the RTK Firmware when it restarts.

Best,
Paul

TCIII, 2025-10-28:
@nseidle 
Then how was I able to use the QGNSS app, the Console tool, and the commands I posted earlier to configure the Sparkfun LG290P RTK GNSS Breakout Board UART3 for 57600 baud on the Rover and have it communicate with my Sparkfun ZED-F9P Base Station which also transmits at 57600 baud on UART3?

Regards,
TCIII

gdt, 2025-10-29:
That can't be right.   "No RTCM correction data is being received" is not the same thing as "the receiver is computing autonomous or WAAS positions".  For any RTK receiver, it is possible, and there's always a brief  period, when RTCM3 reference data (it's really not corrections but that terminology misuse is regrettably near universal) has started but the receiver has not yet gone into FLOAT.

Calling FLOAT "rough but potentially inaccurate" is odd.  I'd call it an "approximate RTK solution, very likely within a meter".   In my experience float solutions are ~never cm accurate and there is almost always a 10s-of-cm jump in the tracklog on the transition from FLOAT to FIX.

But none of that's really about the LG290P.

You're asking about this, but what do you see when you boot up the board and start getting RTCM3?   Does it go from off to blink to solid?   Or do you have evidence that something else is going on?

gdt, 2025-10-29:
When you examined the NMEA stream, what did you find out?  It seems like you are expecting this to work by asking questions instead of by digging in to the data.  That strategy seems unlikely to be successful in my view.

TCIII, 2025-10-29:
@gdt 
The Data View of the QGNSS app indicates the following:
Fix Mode is blank
Quality Indicator is Float RTK
RTK Fixed is 0
RTK Float is an ever increasing four digit value.

Regards,
TCIII