sparkfun/SparkFun_RTK_Everywhere_Firmware #738: Bug in Facet mosaic Base configuration height calculations.

wjc10240, 2025-09-29:
I believe I’ve found a bug in version 2.2 of the Facet mosaic firmware. I’m using the serial interface to set the receiver’s Base configuration. See the attached screen shot. It looks to me as if the sum of Antenna Height and Antenna Phase Center, both in millimeters, is being divided an extra time by 1,000 before being added to the ellipsoidal height from line 3.

I believe the correct value for Total Height Above Ellipsoid (HAE APC) should be 71196.500 mm.

[image]

PaulZC, 2025-10-02:
Hi Bill (@wjc10240 ),

Many thanks for reporting!

It looks like the error crept in in this commit: https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/commit/6fa9a46653c8681dda0a5d54ab60f4853bb88133

We were doing some tidying up, adding the units to some of the variable names, and it looks like we managed to confuse ourselves in that calculation.

[image]

Looking at the code, I'm confident this is limited to that one line. `settings.fixedAltitude` seems to be used correctly everywhere else.

We'll get this fixed in the next release.

Thanks again!
Paul

PaulZC, 2025-10-07:
This is in pcUpdates - commit eca8175c264d00217990b610c2029e50c0a587d4

nseidle, 2025-11-14:
This was included in the [v2.3 release](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/releases/tag/v2.3).

closed 2025-11-14 by nseidle